openbmc/openbmc #3672: Setting bad_pass_thresh is not working on IPMI over LAN

sharady, 2020-07-06:
Limiting invalid username/password attempts must be supported while accessing BMC IPMI over LAN.

**To Reproduce**
Default OpenBMC setting for Bad Password Threshold is `Not Available`:
`$ ipmitool -I lanplus -C 17 -p 5623 -U root -P 0penBmc -H 127.0.0.1 lan print 1
Set in Progress         : Set Complete
Auth Type Support       :
Auth Type Enable        : Callback :
                        : User     :
                        : Operator :
                        : Admin    :
                        : OEM      :
IP Address Source       : DHCP Address
IP Address              : 10.0.2.15
Subnet Mask             : 255.255.255.0
MAC Address             : 36:91:09:fc:bd:1f
Default Gateway IP      : 10.0.2.2
Default Gateway MAC     : 00:00:00:00:00:00
802.1q VLAN ID          : Disabled
RMCP+ Cipher Suites     : 17
Cipher Suite Priv Max   : aaaaaaaaaaaaaaa
                        :     X=Cipher Suite Unused
                        :     c=CALLBACK
                        :     u=USER
                        :     o=OPERATOR
                        :     a=ADMIN
                        :     O=OEM
**Bad Password Threshold  : Not Available**
`

Setting `bad_pass_thresh` ipmi over lan is supported as per below parameters.
`bad_pass_thresh <thresh_num> <1\|0> <reset_interval> <lockout_interval>
Sets the Bad Password Threshold.
<thresh_num> If non-zero, this value determines the number of sequential bad passwords that will be allowed to be entered for the identified user before the user is automatically disabled from access on the channel.
<1\|0> 1 = generate a Session Audit sensor "Invalid password disable" event message. 0 = do not generate an event message when the user is disabled.
<reset_interval> Attempt Count Reset Interval. The interval, in tens of seconds, for which the accumulated count of bad password attempts is retained before being automatically reset to zero.
<lockout_interval> User Lockout Interval. The interval, in tens of seconds, that the user will remain disabled
`

Change the bad_pass_thresh values.
`$ ipmitool -I lanplus -C 17 -p 5623 -U root -P 0penBmc -H 127.0.0.1 lan set 1 bad_pass_thresh 3 1 3 3`

**Unexpected behavior**
The bad password threshold is not getting updated, still shows `Not Available`.
`ipmitool -I lanplus -C 17 -p 5623 -U root -P 0penBmc -H 127.0.0.1 lan print 1
Set in Progress         : Set Complete
Auth Type Support       :
Auth Type Enable        : Callback :
                        : User     :
                        : Operator :
                        : Admin    :
                        : OEM      :
IP Address Source       : DHCP Address
IP Address              : 10.0.2.15
Subnet Mask             : 255.255.255.0
MAC Address             : 36:91:09:fc:bd:1f
Default Gateway IP      : 10.0.2.2
Default Gateway MAC     : 00:00:00:00:00:00
802.1q VLAN ID          : Disabled
RMCP+ Cipher Suites     : 17
Cipher Suite Priv Max   : aaaaaaaaaaaaaaa
                        :     X=Cipher Suite Unused
                        :     c=CALLBACK
                        :     u=USER
                        :     o=OPERATOR
                        :     a=ADMIN
                        :     O=OEM
Bad Password Threshold  : Not Available
`

**Expected behavior**
The Bad Password Threshold must have set and user must be locked for failed login attempts for ipmi over lan.

**OpenBMC Information:**
- /etc/os-release:
ID=openbmc-phosphor
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="2.9.0-dev"
VERSION_ID=2.9.0-dev-303-g9eeaafbe7
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.9.0-dev"
BUILD_ID="2.9.0-dev"

**Additional context**
Account lock works when tried from UI or redfish login.

closed 2021-06-09 by stale[bot]

KillerDAN, 2023-01-16:
Bounce Reopen

dkyadav13, 2024-07-30:
Reopen