openbmc/openbmc #3675: Not able to build Image for portwell-neptune board

athkar123, 2020-07-11:
Not able to build an image for target hardware Neptune. Missing the following dependency in the meta-phosphor layer.  

> openbmc/meta-phosphor/recipes-phosphor/images/obmc-phosphor-image.bb RDEPENDS on or otherwise requires it)
> ERROR: Nothing RPROVIDES '**virtual-obmc-chassis-mgmt**' 
> NOTE: Runtime target 'virtual-obmc-chassis-mgmt' is unbuildable, removing...
> Missing or unbuildable dependency chain was: ['virtual-obmc-chassis-mgmt']
> ERROR: Required build target 'obmc-phosphor-image' has no buildable providers.
> Missing or unbuildable dependency chain was: ['obmc-phosphor-image', 'virtual-obmc-chassis-mgmt']

geissonator, 2020-07-14:
I don't think the portwell-neptune board has had much maintenance. I'm not sure it ever even built. If you're interested in getting it up and running then the layer will need something like https://github.com/openbmc/openbmc/blob/master/meta-intel/meta-common/recipes-intel/packagegroups/packagegroup-intel-apps.bb#L19 that defines where some of these needed virtual packages come from.

jjhatt, 2020-10-14:
I am not able to build the Portwell Neptune card either. It must at one time been built because Portwell will include an OpenBMC image with the purchase of the card. However, they made it clear that I am on my own attempting to reproduce this creation.
Would the facebook/openbmc branch be more likely to have a documented build for the Neptune card?
---jjh

closed 2021-07-08 by stale[bot]