openbmc/openbmc #3676: Error on ASPEED2400 using IPKVM："OF: reserved mem: node jpegbuffer compatible matching fail"

master882020, 2020-07-12:
Hello, I have the following problems when using the ipKVM function
Hello, When I was using the KVM feature on APSEED2400 I had the following problem.

When the kernel is started:

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.48-f77eed1-dirty-b36f9b5 (oe-user@oe-host) (gcc version 10.1.0 (GCC)) #1 Sun Jul 12 01:22:17 UTC 2020
[    0.000000] CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00093177
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] OF: fdt: Machine model: Palmetto BMC
[    0.000000] Memory policy: Data cache writeback
[    0.000000] OF: reserved mem: node jpegbuffer compatible matching fail
[    0.000000] OF: reserved mem: node framebuffer compatible matching fail
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 60960
[    0.000000] Kernel command line: console=ttyS4,115200n8 root=/dev/ram rw
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 232612K/245760K available (6144K kernel code, 390K rwdata, 1312K rodata, 1024K init, 146K bss, 13148K reserved, 0K cma-reserved)
[    0.000000] random: get_random_u32 called from cache_alloc_refill+0x424/0x974 with crng_init=0
[    0.000000] ftrace: allocating 20004 entries in 40 pages
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16

The kenerl print  OF: reserved mem: node jpegbuffer compatible matching fail.

 I added the following new in aspeed-bmc-opp-palmetto.dts :
            gfx_memory: framebuffer {
			size = <0x01000000>;	/* 16M */
			alignment = <0x01000000>;
			compatible = "shared-dma-pool";
			reusable;
		};

		video_engine_memory: jpegbuffer {
			size = <0x02000000>;	/* 32M */
			alignment = <0x01000000>;
			compatible = "shared-dma-pool";
			reusable;
		};
....
&gfx {
	status = "okay";
	//memory-region = <&gfx_memory>;
};

&pinctrl {
	aspeed,external-nodes = <&gfx &lhc>;
};
&video {
	status = "okay";
	memory-region = <&video_engine_memory>;
};

And I modified the aspeed-g4.dtsi file

I think the video device was not created because of memory allocation failure!

Where should I change to use ipKVM functionality on ASPEED2400?

mdmillerii, 2020-07-13:
The message comes from [drivers/of/of_reserved_mem.c line 265](https://github.com/openbmc/linux/blob/2b4829edfc1c225c717652153097470529d171db/drivers/of/of_reserved_mem.c#L265)

> And I modified the aspeed-g4.dtsi file

What did you apply there?  Normally you should not need to patch dtsi files downstream.

The original code development was on the 2500 but I remember someone talk about using it on 2400.

master882020, 2020-07-13:
Thank you！
The ipkvm functionality cannot be used on 2400 without modification.
 I have changed the kernel configuration order and other files, And now I can use the IPKVM functionality.

closed 2020-07-13 by master882020

gooaler, 2021-04-21:
> Thank you！
> The ipkvm functionality cannot be used on AST2400 without modification.
> I have changed the kernel configuration order and other files, And now I can use the IPKVM functionality.

I have the same problem as you. Can you tell me how to modify it?