openbmc/openbmc #3678: Unable to boot the kernel image with full flash size(FLASH_SIZE = 131072)

kiran441-11, 2020-07-20:
I am new to this OpenBMC and trying to set up qemu emulation using Romulus image.
As the inbuilt image doesn't have support for python, I am building with the source by enabling the python because my application requires python.

The overall build got failed with the following error.
### ERROR: obmc-phosphor-image-1.0-r0 do_generate_static: Image '/mnt/workspace/openbmc-2.8.0/build/tmp/work/romulus-openbmc-linux-gnueabi/obmc-phosphor-image/1.0-r0/deploy-obmc-phosphor-image-image-complete/obmc-phosphor-image-romulus.squashfs-xz' is too large!

I have tried two approaches to resolve the build error.
The first approach tried is disabling some other modules and another approach is increasing the flash size to maximum (i.e 131072).

In both cases, the build succeeds and when I am trying boot to the kernel with full flash size got the wrong image format error from U-Boot as below.
> qemu-system-arm: Aspeed iBT has no chardev backend
> 
> 
> U-Boot 2016.07 (Jul 15 2020 - 21:16:53 +0000)
> 
>        Watchdog enabled
> DRAM:  112 MiB
> Flash: 32 MiB
> *** Warning - bad CRC, using default environment
> 
> In:    serial
> Out:   serial
> Err:   serial
> Net:   aspeednic#0
> Error: aspeednic#0 address not set.
> 
> Hit any key to stop autoboot:  0
> Wrong Image Format for bootm command
> ERROR: can't get kernel image!
> ast#
> 

I am not sure any other settings need to modify for using full flash size. Could someone help us, if any other settings need to modify for using maximum flash size?

Below are setup details:
Update the FLASH_SIZE value under meta-phosphor\classes\image_types_phosphor.bbclass with 131072

qemu arguments:
> ./qemu-system-arm -m 1024 -M romulus-bmc -nographic -drive file=./obmc-phosphor-image-romulus.static.mtd,format=raw,if=mtd, -net nic

Thanks in advance,
Kiran

solonfan, 2020-08-05:
Hi,Kiran!

From your log,The detect Flash is still 32 MiB
1.Try to make BMC FLASH_SIZE ? ="**32768**",keep 32 MB if you can build pass with remove other useless module.

2.If meet build fail,modify th**e partion offsets first** according your build error.

Thanks,
SolonFan

usrinivasan2407, 2021-02-03:
I get the same error but not every single time: 
Watchdog enabled
DRAM:  496 MiB
Flash: 32 MiB
*** Warning - bad CRC, using default environment
And I have the correct flash size in my conf file:
FLASH_SIZE = "32768"

What does this mean?  Thanks,
Usha

solonfan, 2021-02-04:
Hi,@usrinivasan2407!

Maybe relate your **boot loader environment**.The location for your uboot enviroment variable.
Try to make size of boot loader **bigger**,make sure have **enough size** for restore the enviroment variable.

Thanks,
SolonFan

usrinivasan2407, 2021-02-04:
Sorry, this is my first time in this area.  Can you tell me or point me to how I can do what you suggest, i.e., make the size bigger and make sure I have enough room?  I also don't know how to check how much room I need.

ushasrinivasan-cornelisnetworks, 2021-08-23:
This was due to an error on my part. It can be closed.

closed 2023-09-25 by williamspatrick