openbmc/openbmc #3679: build s2600wf error:Nothing PROVIDES 'python' ....

master882020, 2020-07-25:
hello!
The overall build got failed with the following error.
export TEMPLATECONF=meta-intel/meta-s2600wf/conf/
. openbmc-env
bitbake obmc-phosphor-image

‘Loading cache: 100% |                                           | ETA:  --:--:--
Loaded 0 entries from dependency cache.
Parsing recipes: 100% |##########################################| Time: 0:02:42
Parsing of 2404 .bb files complete (0 cached, 2404 parsed). 3651 targets, 354 skipped, 0 masked, 0 errors.
WARNING: No bb files in default matched BBFILE_PATTERN_s2600wf ''
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing PROVIDES 'python' (but /home/openbmc/meta-phosphor/recipes-phosphor/flash/obmc-flash-bmc_git.bb, /home/openbmc/meta-phosphor/recipes-phosphor/flash/obmc-mgr-download_git.bb DEPENDS on or otherwise requires it). Close matches:
  python3
  python3-py
  python3-sh
NOTE: Runtime target 'obmc-flash-bmc' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['obmc-flash-bmc', 'python']
NOTE: Runtime target 'virtual-obmc-chassis-mgmt' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['virtual-obmc-chassis-mgmt', 'obmc-flash-bmc', 'python']
ERROR: Required build target 'obmc-phosphor-image' has no buildable providers.
Missing or unbuildable dependency chain was: ['obmc-phosphor-image', 'virtual-obmc-chassis-mgmt', 'obmc-flash-bmc', 'python']
’

mdmillerii, 2020-07-26:
You encountered a recipe for a program that still is written for Python 2 and was not converted to Python 3.   Python 2 had its final release of 2.7.18 this past April and the Yocto project removed all Python 2 code in its Dunfell release and OpenBMC has merged this.

In addition, many BMCs have limited storage space for the filesystem.  The python runtime (both version 2 and 3) takes a significant amount of space.  Many systems have been able to eliminate all Python code and hence the runtime by replacing said code with programs written in C++, ocasionally with helpers in shell.

Either find an alternative package for this function or update the code to python version 3 then change the dependency to python3.

You might also look at the mailing list archives for others who have tried to use the s2600wf platform sometimes there are solutions that have not been sent to the upstream OpenBMC project yet.

master882020, 2020-07-27:
@mdmillerii 
Thank you！

closed 2020-07-27 by master882020