openbmc/openbmc #3680: /var/cache/fontconfig slowly filling up read/write filesystem

geissonator, 2020-07-28:
I noticed on our CI systems that our /var/cache/fontconfig/ directory was slowly filling up over time to the point where CI health checks were starting to fail.

Some internet searching found this:
https://unix.stackexchange.com/questions/545750/fontconfig-issues

But no real solid answers.  I tried a "fc-cache -r -v" which indicated success but did not actually clear any of the cache files.

```
root@witherspoon:/var/cache# fc-cache -r -v
/usr/share/fonts: caching, new cache contents: 0 fonts, 1 dirs
/usr/share/fonts/ttf: caching, new cache contents: 1 fonts, 0 dirs
/home/root/.local/share/fonts: skipping, no such directory
/home/root/.fonts: skipping, no such directory
/usr/share/fonts/ttf: skipping, looped directory detected
/var/cache/fontconfig: cleaning cache directory
/home/root/.cache/fontconfig: not cleaning non-existent cache directory
/home/root/.fontconfig: not cleaning non-existent cache directory
fc-cache: succeeded
#
root@witherspoon:/var/cache# ls /var/cache/fontconfig/
0038c192-83af-457b-adf4-4124cb2b6092-le32d8.cache-7  818eb5ab-ddc1-499d-ae76-db7f51adaef7-le32d8.cache-7
003e4c6f-eff8-4561-b29c-03611d30d362-le32d8.cache-7  81917042-8f2b-47cd-b764-de087d42427b-le32d8.cache-7
0040d067-2552-443d-8ee9-03cbcd4ceae5-le32d8.cache-7  81c26b50-4f82-44af-a4f6-939bcbc19ca6-le32d8.cache-7
00bad75b-d706-4229-b2ce-4c1389e3c848-le32d8.cache-7  81c9a4f1-8664-4371-8b50-731726be9a27-le32d8.cache-7

````

It was taking over 8MB uncompressed
```
root@witherspoon:/var/cache/fontconfig# du -sh
8.0M	.
#
root@witherspoon:/var/cache/fontconfig# df
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/root                18816     18816         0 100% /
devtmpfs                180892         0    180892   0% /dev
ubi0:rwfs                 4644      3560       824  81% /var
```

I did an "rm /var/cache/fontconfig/*" on the system having issues but this is a concerning issue for systems running for a long time.

I'm hoping this is more of a CI issue due to the fact it's running so many different levels of code on a daily basis (i.e. cache building up over time) but it does seem like we may want to consider some sort of cron that clears up our cache directories. Or maybe we log a bug on fontconfig (https://gitlab.freedesktop.org/fontconfig/fontconfig/-/issues)?

mdmillerii, 2020-07-29:
https://gitlab.freedesktop.org/fontconfig/fontconfig/-/issues/145

geissonator, 2021-06-09:
Looking at the CI system I cleared all of this from last time, we still have the issue. It's back up from a few KB to 6MB again.
```
root@witherspoon-Y230UF71K03T:/var/cache/fontconfig# du -sh
6.0M
```

I'm going to manually clear it again.

geissonator, 2021-10-21:
Still happening:
```
root@witherspoon-Y230UF71K03T:/var/cache/fontconfig# du -sh
7.8M	.
```

geissonator, 2022-04-28:
Still happening with 2.12.0-dev-1029

williamspatrick, 2023-09-25:
@geissonator How should we resolve this?  I suspect this is only an issue for people using the kvm support, right?

geissonator, 2023-09-25:
Yeah, I'm not seeing this issue anymore on our witherspoon (but we've removed the webui and kvm from the image to save flash space). Also, our p10bmc images do not include kvm and is not seeing this issue. So I no longer have an issue here on my systems, will close.

closed 2023-09-25 by geissonator