openbmc/openbmc #3681: How to use SPI1 ?

AlooGS, 2020-07-29:
Where is set 0x2 on barreleye and palmetto ? The FMC is used for BMC, so I want to use the SMC on BMC system to update the host BIOS, but set SCU70[13:12] = 1 on uboot flash_init(), uboot will hang :-(

AlooGS, 2020-07-29:
[image]
 

I want to use th blue.

solonfan, 2020-08-05:
Hi,AlooGS!

1.Check the HW schematic first,which type are your board is using.
[image]

2. Switching to Blue(Master SPI mode) from Red( SPI bypass mode) should later than flash_init,BMC need init&booting finished at bypass mode.

3.You can make BMC run into uboot cmdline mode,there is memory etc cmd to r/w register for test your purpose.

4.Or run into kernel.modify the flash device file-/dev/host_spi_flash0.Modify it to MASTER_MODE.Then acess the mtd device to flash your BIOS image.

Thanks,
SolonFan

closed 2021-07-08 by stale[bot]