openbmc/openbmc #3682: How to build the openbmc firmware image that can running with the ast2500 evb board?

whitehu81, 2020-07-29:
**Unexpected behavior you saw**
I try to setup the openbmc demo with AST2500 EVB to demonstrate the possibility of the openbmc with AST2500 Soc in team.
I assume the openbmc project officially support the AST2500 EVB, it should be very easy to build the image that can run with the EVB.

I think the *.mtd is the image that can be download to the EVB SPI flash by dedi-prog, please correct me if I am wrong.
But I don’t see the *.mtd file in the tmp/deploy/images/evb-ast2500/ after build complete, not sure something wrong or NOT in my local build
I am very new for the openbmc stack and build environment, do I need to change the configuration file then build the image after pull the code from the github?

Do you have any guideline doc to help me build and run the openbmc firmware with the AST-2500 EVB?

**To Reproduce**
Steps to reproduce the behavior:
 1. export TEMPLATECONF=meta-evb/meta-evb-aspeed/meta-evb-ast2500/conf
 2.  . openbmc-env
 3. bitbake obmc-phosphor-image

**Screenshots**
huw8@whitehu-virtual-machine:~/openbmc/openbmc$ . openbmc-env
### Initializing OE build env ###
Common targets are:
     obmc-phosphor-image
huw8@whitehu-virtual-machine:~/openbmc/openbmc/build$ bitbake obmc-phosphor-image
WARNING: Layer evb should set LAYERSERIES_COMPAT_evb in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer evb-aspeed should set LAYERSERIES_COMPAT_evb-aspeed in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer evb-ast2500 should set LAYERSERIES_COMPAT_evb-ast2500 in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer evb should set LAYERSERIES_COMPAT_evb in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer evb-aspeed should set LAYERSERIES_COMPAT_evb-aspeed in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer evb-ast2500 should set LAYERSERIES_COMPAT_evb-ast2500 in its conf/layer.conf file to list the core layer names it is compatible with.
Loading cache: 100% |############################################| Time: 0:00:00
Loaded 3500 entries from dependency cache.
Parsing recipes: 100% |##########################################| Time: 0:00:00
Parsing of 2318 .bb files complete (2317 cached, 1 parsed). 3501 targets, 351 skipped, 0 masked, 0 errors.
WARNING: No bb files in default matched BBFILE_PATTERN_evb ''
WARNING: No bb files in default matched BBFILE_PATTERN_evb-aspeed ''
WARNING: No bb files in default matched BBFILE_PATTERN_evb-ast2500 ''
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.47.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-16.04"
TARGET_SYS           = "arm-openbmc-linux-gnueabi"
MACHINE              = "evb-ast2500"
DISTRO               = "openbmc-phosphor"
DISTRO_VERSION       = "0.1.0"
TUNE_FEATURES        = "arm thumb arm1176jzs"
TARGET_FPU           = "soft"
meta
meta-poky
meta-oe
meta-networking
meta-python
meta-webserver
meta-phosphor
meta-aspeed
meta-evb
meta-evb-aspeed
meta-evb-ast2500     = "master:d221e0389dc9bafd72e541090a9c401a25a84ce3"

Initialising tasks: 100% |#######################################| Time: 0:00:02
Sstate summary: Wanted 0 Found 0 Missed 0 Current 1295 (0% match, 100% complete)
NOTE: Executing Tasks
NOTE: Tasks Summary: Attempted 3792 tasks of which 3792 didn't need to be rerun and all succeeded.

Summary: There were 9 WARNING messages shown.
huw8@whitehu-virtual-machine:~/openbmc/openbmc/build$ ls
bitbake-cookerdaemon.log  conf       evb-ast2500  sstate-cache
cache                     downloads  palmetto     tmp
huw8@whitehu-virtual-machine:~/openbmc/openbmc/build$ ls tmp/deploy/images/evb-ast2500/
aspeed-ast2500-evb--5.4.48+git0+f77eed1af7-r0-evb-ast2500-20200720041220.dtb
aspeed-ast2500-evb.dtb
aspeed-ast2500-evb-evb-ast2500.dtb
aspeed-image-initramfs-evb-ast2500-20200720041220.rootfs.cpio.xz
aspeed-image-initramfs-evb-ast2500-20200720041220.rootfs.manifest
aspeed-image-initramfs-evb-ast2500-20200720041220.testdata.json
aspeed-image-initramfs-evb-ast2500.cpio.xz
aspeed-image-initramfs-evb-ast2500.manifest
aspeed-image-initramfs-evb-ast2500.testdata.json
fitImage
fitImage--5.4.48+git0+f77eed1af7-r0-evb-ast2500-20200720041220.bin
fitImage-aspeed-image-initramfs-evb-ast2500--5.4.48+git0+f77eed1af7-r0-evb-ast2500-20200720041220.bin
fitImage-aspeed-image-initramfs-evb-ast2500-evb-ast2500
fitImage-evb-ast2500.bin
fitImage-its--5.4.48+git0+f77eed1af7-r0-evb-ast2500-20200720041220.its
fitImage-its-aspeed-image-initramfs-evb-ast2500--5.4.48+git0+f77eed1af7-r0-evb-ast2500-20200720041220.its
fitImage-its-aspeed-image-initramfs-evb-ast2500-evb-ast2500
fitImage-its-evb-ast2500
fitImage-linux.bin--5.4.48+git0+f77eed1af7-r0-evb-ast2500-20200720041220.bin
fitImage-linux.bin-evb-ast2500
obmc-phosphor-image-evb-ast2500-20200720041220.rootfs.manifest
obmc-phosphor-image-evb-ast2500-20200720041220.rootfs.tar.gz
obmc-phosphor-image-evb-ast2500-20200720041220.testdata.json
obmc-phosphor-image-evb-ast2500.manifest
obmc-phosphor-image-evb-ast2500.tar.gz
obmc-phosphor-image-evb-ast2500.testdata.json

g6949, 2020-09-05:
I also want to resolve this problem, but I still don't understand the reason.
Does anyone know the reason?

Attached is a diff of bitbake -v obmc-phosphor-image from 2.8.0-dev and 2.8.0-rc1 around the do_image.
[ast2500.log](https://github.com/openbmc/openbmc/files/5177752/ast2500.log)

mdmillerii, 2020-09-06:
Looks like you are getting the code update image pckaged but not the mtd file.  As q work around you can try unpacking the `obmc-phosphor-image-evb-ast2500.tar.gz` file to see if you it contains the needed image (it will also contain some additional meta data files). Hope this helps keeping you progressing over the long weekend for many.

mine260309, 2020-09-06:
Please see the `README` of https://github.com/openbmc/openbmc for `evb-ast2500`, there is special note:
>For evb-ast2500, please use the below command to specify the machine config, because the machine in meta-aspeed layer is in a BSP layer and does not build the openbmc image.
>
> ```TEMPLATECONF=meta-evb/meta-evb-aspeed/meta-evb-ast2500/conf . openbmc-env```

With the above config, you are able to build the mtd image.

g6949, 2020-09-06:
Thank you for your comment, @mine260309 @mdmillerii 

I've already specified TEMPATECONF.
Here is my command history.
```
  git clone https://github.com/openbmc/openbmc.git openbmc-2.8.0-rc1 -b 2.8.0-rc1 --depth 1
  cd openbmc-2.8.0-rc1/
  export TEMPLATECONF=meta-evb/meta-evb-aspeed/meta-evb-ast2500/conf
  TEMPLATECONF=meta-evb/meta-evb-aspeed/meta-evb-ast2500/conf . openbmc-env
  bitbake -v obmc-phosphor-image
```
I've tried once more. I still can't get the mtd image.
```
$ ll tmp/deploy/images/evb-ast2500/*.mtd
ls: cannot access 'tmp/deploy/images/evb-ast2500/*.mtd': No such file or directory
$ tar tvzf tmp/deploy/images/evb-ast2500/obmc-phosphor-image-evb-ast2500.tar.gz | grep mtd
-rw-r--r-- 0/0             174 2020-09-06 17:40 ./lib/udev/rules.d/75-probe_mtd.rules
-rw-r--r-- 0/0              45 2020-09-06 18:03 ./lib/udev/rules.d/76-aspeed-mtd-partitions.rules
-rwxr-xr-x 0/0           19584 2020-09-06 18:10 ./lib/udev/mtd_probe
$
```
In case of 2.8.0-dev, I can get mtd image with the same procedure.
```
$ ls -1 tmp/deploy/images/evb-ast2500/*.mtd
tmp/deploy/images/evb-ast2500/obmc-phosphor-image-evb-ast2500-20200905021036.static.mtd
tmp/deploy/images/evb-ast2500/obmc-phosphor-image-evb-ast2500.static.mtd
$ ls -1 tmp/deploy/images/evb-ast2500/image-bmc
tmp/deploy/images/evb-ast2500/image-bmc
$
```
I am looking forward more information.
Thanks.

mine260309, 2020-09-07:
There was known issue on evb-ast2500 build and it was fixed in 185ce9eb0, 82583ec0d, bfe22e8c4 and 56f86d23c
Please try a revision after the fixes, e.g. use the latest master.

g6949, 2020-09-07:
Hi, @mine260309 

I've got mtd image by https://github.com/openbmc/openbmc/commit/185ce9eb054c2c8b43cd169dc34ec53aaf62ce0e finally.

Thank you for your help!

closed 2021-07-08 by stale[bot]