openbmc/openbmc #3683: w83795 can not display sensor value in web

229024527, 2020-07-29:
Hello everyone when i use w83795 as a sensor detect，But nothing display in my web, I can get value under /sys/class/hwmon/hwmon0/device/

1.Set ip address 192.168.5.15 by uart 
2. use firefox open https://192.168.5.15/#/server-health/sensors-overview

root@palmetto:/sys/class/hwmon/hwmon0/device# cat /etc/issue
Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 \n \l

root@palmetto:/sys/class/hwmon/hwmon0/device# uname -a
Linux palmetto 5.4.28 #1 Tue Jul 28 05:45:17 UTC 2020 armv5tejl GNU/Linux

my dts
&i2c0 {
        status = "okay";

        w83795adg@2c {
                compatible = "nuvoton,w83795adg";
                reg = <0x2c>;
        };
}

phosphor-hwmon_%.bbappend

FILESEXTRAPATHS_prepend_palmetto := "${THISDIR}/${PN}:"

EXTRA_OECONF_append_palmetto = " --enable-negative-errno-on-fail"

CHIPS = " \
        bus@1e78a000/i2c-bus@40/w83795adg@2c \
        "
ITEMSFMT = "ahb/apb/{0}.conf"

ITEMS = "${@compose_list(d, 'ITEMSFMT', 'CHIPS')}"

#OCCSFMT = "device/w83795adg@2c.conf"
#OCCITEMS = "${@compose_list(d, 'OCCSFMT', 'OCCS')}"

ENVS = "obmc/hwmon/{0}"
SYSTEMD_ENVIRONMENT_FILE_${PN}_append_palmetto = " ${@compose_list(d, 'ENVS', 'ITEMS')}"
#SYSTEMD_ENVIRONMENT_FILE_${PN}_append_palmetto = " ${@compose_list(d, 'ENVS', 'OCCITEMS')}"

openbmc@z:~$ cat openbmc/meta-ibm/meta-palmetto/recipes-phosphor/sensors/phosphor-hwmon/obmc/hwmon/ahb/apb/bus@1e78a000/i2c-bus@40/w83795adg@2c.conf 
LABEL_temp1="CPU"
WARNLO_temp1=10000
WARNHI_temp1=40000

LABEL_temp2="SYSTEM"
WARNLO_temp2=10000
WARNHI_temp2=80000

LABEL_temp6="7A"
WARNLO_temp6=10000
WARNHI_temp6=80000

LABEL_fan1 = "CPU0"
PWM_TARGET_fan1 = "1"

LABEL_fan2 = "CPU2"
PWM_TARGET_fan2 = "2"

LABEL_fan3 = "SYSTEM1"
PWM_TARGET_fan3 = "3"

LABEL_fan4 = "SYSTEM2"
PWM_TARGET_fan4 = "4"

LABEL_fan5 = "SYSTEM11"
PWM_TARGET_fan5 = "5"

LABEL_fan6 = "SYSTEM4"
PWM_TARGET_fan6 = "6"

LABEL_fan7 = "CPU1"
PWM_TARGET_fan7 = "7"

LABEL_fan8 = "CPU3"
PWM_TARGET_fan8 = "8"                                                                                         

root@palmetto:/sys/class/hwmon/hwmon0/device# cat fan1_input
3739

Tommy-git2033, 2021-03-23:
Hello, it seems that i encouter almost the same issue, please, have you solve it ?

Tommy-git2033, 2021-03-23:
Hello, Have you find how to solve this problem?

229024527, 2021-04-27:
> Hello, Have you find how to solve this problem?

This problem sould connet Nuvoton to update w83795 driver

williamspatrick, 2024-12-20:
Missing information on if the sensor object is created.  You need an association to the chassis in order for it to show up in the webui.

Most people are using entity-manager / dbus-sensors now for proper functionality.

closed 2024-12-20 by williamspatrick