openbmc/openbmc #3684: hwclock cannot set time

shenki, 2020-08-07:
On one of our systems running openbmc master we noticed hwclock could no longer set the time:

```
root@rain15bmc:~# timedatectl set-time '2020-08-04 16:14:50'
root@rain15bmc:~# date
Tue Aug  4 16:14:51 UTC 2020
root@rain15bmc:~# hwclock -w
root@rain15bmc:~# hwclock -s
hwclock: settimeofday: Invalid argument
```

This was reported to the busybox mailing list last month:

  http://lists.busybox.net/pipermail/busybox/2020-July/088094.html

The root cause is a change in the glibc for the gettimeofday/settimeofday system call wrappers:

 https://www.mail-archive.com/info-gnu@gnu.org/msg02694.html

> However, to reduce the odds of this offset being set by accident,
> settimeofday can no longer be used to set the time and the offset
> simultaneously.  If both of its two arguments are non-null, the call
> will fail (setting errno to EINVAL).
>
> settimeofday itself is obsolescent according to POSIX.  Programs
> that set the system time should use clock_settime and/or the adjtime
> family of functions instead.  We may cease to make settimeofday
> available to newly linked binaries after there is a replacement for
> Linux's time-zone-like offset API.

This issue being the root cause was confirmed by using the util-linux hwclock binary from Debian unstable:

```
root@rain15bmc:~# LD_LIBRARY_PATH=$PWD ./hwclock -r
2020-08-04 17:43:59.282924+00:00
root@rain15bmc:~# LD_LIBRARY_PATH=$PWD ./hwclock -s
root@rain15bmc:~# LD_LIBRARY_PATH=$PWD ./hwclock -r
2020-08-04 17:44:12.448413+00:00
```

The util-linux change is https://github.com/karelzak/util-linux/commit/cd781c405be82540484da3bfe3d3f17a39b8eb5c and https://github.com/karelzak/util-linux/commit/9c6139a72017cecb9145e46102152cb4f456ada6. The second commit fixes it by using the system call directly, without the glibc wrapper. This is probably the approach that needs to be taken by busybox in order to remain portable.

eddiejames, 2020-08-10:
http://lists.busybox.net/pipermail/busybox/2020-August/088187.html

vishwabmc, 2020-08-12:
> http://lists.busybox.net/pipermail/busybox/2020-August/088187.html

@eddiejames Are you planning on taking that and putting in Openbmc ?

bjwyman, 2020-09-14:
Is this the fix: https://github.com/openbmc/openbmc/blob/748a483a8f515f7aa0ce999ebeeebed4ed17ae10/poky/meta/recipes-core/busybox/busybox/0001-hwclock-make-glibc-2.31-compatible.patch ?

mdmillerii, 2020-09-14:
@bjwyman yes it appears so

geissonator, 2020-10-20:
This was pulled into openbmc via https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/35078

busybox has not tagged a release with the fix yet but the patch was put into upstream yocto so we picked it up.

closed 2020-10-20 by geissonator