openbmc/openbmc #3685: UBI failing on 5.8 kernel

shenki, 2020-08-11:
When attempting to boot a witherspoon system with dev-5.8, the UBI filesystem fails to mount.

Commonly we were seeing this error:
```
[    3.404365] ubi0: scanning is finished
[    3.408365] ubi0 error: ubi_read_volume_table: the layout volume was not found
```

However, on a Witherspoon that was running op940.00-16 (Linux 5.3.1-7e560ad) I saw the following:
```
[    4.520873] ubi0: default fastmap pool size: 25
[    4.525431] ubi0: default fastmap WL pool size: 12
[    4.530430] ubi0: attaching mtd3
[    4.545028] ubi0: scanning is finished
[    4.559527] ubi0: attached mtd3 (name "obmc-ubi", size 31 MiB)
[    4.565388] ubi0: PEB size: 65536 bytes (64 KiB), LEB size: 65408 bytes
[    4.572155] ubi0: min./max. I/O unit sizes: 1/256, sub-page size 1
[    4.578469] ubi0: VID header offset: 64 (aligned 64), data offset: 128
[    4.585013] ubi0: good PEBs: 504, bad PEBs: 0, corrupted PEBs: 0
[    4.591137] ubi0: user volume: 4, internal volumes: 1, max. volumes count: 128
[    4.598487] ubi0: max/mean erase counter: 634/157, WL threshold: 4096, image sequence number: 725656936
[    4.608002] ubi0: available PEBs: 34, total reserved PEBs: 470, PEBs reserved for bad PEB handling: 0
[    4.617414] ubi0: background thread "ubi_bgt0d" started, PID 46
[    4.623435] ubi4: default fastmap pool size: 25
[    4.628109] ubi4: default fastmap WL pool size: 12
[    4.632921] ubi4: attaching mtd7
[    4.641992] ubi4 error: ubi_add_to_av: two LEBs with same sequence number 33154
[    4.649463] eraseblock attaching information dump:
[    4.654271] 	ec       1
[    4.656844] 	pnum     15
[    4.659391] 	lnum     33
[    4.661926] 	scrub    0
[    4.664371] 	sqnum    33154
[    4.667266] Volume identifier header dump:
[    4.671381] 	magic     55424921
[    4.674531] 	version   1
[    4.677176] 	vol_type  2
[    4.679724] 	copy_flag 0
[    4.682259] 	compat    0
[    4.684788] 	vol_id    1
[    4.687429] 	lnum      33
[    4.690062] 	data_size 65408
[    4.692946] 	used_ebs  43
[    4.695570] 	data_pad  0
[    4.698210] 	sqnum     33154
[    4.701105] 	hdr_crc   71404ef8
[    4.704246] Volume identifier header hexdump:
[    4.708959] ubi4 error: ubi_attach_mtd_dev: failed to attach mtd7, error -22
[    4.716216] UBI error: cannot attach mtd7
[    4.720259] UBI: block: can't open volume on ubi4_0, err=-19
[    4.726142] printk: console [netcon0] enabled
[    4.730524] netconsole: network logging started
[    4.737876] input: gpio-keys as /devices/platform/gpio-keys/input/input2
[    4.747955] /dev/root: Can't open blockdev
[    4.752104] VFS: Cannot open root device "ubiblock4_0" or unknown-block(0,0): error -6
```

This does not reproduce in qemu, however it needs to be confirmed that someone has done a test where the old kernel read and wrote the filesystem, and then the new kernel was used.

shenki, 2020-08-11:
dev-5.7 works fine on the same machine. The changes between these releases for ubifs and ubi are:

4b68bf9a69d2 ubi: Select fastmap anchor PEBs considering wear level rules
66aaba3a07cb ubi: Relax the 'no MLC' rule and allow MLCs operating in SLC mode
f80df3851246 ubifs: use crypto_shash_tfm_digest()

There are some other changes related to memory management, but it is unlikely they are the cause:

ca5999fde0a1 mm: introduce include/linux/pgtable.h
88dca4ca5a93 mm: remove the pgprot argument to __vmalloc
88dca4ca5a93 mm: remove the pgprot argument to __vmalloc

shenki, 2020-08-11:
I tried reverting 4b68bf9a69d2, and then tried reverting all three ubi/ubifs patches, and dev-5.8 would not mount the filesystem on my test machine.

shenki, 2020-08-11:
Reverting the three commits on top of dev-5.8 failed to mount:
```
[    4.512680] ubi4: default fastmap pool size: 25
[    4.517233] ubi4: default fastmap WL pool size: 12
[    4.522155] ubi4: attaching mtd7
[    4.531352] ubi4 error: ubi_add_to_av: two LEBs with same sequence number 33154
```

Notably mtd7 is the second flash chip. This is the system's command line:
```
[    0.000000] Kernel command line: console=ttyS4,115200n8 ubi.mtd=obmc-ubi,0,0,0 ubi.mtd=alt-obmc-ubi,0,0,4 ro rootfstype=squashfs ubi.block=4,0 root=/dev/ubiblock4_0
```

shenki, 2020-08-11:
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.8.0-00137-ge1ef3f980349 (joel@voyager) (arm-linux-gnueabi-gcc (Debian 9.3.0-13) 9.3.0, GNU ld (GNU Binutils for Debian) 2.35) #131 SMP Tue Aug 11 21:49:14 ACST 2020
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Witherspoon BMC
[    0.000000] Memory policy: Data cache writeback
[    0.000000] Reserved memory: created CMA memory pool at 0x9c000000, size 32 MiB
[    0.000000] OF: reserved mem: initialized node jpegbuffer, compatible id shared-dma-pool
[    0.000000] Reserved memory: created CMA memory pool at 0x97000000, size 16 MiB
[    0.000000] OF: reserved mem: initialized node framebuffer, compatible id shared-dma-pool
[    0.000000] Reserved memory: created DMA memory pool at 0x9f000000, size 16 MiB
[    0.000000] OF: reserved mem: initialized node region@9f000000, compatible id shared-dma-pool
[    0.000000] cma: Reserved 16 MiB at 0x96000000
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000080000000-0x000000009effffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080000000-0x0000000097ffffff]
[    0.000000]   node   0: [mem 0x000000009c000000-0x000000009effffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080000000-0x000000009effffff]
[    0.000000] CPU: All CPU(s) started in SVC mode.
[    0.000000] percpu: Embedded 18 pages/cpu s50444 r0 d23284 u73728
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 109600
[    0.000000] Kernel command line: console=ttyS4,115200n8 ubi.mtd=obmc-ubi,0,0,0 ubi.mtd=alt-obmc-ubi,0,0,4 ro rootfstype=squashfs ubi.block=4,0 root=/dev/ubiblock4_0
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 360572K/442368K available (8192K kernel code, 566K rwdata, 1832K rodata, 1024K init, 178K bss, 16260K reserved, 65536K cma-reserved, 0K highmem)
[    0.000000] random: get_random_u32 called from cache_alloc_refill+0x434/0x9b0 with crng_init=0
[    0.000000] ftrace: allocating 26723 entries in 53 pages
[    0.000000] ftrace: allocated 53 pages with 4 groups
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=2 to nr_cpu_ids=1.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=1
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] i2c controller registered, irq 17
[    0.000000] clocksource: FTTMR010-TIMER2: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 77222644334 ns
[    0.000019] sched_clock: 32 bits at 24MHz, resolution 40ns, wraps every 86767015915ns
[    0.000109] Switching to timer-based delay loop, resolution 40ns
[    0.001943] Calibrating delay loop (skipped), value calculated using timer frequency.. 49.50 BogoMIPS (lpj=247500)
[    0.002000] pid_max: default: 32768 minimum: 301
[    0.002792] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.002847] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.004903] CPU: Testing write buffer coherency: ok
[    0.007964] Setting up static identity map for 0x80100000 - 0x80100078
[    0.008489] rcu: Hierarchical SRCU implementation.
[    0.013306] smp: Bringing up secondary CPUs ...
[    0.013350] smp: Brought up 1 node, 1 CPU
[    0.013372] SMP: Total of 1 processors activated (49.50 BogoMIPS).
[    0.013386] CPU: All CPU(s) started in SVC mode.
[    0.014611] devtmpfs: initialized
[    0.039035] VFP support v0.3: not present
[    0.040311] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.040384] futex hash table entries: 256 (order: 2, 16384 bytes, linear)
[    0.052148] pinctrl core: initialized pinctrl subsystem
[    0.053457] NET: Registered protocol family 16
[    0.058230] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.061143] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.061180] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.141853] usbcore: registered new interface driver usbfs
[    0.142016] usbcore: registered new interface driver hub
[    0.142252] usbcore: registered new device driver usb
[    0.142504] mc: Linux media interface: v0.10
[    0.142654] videodev: Linux video capture interface: v2.00
[    0.142827] EDAC MC: Ver: 3.0.0
[... 167 lines cut ...]
[    4.741096] [<8010db58>] (dump_backtrace) from [<8010ddb4>] (show_stack+0x20/0x24)
[    4.748669]  r7:80a0dd94 r6:60000093 r5:00000000 r4:80c6b694
[    4.754347] [<8010dd94>] (show_stack) from [<808b6498>] (dump_stack+0xcc/0xe8)
[    4.761590] [<808b63cc>] (dump_stack) from [<80122cdc>] (panic+0x110/0x344)
[    4.768556]  r7:80a0dd94 r6:80c0df70 r5:00000000 r4:80c8e388
[    4.774247] [<80122bcc>] (panic) from [<80b01b00>] (mount_block_root+0x2b4/0x358)
[... cut ...]
[    4.840370]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:808cf54c
[    4.848196]  r4:00000000
[    4.850747] [<808cf54c>] (kernel_init) from [<80100168>] (ret_from_fork+0x14/0x2c)
[    4.858311] Exception stack(0x9e4b3fb0 to 0x9e4b3ff8)
[    4.863369] 3fa0:                                     00000000 00000000 00000000 00000000
[    4.871542] 3fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[    4.879712] 3fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[    4.886323]  r5:808cf54c r4:00000000
```

shenki, 2020-08-11:
Fixing the system to boot from the first chipselect, the dev-5.8 tree (with reverts applied) was able to find and mount the ubi volume, but then hit errors. It produced pages of oops similar to the one below:

```
[    6.544168] block ubiblock0_0: created from ubi0:0(rofs-730944ed)
[    6.576761] VFS: Mounted root (squashfs filesystem) readonly on device 253:0.
[    6.589041] devtmpfs: mounted
[    6.597965] Freeing unused kernel memory: 1024K
[    6.620711] Checked W+X mappings: passed, no W+X pages found
[    6.626431] Run /sbin/init as init process
[    7.967805] UBIFS (ubi0:2): Mounting in unauthenticated mode
[    7.980593] UBIFS (ubi0:2): background thread "ubifs_bgt0_2" started, PID 49
[    8.009996] UBIFS (ubi0:2): UBIFS: mounted UBI device 0, volume 2, name "rwfs"
[    8.017254] UBIFS (ubi0:2): LEB size: 65408 bytes (63 KiB), min./max. I/O unit sizes: 8 bytes/256 bytes
[    8.026876] UBIFS (ubi0:2): FS size: 5690496 bytes (5 MiB, 87 LEBs), journal size 981120 bytes (0 MiB, 15 LEBs)
[    8.037106] UBIFS (ubi0:2): reserved for root: 0 bytes (0 KiB)
[    8.043080] UBIFS (ubi0:2): media format: w4/r0 (latest is w5/r0), UUID 7CC6FFB3-5D94-4D02-BCA5-AA8F1047475A, small LPT model
[    8.451346] UBIFS error (ubi0:2 pid 53): ubifs_read_node: bad node type (255 but expected 2)
[    8.460047] UBIFS error (ubi0:2 pid 53): ubifs_read_node: bad node at LEB 58:64856, LEB mapping status 1
[    8.469672] Not a node, first 24 bytes:
[    8.469704] 00000000: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff                          ........................
[    8.486543] CPU: 0 PID: 53 Comm: mount Not tainted 5.8.0-00137-ge1ef3f980349 #131
[    8.494036] Hardware name: Generic DT based system
[    8.498833] Backtrace: 
[    8.501325] [<8010db58>] (dump_backtrace) from [<8010ddb4>] (show_stack+0x20/0x24)
[    8.508904]  r7:0000003d r6:60000013 r5:00000000 r4:80c6b694
[    8.514584] [<8010dd94>] (show_stack) from [<808b6498>] (dump_stack+0xcc/0xe8)
[    8.521815] [<808b63cc>] (dump_stack) from [<80402268>] (ubifs_read_node+0x284/0x2c8)
[    8.529649]  r7:0000003d r6:0000fd58 r5:00000000 r4:95a07000
[    8.535311] [<80401fe4>] (ubifs_read_node) from [<80402474>] (ubifs_read_node_wbuf+0x1c8/0x314)
[    8.544011]  r10:95a07000 r9:0000003d r8:00000002 r7:0000003a r6:959ebcc0 r5:0000fd58
[    8.551838]  r4:95a01090
[    8.554406] [<804022ac>] (ubifs_read_node_wbuf) from [<80423a98>] (ubifs_tnc_read_node+0x70/0x1e4)
[    8.563367]  r10:00000002 r9:95a070d4 r8:95a07000 r7:95a24000 r6:00000002 r5:959ebcc0
[    8.571189]  r4:95a6bc40
[    8.573736] [<80423a28>] (ubifs_tnc_read_node) from [<80403db4>] (tnc_read_hashed_node+0xa8/0x1c0)
[    8.582691]  r8:959ebcc0 r7:95a07000 r6:959ebcc0 r5:95a25cb8 r4:95a6bc40
[    8.589400] [<80403d0c>] (tnc_read_hashed_node) from [<80407c68>] (ubifs_tnc_next_ent+0x244/0x284)
[    8.598356]  r7:95a24000 r6:95a07000 r5:95a25cb8 r4:00000000
[    8.604036] [<80407a24>] (ubifs_tnc_next_ent) from [<803f7dc0>] (ubifs_readdir+0x4a8/0x540)
[    8.612394]  r10:959ebc00 r9:9e343298 r8:00000000 r7:9e343218 r6:959f1c00 r5:95a25d30
[    8.620217]  r4:9e342440
[    8.622774] [<803f7918>] (ubifs_readdir) from [<802d5178>] (iterate_dir+0x88/0x178)
[    8.630437]  r10:959ebc00 r9:9e343298 r8:95a25d30 r7:00000001 r6:959f1c08 r5:00000000
[    8.638262]  r4:959f1c00
[    8.640810] [<802d50f0>] (iterate_dir) from [<80438cbc>] (ovl_check_d_type_supported+0xac/0x124)
[    8.649599]  r10:959ebc00 r9:00000002 r8:00000000 r7:95a25dec r6:95a24000 r5:959f1c00
[    8.657423]  r4:95925580
[    8.659971] [<80438c10>] (ovl_check_d_type_supported) from [<8042c854>] (ovl_fill_super+0x594/0x19dc)
[    8.669193]  r8:95a6c550 r7:95a25e10 r6:95a47000 r5:00000000 r4:95925580
[    8.675903] [<8042c2c0>] (ovl_fill_super) from [<802c2c08>] (mount_nodev+0x54/0xa4)
[    8.683562]  r10:00000000 r9:95a18000 r8:00000020 r7:95a18000 r6:8042c2c0 r5:00000000
[    8.691389]  r4:95a47000
[    8.693936] [<802c2bb4>] (mount_nodev) from [<8042af98>] (ovl_mount+0x20/0x28)
[    8.701162]  r7:95925e80 r6:959eb300 r5:8042af78 r4:95925e80
[    8.706830] [<8042af78>] (ovl_mount) from [<802fcdf0>] (legacy_get_tree+0x34/0x5c)
[    8.714409] [<802fcdbc>] (legacy_get_tree) from [<802c2d90>] (vfs_get_tree+0x34/0xf4)
[    8.722236]  r5:95925e80 r4:00000000
[    8.725830] [<802c2d5c>] (vfs_get_tree) from [<802e860c>] (do_mount+0x760/0x97c)
[    8.733227]  r6:959eb300 r5:95a24000 r4:00000000
[    8.737852] [<802e7eac>] (do_mount) from [<802e8c1c>] (sys_mount+0x98/0xc4)
[    8.744820]  r10:00000015 r9:95a24000 r8:c0ed0000 r7:0008d3c0 r6:95a18000 r5:959eb300
[    8.752643]  r4:959eba40
[    8.755186] [<802e8b84>] (sys_mount) from [<80100080>] (ret_fast_syscall+0x0/0x54)
[    8.762750] Exception stack(0x95a25fa8 to 0x95a25ff0)
[    8.767811] 5fa0:                   0008d2f8 46e28000 0008d3b0 0008d3c0 0008d3a0 c0ed0000
[    8.775994] 5fc0: 0008d2f8 46e28000 46e28790 00000015 c0ed0000 0008d3c0 0008d3b0 0008d3a0
[    8.784170] 5fe0: 46e28680 7ee35bb0 46df14e8 46b96620
[    8.789227]  r8:80100284 r7:00000015 r6:46e28790 r5:46e28000 r4:0008d2f8
[    8.796156] UBIFS error (ubi0:2 pid 53): ubifs_readdir: cannot find next direntry, error -22
mount: /etc: wrong fs type, bad option, bad superblock on overlay, missing codepage or helper program, or other error.
```

The error had this in common:
```
[   29.213605] UBIFS error (ubi0:2 pid 95): ubifs_read_node: bad node type (255 but expected 2)
[   29.222248] UBIFS error (ubi0:2 pid 95): ubifs_read_node: bad node at LEB 65:416, LEB mapping status 1
[   29.231705] Not a node, first 24 bytes:
[   29.231736] 00000000: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  
```

```

shenki, 2020-08-11:
There were no changes of note in the spi-nor driver between versions:
```
$ git diff dev-5.7..dev-5.8 drivers/mtd/spi-nor/controllers/aspeed-smc.c
diff --git a/drivers/mtd/spi-nor/controllers/aspeed-smc.c b/drivers/mtd/spi-nor/controllers/aspeed-smc.c
index 8daf423d02de..a591ce1d24c5 100644
--- a/drivers/mtd/spi-nor/controllers/aspeed-smc.c
+++ b/drivers/mtd/spi-nor/controllers/aspeed-smc.c
@@ -1202,7 +1202,7 @@ static int aspeed_smc_chip_setup_finish(struct aspeed_smc_chip *chip)
 
        /*
         * TODO: Adjust clocks if fast read is supported and interpret
-        * SPI-NOR flags to adjust controller settings.
+        * SPI NOR flags to adjust controller settings.
         */
        io_mode = aspeed_smc_get_io_mode(chip);
        if (io_mode < 0)
```

shenki, 2020-08-11:
After the error above the flash chip was not able to be read by u-boot's ubi driver:
```
ast# ubi part obmc-ubi
ubi0: attaching mtd1
ubi0: scanning is finished
UBI init error 22
```

shenki, 2020-08-12:
test setup:
 - witherspoon with dual 32MB NOR
 - op940 UBI layout flashed to both
 - netbooting kernel with initrd userspace
 - aspeed_g5_defconfig
 - kernel will probe ubi devices and attach the volumes

good:
```
[    5.274571] ubi0: default fastmap pool size: 25
[    5.279337] ubi0: default fastmap WL pool size: 12
[    5.284151] ubi0: attaching mtd3
[    5.306238] ubi0: scanning is finished
[    5.323686] ubi0: attached mtd3 (name "obmc-ubi", size 31 MiB)
[    5.329693] ubi0: PEB size: 65536 bytes (64 KiB), LEB size: 65408 bytes
[    5.336325] ubi0: min./max. I/O unit sizes: 1/256, sub-page size 1
[    5.342625] ubi0: VID header offset: 64 (aligned 64), data offset: 128
[    5.349259] ubi0: good PEBs: 504, bad PEBs: 0, corrupted PEBs: 0
[    5.355286] ubi0: user volume: 3, internal volumes: 1, max. volumes count: 128
[    5.362626] ubi0: max/mean erase counter: 2/0, WL threshold: 4096, image sequence number: 2060944082
[    5.371878] ubi0: available PEBs: 70, total reserved PEBs: 434, PEBs reserved for bad PEB handling: 0
[    5.381300] ubi0: background thread "ubi_bgt0d" started, PID 45
[    5.387429] ubi4: default fastmap pool size: 25
[    5.391980] ubi4: default fastmap WL pool size: 12
[    5.396906] ubi4: attaching mtd7
[    5.453835] ubi4: scanning is finished
[    5.493428] ubi4: attached mtd7 (name "alt-obmc-ubi", size 31 MiB)
[    5.499776] ubi4: PEB size: 65536 bytes (64 KiB), LEB size: 65408 bytes
[    5.506408] ubi4: min./max. I/O unit sizes: 1/256, sub-page size 1
[    5.512712] ubi4: VID header offset: 64 (aligned 64), data offset: 128
[    5.519345] ubi4: good PEBs: 504, bad PEBs: 0, corrupted PEBs: 0
[    5.525371] ubi4: user volume: 4, internal volumes: 1, max. volumes count: 128
[    5.532708] ubi4: max/mean erase counter: 226/66, WL threshold: 4096, image sequence number: 725656936
[    5.542136] ubi4: available PEBs: 27, total reserved PEBs: 477, PEBs reserved for bad PEB handling: 0
[    5.551600] ubi4: background thread "ubi_bgt4d" started, PID 46
[    9.283129] block ubiblock0_1: created from ubi0:1(rofs-730944ed)
```
bad:
```
[    4.832449] ubi0: default fastmap pool size: 25
[    4.837006] ubi0: default fastmap WL pool size: 12
[    4.842006] ubi0: attaching mtd3
[    4.861052] ubi0: scanning is finished
[    4.864973] ubi0 error: ubi_read_volume_table: the layout volume was not found
[    4.872588] ubi0 error: ubi_attach_mtd_dev: failed to attach mtd3, error -22
[    4.879840] UBI error: cannot attach mtd3
[    4.883920] ubi4: default fastmap pool size: 25
[    4.888465] ubi4: default fastmap WL pool size: 12
[    4.893398] ubi4: attaching mtd7
[    4.946315] ubi4: scanning is finished
[    4.950360] ubi4 error: ubi_read_volume_table: the layout volume was not found
[    4.957842] ubi4 error: ubi_attach_mtd_dev: failed to attach mtd7, error -22
[    4.965101] UBI error: cannot attach mtd7
[    4.969271] UBI: block: can't open volume on ubi0_1, err=-19
```

v5.7 can attach mtd devices
v5.8 cannot. 

Bisection log:
```
$ git bisect log 
git bisect start
# bad: [bcf876870b95592b52519ed4aafcf9d95999bc9c] Linux 5.8
git bisect bad bcf876870b95592b52519ed4aafcf9d95999bc9c
# good: [3d77e6a8804abcc0504c904bd6e5cdf3a5cf8162] Linux 5.7
git bisect good 3d77e6a8804abcc0504c904bd6e5cdf3a5cf8162
# good: [694b5a5d313f3997764b67d52bab66ec7e59e714] Merge tag 'arm-soc-5.8' of git://git.kernel.org/pub/scm/linux/kernel/git/soc/soc
git bisect good 694b5a5d313f3997764b67d52bab66ec7e59e714
# good: [595a56ac1b0d5f0a16a89589ef55ffd35c1967a2] Merge tag 'linux-kselftest-kunit-5.8-rc1' of git://git.kernel.org/pub/scm/linux/kernel/git/shuah/linux-kselftest
git bisect good 595a56ac1b0d5f0a16a89589ef55ffd35c1967a2
# bad: [75613939084f59c0848b146e54ba463dc494c433] Merge tag 'powerpc-5.8-3' of git://git.kernel.org/pub/scm/linux/kernel/git/powerpc/linux
git bisect bad 75613939084f59c0848b146e54ba463dc494c433
```

shenki, 2020-08-12:
This is the culprit: https://github.com/torvalds/linux/commit/f9acd7fa80be6ee14aecdc54429f2a48e56224e8. Reverting it from dev-5.8 allows the ubi volume to attach again.

```
commit f9acd7fa80be6ee14aecdc54429f2a48e56224e8 (refs/bisect/bad)
Author: Pratyush Yadav <p.yadav@ti.com>
Date:   Mon May 25 14:45:31 2020 +0530

    mtd: spi-nor: sfdp: default to addr_width of 3 for configurable widths
    
    JESD216D.01 says that when the address width can be 3 or 4, it defaults
    to 3 and enters 4-byte mode when given the appropriate command. So, when
    we see a configurable width, default to 3 and let flash that default to
    4 change it in a post-bfpt fixup.
    
    This fixes SMPT parsing for flashes with configurable address width. If
    the SMPT descriptor advertises variable address width, we use
    nor->addr_width as the address width. But since it was not set to any
    value from the SFDP table, the read command uses an address width of 0,
    resulting in an incorrect read being issued.
    
    Signed-off-by: Pratyush Yadav <p.yadav@ti.com>
    Signed-off-by: Tudor Ambarus <tudor.ambarus@microchip.com>

diff --git a/drivers/mtd/spi-nor/sfdp.c b/drivers/mtd/spi-nor/sfdp.c
index f6038d3a3684..688aa36e863a 100644
--- a/drivers/mtd/spi-nor/sfdp.c
+++ b/drivers/mtd/spi-nor/sfdp.c
@@ -460,6 +460,7 @@ static int spi_nor_parse_bfpt(struct spi_nor *nor,
        /* Number of address bytes. */
        switch (bfpt.dwords[BFPT_DWORD(1)] & BFPT_DWORD1_ADDRESS_BYTES_MASK) {
        case BFPT_DWORD1_ADDRESS_BYTES_3_ONLY:
+       case BFPT_DWORD1_ADDRESS_BYTES_3_OR_4:
                nor->addr_width = 3;
                break;
```

geissonator, 2021-06-07:
Fixed per Joels find above.

closed 2021-06-07 by geissonator