openbmc/openbmc #3687: How to use Virtual Media to mount files or folders to host？

xflyh1230, 2020-08-20:
Hi，I want to use VirtualMedia feature to mount folder or file . Although the ISO file and cdrom device can be mounted to the host through NBD, but the folder and file other than ISO  cannot be mounted to the host.
How to use Virtual Media feature to mount files or folders to the host？Could you give me a configuration method to deal with this?

xflyh1230, 2020-08-24:
I'm reading VirtualMedia.md Document, ask for help for several questions:

1. Has the community updated and supported virtual media mounting of folders and files?
 If so, what do I need to do? How to update which modules to support?

2. If I want to mount folders and files, do I need to use CIFS / NFS? How to do this

3. In the document, I see the following DBUS interfaces, but my environment does not support these interfaces. 
Are they necessary for folder mounting? Which module needs to be updated to get these DBUS interfaces?
 At the same time, what services are needed to support it?
/xyz/openbmc_ project/VirtualMedia/Proxy/ISO0
/xyz/openbmc_ project/VirtualMedia/Proxy/1
/xyz/openbmc_ project/VirtualMedia/Legacy/0
/xyz/openbmc_ project/VirtualMedia/Legacy/1

My current adaptation situation is as follows:

1. The ISO or device can be mounted to the host through NBD,
That is: nbd-server is on the server side, BMC executes nbd-client through DBUS interface to load into / dev / nbdx,
 and the virtual device can be seen on the host side.

2. However, folders and files other than ISO and image cannot be mounted to the host computer.
I tried NFS and CIFS. Now I just execute client on BMC side to see the folder and file. 
But how to make the folder visible on the host side is making me scratch my head :-(
（How do files and folders mount like NBD devices?）

kingzmm, 2021-04-08:
@xflyh1230 Have you solved this problem? I am now studying this function (virtual media realizes USB mounting), thank you very much

jk-ozlabs, 2021-04-08:
No, the virtual media system does not support access to a generic file / folder. We don't currently have code to emulate a complete filesystem image to the host, and so the file that you export needs to already be in a filesystem image format.

However, the image that you serve does not necessarily need to be an ISO - any image format that the host system understands will work. We just export it as a normal block device to the host, which the host sees as a USB mass-storage device.

It's unlikely that we'll implement folder support with the current infrastructure, as that requires a facility for the JS code (running in the client browser) to have fairly arbitrary access to the underlying client filesystem, which is (sensibly) not premitted by the usual JS security models.

If you're doing this without the JS/NBD infrastructure, you'd still need to provide a block device to the host over USB, which requires constructing a filesystem image anyway; there may be a way to dynamically emulate a simple filesystem format, but the easiest approach is to pack your file/folder into a filesystem image, and then share that to the host using the existing infrastructure.

kingzmm, 2021-04-08:
@jk-ozlabs Thank you very much for your reply. Can I ask again, does virtual media support remote mounting of USB devices? If I want to implement remote mounting of USB devices through virtual media, what service should I choose? Can I trouble you for advice? Thank you very much.

jk-ozlabs, 2021-04-08:
@kingzmm: I'm not 100% not sure what you mean by "remote mounting" there - I assume you have a USB mass-storage device (say, plugged in to a laptop), and want to have that appear as virtual media to the host - is that correct?

If so, that's definitely possible - just select the device node of the USB device on the laptop (eg `/dev/sdb`) as the file to share over remote virtual media, when using the OpenBMC web interface.

I'm not sure if Windows/MacOS provide a suitable device node though; we may need someone familiar with those client OSes to help with any advice there.

kingzmm, 2021-04-08:
@jk-ozlabs：Thank you very much for your reply.I hope to log in to the BMC Web interface on the local PC and use the virtual media function to view the USB device on the remote server.Can you give me some suggestions?

kingzmm, 2021-04-08:
Now I can perform virtual media operations on ISO files, but USB devices cannot

jk-ozlabs, 2021-04-08:
> @jk-ozlabs：Thank you very much for your reply.I hope to log in to the BMC Web interface on the local PC and use the virtual media function to view the USB device on the remote server.Can you give me some suggestions?

We don't provide a facility for that; the USB devices on the host system are not shared with the BMC.

kingzmm, 2021-04-08:
@jk-ozlabs 👍 Okay, I'm sorry to disturb you, thank you very much for your reply.

jack206, 2023-06-12:
hI, I want to mount /dev/cdrom in Linux。but failed。here is log：
[Info    ] [system.hpp:311] operator()(): [DeviceMonitor]: "/dev/nbd0" removed.
[Info    ] [state_machine.hpp:68] emitEvent(): Slot_0 received UdevStateChangeEvent while in ReadyState
[Error   ] [ready_state.hpp:72] handleEvent(): Invalid event: UdevStateChangeEvent
@jk-ozlabs May I ask how to solve it？

jk-ozlabs, 2023-06-15:
@jack206 : I'm not sure what you're trying to do here, nor where those logs are from. Could you provide a bit of info? (preferably in a new github issue, as it doesn't seem related to this one).

Hong1230, 2023-08-25:
@jk-ozlabs Thank you very much for your reply and suggestion. For now, it is to package files/folders into a file system image, and then share that to the host using the existing infrastructure.

There is a new issue now. When using virtual media to mount a Windows Server 2022 image (5.1G), the mounting process is normal, but during the mounting boot installation, a driver loading failure will be reported. The error message is that the required media driver for the computer is missing.

May I ask if there is an issue with OpenBMC's support for Windows x86, or is this Windows image larger than 4G?

How should we solve it? Do we need to modify the kernel driver and how to do so?

Can I modify the window image to complete the mount boot installation? How to modify it?

LeeTroy, 2023-08-29:
Try this patch
https://gerrit.openbmc.org/c/openbmc/openbmc/+/42986

williamspatrick, 2023-08-29:
Closing issue as it isn’t an issue.

closed 2023-08-29 by williamspatrick