openbmc/openbmc #3692: Feature 'xyz' not found, but KERNEL_DANGLING_FEATURES_WARN_ONLY

geissonator, 2020-09-01:
With #3690 we set a warning flag to handle a new upstream poky "feature" in the kernel-yocto.bbclass. This issue is to track the needed refactor to get these kernel features out of the meta-phosphor layer and into the appropriate meta-aspeed and/or meta-nuvoton layers.

geissonator, 2020-09-01:
@amboar @shenki

geissonator, 2020-09-01:
The features in question come from the meta-phosphor/aspeed-layer/ and meta-phosphor/nuvoton-layer/ sub directories.

shenki, 2020-10-28:
AIUI, these features should have been placed in meta-aspeed and then the various machines can enable them from their layers.

We should make sure that this is the case, and put up reviews with each of the features. I suspect some are no longer required.

closed 2021-07-08 by stale[bot]