openbmc/openbmc #3693: gsj: libtraceevent-dynamic-list:2: syntax error in dynamic list

geissonator, 2020-09-02:
Moving to latest upstream poky his hitting issues compiling the gsj machine.

https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/35873

https://jenkins.openbmc.org/job/ci-openbmc/528/distro=ubuntu,label=docker-builder,target=gsj/console

```
/perf/1.0-r9/perf-1.0/tools/lib/ -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -DHAVE_SYNC_COMPARE_AND_SWAP_SUPPORT -DHAVE_PTHREAD_ATTR_SETAFFINITY_NP -DHAVE_PTHREAD_BARRIER -DHAVE_EVENTFD -DHAVE_GET_CURRENT_DIR_NAME -DHAVE_GETTID -DHAVE_DWARF_GETLOCATIONS_SUPPORT -DHAVE_GLIBC_SUPPORT -DHAVE_AIO_SUPPORT -DHAVE_SCHED_GETCPU_SUPPORT -DHAVE_SETNS_SUPPORT -DHAVE_LIBELF_SUPPORT -DHAVE_LIBELF_MMAP_SUPPORT -DHAVE_ELF_GETPHDRNUM_SUPPORT -DHAVE_GELF_GETNOTE_SUPPORT -DHAVE_ELF_GETSHDRSTRNDX_SUPPORT -DHAVE_LIBBPF_SUPPORT -DHAVE_JITDUMP -DHAVE_DWARF_UNWIND_SUPPORT -DHAVE_LIBUNWIND_SUPPORT -I/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/perf/1.0-r9/recipe-sysroot/usr/include -DHAVE_LIBCRYPTO_SUPPORT -DHAVE_SLANG_SUPPORT -DNO_LIBPERL -DHAVE_TIMERFD_SUPPORT -DNO_LIBPYTHON -DHAVE_CPLUS_DEMANGLE_SUPPORT -DHAVE_LIBBFD_SUPPORT -DHAVE_ZLIB_SUPPORT -DHAVE_LZMA_SUPPORT -DHAVE_BACKTRACE_SUPPORT -DDISASM_FOUR_ARGS_SIGNATURE -DHAVE_AUXTRACE_SUPPORT -I/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/ -ldl -lutil -Wl,-z,noexecstack -lunwind -lunwind-arm -L/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/perf/1.0-r9/recipe-sysroot/usr/lib -Xlinker --dynamic-list=/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/plugins/libtraceevent-dynamic-list \
| 	/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/perf-in.o /tmp/openbmc/work/gsj-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/pmu-events/pmu-events-in.o -Wl,--whole-archive /tmp/openbmc/work/gsj-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/libapi.a /tmp/openbmc/work/gsj-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/libtraceevent.a /tmp/openbmc/work/gsj-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/libsubcmd.a /tmp/openbmc/work/gsj-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/libperf.a /tmp/openbmc/work/gsj-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/libbpf.a  -Wl,--no-whole-archive -Wl,--start-group -lpthread -lrt -lm -ldl -lelf -lunwind -lunwind-arm -lcrypto -lslang -lbfd -lopcodes -liberty -lz -llzma -Wl,--end-group -o /tmp/openbmc/work/gsj-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/perf
| /tmp/openbmc/work/gsj-openbmc-linux-gnueabi/perf/1.0-r9/recipe-sysroot-native/usr/bin/arm-openbmc-linux-gnueabi/../../libexec/arm-openbmc-linux-gnueabi/gcc/arm-openbmc-linux-gnueabi/10.2.0/ld:/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/plugins/libtraceevent-dynamic-list:2: ignoring invalid character `@' in script
| /tmp/openbmc/work/gsj-openbmc-linux-gnueabi/perf/1.0-r9/recipe-sysroot-native/usr/bin/arm-openbmc-linux-gnueabi/../../libexec/arm-openbmc-linux-gnueabi/gcc/arm-openbmc-linux-gnueabi/10.2.0/ld:/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/plugins/libtraceevent-dynamic-list:2: ignoring invalid character `@' in script
| /tmp/openbmc/work/gsj-openbmc-linux-gnueabi/perf/1.0-r9/recipe-sysroot-native/usr/bin/arm-openbmc-linux-gnueabi/../../libexec/arm-openbmc-linux-gnueabi/gcc/arm-openbmc-linux-gnueabi/10.2.0/ld:/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/plugins/libtraceevent-dynamic-list:2: syntax error in dynamic list
| collect2: error: ld returned 1 exit status
| Makefile.perf:609: recipe for target '/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/perf' failed
| make[2]: *** [/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/perf] Error 1
| Makefile.perf:220: recipe for target 'sub-make' failed
| make[1]: *** [sub-make] Error 2
| Makefile:69: recipe for target 'all' failed
| make: *** [all] Error 2
| make: Leaving directory '/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/tools/perf'
| WARNING: exit code 1 from a shell command.
| 
[1;31mERROR[0m: [31mTask (/home/jenkins-slave/workspace/ci-openbmc/distro/ubuntu/label/docker-builder/target/gsj/openbmc/meta/recipes-kernel/perf/perf.bb:do_compile) failed with exit code '1'[0m
```

When I compile on my system, I hit something different, so not sure if we have multiple issues here or what:

```
| make[1]: Leaving directory '/home/andrewg/Code/openbmc/build/gsj/tmp/work/gsj-openbmc-linux-gnueabi/u-boot-nuvoton/git+npcm7xx-v2019.01+-r0/build'
| make: *** [Makefile:148: sub-make] Error 2
| make: Leaving directory '/home/andrewg/Code/openbmc/build/gsj/tmp/work/gsj-openbmc-linux-gnueabi/u-boot-nuvoton/git+npcm7xx-v2019.01+-r0/git'
| WARNING: /home/andrewg/Code/openbmc/build/gsj/tmp/work/gsj-openbmc-linux-gnueabi/u-boot-nuvoton/git+npcm7xx-v2019.01+-r0/temp/run.do_compile.2138426:221 exit 1 from 'exit 1'
| WARNING: Backtrace (BB generated script):
| 	#1: bbfatal_log, /home/andrewg/Code/openbmc/build/gsj/tmp/work/gsj-openbmc-linux-gnueabi/u-boot-nuvoton/git+npcm7xx-v2019.01+-r0/temp/run.do_compile.2138426, line 221
| 	#2: die, /home/andrewg/Code/openbmc/build/gsj/tmp/work/gsj-openbmc-linux-gnueabi/u-boot-nuvoton/git+npcm7xx-v2019.01+-r0/temp/run.do_compile.2138426, line 196
| 	#3: oe_runmake, /home/andrewg/Code/openbmc/build/gsj/tmp/work/gsj-openbmc-linux-gnueabi/u-boot-nuvoton/git+npcm7xx-v2019.01+-r0/temp/run.do_compile.2138426, line 191
| 	#4: do_compile, /home/andrewg/Code/openbmc/build/gsj/tmp/work/gsj-openbmc-linux-gnueabi/u-boot-nuvoton/git+npcm7xx-v2019.01+-r0/temp/run.do_compile.2138426, line 185
| 	#5: main, /home/andrewg/Code/openbmc/build/gsj/tmp/work/gsj-openbmc-linux-gnueabi/u-boot-nuvoton/git+npcm7xx-v2019.01+-r0/temp/run.do_compile.2138426, line 225
| 
| Backtrace (metadata-relative locations):
| 	#1: bbfatal_log, /home/andrewg/Code/openbmc/meta/classes/logging.bbclass, line 72
| 	#2: die, /home/andrewg/Code/openbmc/meta/classes/base.bbclass, line 56
| 	#3: oe_runmake, /home/andrewg/Code/openbmc/meta/classes/base.bbclass, line 65
| 	#4: do_compile, /home/andrewg/Code/openbmc/meta-nuvoton/recipes-bsp/u-boot/u-boot-nuvoton.inc, line 136
ERROR: Task (/home/andrewg/Code/openbmc/meta-nuvoton/recipes-bsp/u-boot/u-boot-nuvoton_git.bb:do_compile) failed with exit code '1'

```

geissonator, 2020-09-03:
Sounds a lot like https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1887397

geissonator, 2020-09-03:
I wonder if moving gsj up to linux 5.8 would fix this?

BenjaminFair, 2020-09-03:
We can move to 5.8 to fix the CI, but that version currently has issues on real hardware. I'm still working with Nuvoton and Quanta to resolve this, but moving to 5.8 in the meantime is ok with me.

geissonator, 2020-09-03:
ok cool, I put up https://gerrit.openbmc-project.xyz/c/openbmc/meta-nuvoton/+/36211 and then cherry-picked it into openbmc/openbmc via https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/36214/1 to see if it fixes the build for us.

geissonator, 2020-09-03:
Looks like that did the trick and got the system to pass. Be great if we could get https://gerrit.openbmc-project.xyz/c/openbmc/meta-nuvoton/+/36211 merged @BenjaminFair

geissonator, 2020-09-04:
Everything now merged into openbmc/openbmc master.

closed 2020-09-04 by geissonator