openbmc/openbmc #3694: No IP on qemuarm machine

xXorAa, 2020-09-08:
**Unexpected behavior you saw**
eth0 should have the IP passed on cmdline, it has no IP address.

**Expected behavior**
eth0 should have the passed IP address

**To Reproduce**
runqemu qemuarm nographic

**Screenshots**
qemuarm (v2.8.0-build)]$ runqemu qemuarm nographic                                                                                                                                            
runqemu - INFO - Running MACHINE=qemuarm bitbake -e ...                                                                                                                                                            
runqemu - INFO - Continuing with the following parameters:                                                                                                                                                         
KERNEL: [/mnt/work/graeme/Nuvia/Work/openbmc/qemuarm/tmp/deploy/images/qemuarm/zImage--5.4.34+git0+054d410d1a_440e41f2f4-r0-qemuarm-20200907191552.bin]                                                            
MACHINE: [qemuarm]                                                                                                                                                                                                 
FSTYPE: [ext4]                                                                                                                                                                                                     
ROOTFS: [/mnt/work/graeme/Nuvia/Work/openbmc/qemuarm/tmp/deploy/images/qemuarm/obmc-phosphor-image-qemuarm-20200907191552.rootfs.ext4]                                                                             
CONFFILE: [/mnt/work/graeme/Nuvia/Work/openbmc/qemuarm/tmp/deploy/images/qemuarm/obmc-phosphor-image-qemuarm-20200907191552.qemuboot.conf]                                                                         
                                                                                                                                                                                                                   
runqemu - INFO - Acquiring lockfile /tmp/qemu-tap-locks/tap0.lock failed: [Errno 11] Resource temporarily unavailable                                                                                              
runqemu - INFO - Setting up tap interface under sudo                                                                                                                                                               
[sudo] password for graeme:                                                                                                                                                                                        
runqemu - INFO - Network configuration: ip=192.168.7.4::192.168.7.3:255.255.255.0                        
runqemu - INFO - Running /mnt/work/graeme/Nuvia/Work/openbmc/qemuarm/tmp/work/x86_64-linux/qemu-helper-native/1.0-r1/recipe-sysroot-native/usr/bin/qemu-system-arm -device virtio-net-device,netdev=net0,mac=52:54:
00:12:34:04 -netdev tap,id=net0,ifname=tap1,script=no,downscript=no -drive id=disk0,file=/mnt/work/graeme/Nuvia/Work/openbmc/qemuarm/tmp/deploy/images/qemuarm/obmc-phosphor-image-qemuarm-20200907191552.rootfs.ex
t4,if=none,format=raw -device virtio-blk-device,drive=disk0 -show-cursor -device VGA,edid=on -device qemu-xhci -device usb-tablet -device usb-kbd -object rng-random,filename=/dev/urandom,id=rng0 -device virtio-r
ng-pci,rng=rng0  -nographic -machine virt,highmem=off -cpu cortex-a15 -m 256 -serial mon:stdio -serial null -kernel /mnt/work/graeme/Nuvia/Work/openbmc/qemuarm/tmp/deploy/images/qemuarm/zImage--5.4.34+git0+054d4
10d1a_440e41f2f4-r0-qemuarm-20200907191552.bin -append 'root=/dev/vda rw  console=ttyS0 mem=256M ip=192.168.7.4::192.168.7.3:255.255.255.0 console=ttyAMA0 ' 

root@qemuarm:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast qlen 1000
    link/ether 52:54:00:12:34:04 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::5054:ff:fe12:3404/64 scope link 
       valid_lft forever preferred_lft forever
3: sit0@NONE: <NOARP,UP,LOWER_UP> mtu 1480 qdisc noqueue qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
    inet6 ::192.168.7.4/96 scope global 
       valid_lft forever preferred_lft forever
    inet6 ::127.0.0.1/96 scope host 
       valid_lft forever preferred_lft forever

root@qemuarm:~# cat /proc/cmdline 
root=/dev/vda rw  console=ttyS0 mem=256M ip=192.168.7.4::192.168.7.3:255.255.255.0 console=ttyAMA0 

**OpenBMC Information:**
 - /etc/os-release:
root@qemuarm:~# cat /etc/os-release 
ID=openbmc-phosphor
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="2.8.0-0"
VERSION_ID=2.8.0-0-g314c4ecc0
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.8.0-0"
BUILD_ID="2.8.0"
OPENBMC_TARGET_MACHINE="qemuarm"

(but also affects master)

 - uname -a:
root@qemuarm:~# uname -a
Linux qemuarm 5.4.34-yocto-s-dirty-440e41f #1 SMP PREEMPT Mon Sep 7 20:53:38 UTC 2020 armv7l GNU/Linux

mdmillerii, 2020-09-08:
OpenBMC Phosphor has a explicity networking setup for static routing but by default will be a DHCP client.   The `ip=` kernel command line is not configured or used in most systems and we do not push the ip information to u-boot to feed back to the kernel.  (Note: the `ip=` parameter can be useful in the static layout `phosphor-initrd-scripts` initramfs environment as it will allow fetching files to allow update or recovery from a broken flash image (if Das U-Boot is intact).

xXorAa, 2020-09-08:
So the runqemu script does not setup things correctly for DHCP, and adds the ip= so it seems there is a disconnect there from what it does and the intent in the image. Note by default it sets up a tap device with static IPs and no DHCP server on the gateway tap device.

legoater, 2020-09-08:
The above command line is running `-machine virt,highmem=off -cpu cortex-a15` which is not a QEMU Aspeed machine.

closed 2021-07-08 by stale[bot]