openbmc/openbmc #3696: Linux Kernel Enable CIFS and NFS Failed

chineweff, 2020-09-09:
We need to mount CIFS/NFS from remote server to implement virtual-media.

I try command like this:
mount -t nfs ${IP}:filepath /mnt
mount -t cifs //${IP}/filepath /mnt

And I got error messages
[NFS]
mount.nfs: No such device

[CIFS]
Password for wayne@//10.248.43.97/public:  ****
mount error: cifs filesystem not supported by the system
mount error(19): No such device

After that I enable some tools:
cifs-utils
nfs-utils
nfs-export-root

Also I added some options in my kernel config to enable CIFS and NFS filesystem
CONFIG_NFS_FS=y
CONFIG_CIFS=y

but I found that linux-aspeed did't build these two config.
The .o file did not appear in my build folder.

And in linux-aspeed build folder image/boot/config doesn't have CONFIG_CIFS and CONFIG_NFS_FS.
Even did not show #XXX is not set

My linux kernel version:
uname -a
Linux c2080 5.4.23-6ba7086-dirty-4db3074 #1 Wed Sep 9 01:55:40 UTC 2020 armv6l GNU/Linux

Does anyone have any ideas?

Tulasinathreddy, 2021-06-14:
even i am facing the same issue !!

Yanglizhi, 2023-07-26:
Has anyone updated on this issue? CIFS appears to be available, but NFS is not implemented

thangqn-ampere, 2023-07-26:
> Has anyone updated on this issue? CIFS appears to be available, but NFS is not implemented

Let try to add below if you can enable CONFIG_NFS
```
CONFIG_NFS_FS=y
CONFIG_NFS_V2=y
CONFIG_NFS_V3=y
CONFIG_NFS_V4=y
CONFIG_NFS_COMMON=y
CONFIG_NFSD=y
```

On my system's BMC console:

```
# zcat /proc/config.gz | grep NFS
# CONFIG_USB_FUNCTIONFS is not set
CONFIG_KERNFS=y
CONFIG_NFS_FS=y
CONFIG_NFS_V2=y
CONFIG_NFS_V3=y
# CONFIG_NFS_V3_ACL is not set
CONFIG_NFS_V4=y
# CONFIG_NFS_V4_1 is not set
# CONFIG_NFS_USE_LEGACY_DNS is not set
CONFIG_NFS_USE_KERNEL_DNS=y
CONFIG_NFS_DISABLE_UDP_SUPPORT=y
CONFIG_NFSD=y
# CONFIG_NFSD_V3_ACL is not set
# CONFIG_NFSD_V4 is not set
CONFIG_NFS_COMMON=y
```

Yanglizhi, 2023-07-26:
> Let try to add below if you can enable CONFIG_NFS
>
> ```
> CONFIG_NFS_FS=y
> CONFIG_NFS_V2=y
> CONFIG_NFS_V3=y
> CONFIG_NFS_V4=y
> CONFIG_NFS_COMMON=y
> CONFIG_NFSD=y
> ```
>
> On my system's BMC console:
>
> ```
> # zcat [...]

Thanks！It seems that enable nfs  config only work for bmc itself, how can host virtual media support nfs?

In the introduction to virtual media development guide , it seems that only cifs is supported.
[image]

thangqn-ampere, 2023-07-26:
> Thanks！It seems that enable nfs config only work for bmc itself, how can host virtual media support nfs?

I don't understand your question. BMC VirtualMedia mounts ISO files from CIFS or NFS and export to Host as a virtual USB CD-ROM device. From Host side you can see a USB CD-ROM device that you can use.
About VirtualMedia supports CIFS and/or NFS, it is on the virtual media backend support which is missing now. refer to https://github.com/openbmc/bmcweb/issues/188

closed 2023-09-25 by williamspatrick