openbmc/openbmc #3697: evb-ast2500: systemd hang and QEMU CI failure

leiyu-bytedance, 2020-09-10:
We have an internal Jenkins job to do QEMU CI on `evb-ast2500`, and it's noticed that the CI fails recently.
We do not preserve a long history, and the earliest failure occurs on 3c0e2b42742151, but it could be earlier.

The CI fails because:
* ssh to BMC hangs
* `systemctl` hangs

The top shows that the systemd is consuming most of the CPU:
```
Mem: 98016K used, 395456K free, 11948K shrd, 0K buff, 56920K cached
CPU:  46% usr  53% sys   0% nic   0% idle   0% io   0% irq   0% sirq
Load average: 1.39 1.35 0.96 2/88 342
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
    1     0 root     R    26348   5%  79% {systemd} /sbin/init
  161   160 messageb S     2808   1%  13% dbus-broker --log 4 --controller 9 --machine-id 2c0824c702914e4eb83641bcc05ef314 --max-bytes 536870912 --max-fds 4096 --max-matches 16384 --audit
```

The journal log is attached.
[log_ast2500_evb_qemu_ci.log](https://github.com/openbmc/openbmc/files/5198861/log_ast2500_evb_qemu_ci.log)

geissonator, 2020-09-11:
Once we get this fixed, feel free to post the results of your CI to the gerrit reviews so we can prevent breaks from getting in.
Should be able to follow similar path as https://github.com/openbmc/openbmc/wiki/Adding-a-System-to-Hardware-Continuous-Integration.

leiyu-bytedance, 2020-09-28:
@amboar This issue looks related to the finding that you posted in the IRC.
Would you please cherry-pick the systemd fixes?

amboar, 2020-09-30:
@geissonator sent a systemd bump upstream for openembedded, I guess we'll pick that up @leiyu-bytedance

leiyu-bytedance, 2020-10-09:
Confirmed that on c3d88e4d9fcc08e1aae7cc9d0337c0261e996c64 the issue is gone.

closed 2020-10-09 by leiyu-bytedance