openbmc/openbmc #3698: Factory Reset does't work

229024527, 2020-09-10:
Hello everyone, when i  use restapi to reset bmc firmware but it response Unauthorized , who can help me ?

root@palmetto:~# cat /etc/issue
Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 \n \l

root@palmetto:~# uname -a 
Linux palmetto 5.4.28-cb30920 #1 Tue Apr 7 08:29:06 UTC 2020 armv5tejl GNU/Linux

1. export bmc=192.168.11.129
2. export token=`curl -k -H "Content-Type: application/json" -X POST https://${bmc}:2443/login -d '{"username" :  "root", "password" :  "0penBmc"}' | grep token | awk '{print $2;}' | tr -d '"'`
3.curl -b cjar -k -H 'Content-Type: application/json' -X POST -d '{"data":[]}' https://${bmc}:2443/xyz/openbmc_project/software/action/Reset
Unauthorized

anoo1, 2020-09-10:
> 2\. 3.curl -b cjar -k -H 'Content-Type: application/json' -X POST -d '{"data":[]}' https://${bmc}:2443/xyz/openbmc_project/software/action/Reset

Try passing the token generated in step 2 instead of using the cookie cjar:
```
curl -k -H "X-Auth-Token: $token" -H 'Content-Type: application/json' -X POST -d '{"data":[]}' https://${bmc}:2443/xyz/openbmc_project/software/action/Reset
```

anoo1, 2020-09-10:
For reference, the doc with examples for the REST API have been updated to deprecate the cookie argument: https://github.com/openbmc/docs/blob/master/REST-cheatsheet.md

229024527, 2020-09-10:
> Try passing the token generated in step 2 instead of using the cookie cjar:
>
> ```
> curl -k -H "X-Auth-Token: $token" -H 'Content-Type: application/json' -X POST -d '{"data":[]}' [...]

Hi，i try to use cjar but response Unauthorized again
1.export bmc=192.168.11.129
2.curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST https://${bmc}:2443/login -d "{\"data\": [ \"root\", \"0penBmc\" ] }"
{
  "data": "User 'root' logged in",
  "message": "200 OK",
  "status": "ok"
}
3.curl -b cjar -k -H 'Content-Type: application/json' -X POST -d '{"data":[]}' https://${bmc}:2443/xyz/enbmc_project/software/action/Reset
Unauthorized

229024527, 2020-09-11:
> Try passing the token generated in step 2 instead of using the cookie cjar:
>
> ```
> curl -k -H "X-Auth-Token: $token" -H 'Content-Type: application/json' -X POST -d '{"data":[]}' [...]

Thank you very much， it works

closed 2020-09-11 by 229024527