openbmc/openbmc #3703: command not response ,journal not message

kingzmm, 2020-09-27:
**Note on Bugs**
Please note that questions or feature requests should be discussed on the
mailing list or IRC. Please see https://github.com/openbmc/openbmc#contact
for info on how to contact us. Github Issues in openbmc/openbmc are for bugs
only. Please remove this section when opening a bug.

**Unexpected behavior you saw**
A clear and concise description of what unexpectedly happened.

**Expected behavior**
A clear and concise description of what you expected to happen.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Screenshots**
If applicable, add screenshots to help explain your problem.

**OpenBMC Information:**
 - /etc/os-release:
 - uname -a:
 - Any additional patches or changes:

**Additional context**
Add any other context about the problem here.

solonfan, 2020-10-20:
What is your issue?

kingzmm, 2020-10-20:
yes,but I have solved this problem

closed 2020-11-03 by kingzmm