openbmc/openbmc #3705: Error writing to eMMC with kernel 5.8.8

anoo1, 2020-09-28:
A mmc code update on Tacoma running the latest master with kernel 5.8.8 fails with:

```
Sep 28 20:08:05 witherspoon-tacoma-YA30UF03400F obmc-flash-bmc[572]: Error: Partition(s) 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 48, 49, 50, 51, 52, 53, 54, 55, 56, 57, 58, 59, 60, 61, 62, 63, 64 on /dev/mmcblk0boot0 have been written, but we have been unable to inform the kernel of the change, probably because it/they are in use.  As a result, the old partition(s) will remain in use.  You should reboot now before making further changes.
```

mdmillerii, 2020-09-28:
The error is likely the ioctl to reread the partition table is failing.
This will fail if any partition is mounted.

Are you trying to re-partition or just modify the existing partition contents?

Also why is it printing all 64(-1) possible partitions?

anoo1, 2020-10-07:
> Are you trying to re-partition or just modify the existing partition contents?

Just dd to an existing partition to write a new image.

> Also why is it printing all 64(-1) possible partitions?

Yeah good question, not sure why it's printing all those partitions, including the boot ones.

anoo1, 2020-10-07:
I've been trying git bisect a few times without success on finding a commit that introduced this issue.
Today I moved to 5.8.12 and this is not seen anymore, so something along the way got fixed. Closing.

closed 2020-10-07 by anoo1