openbmc/openbmc #3708: restapi factory reset "description": "org.freedesktop.DBus.Error.NoReply"

kingzmm, 2020-09-30:
factory reset error
{
  "data": {
    "description": "org.freedesktop.DBus.Error.NoReply"
  },
  "message": "Remote peer disconnected",
  "status": "error"
}

mdmillerii, 2020-10-03:
Are you noting that you are getting this error?

Note that getting a reply to a reset is inherently racy because performing the reset involves loosing all state.  There have been hacks to insert a short delay for the ack to make it back to the client in some cases (possibly reboot BMC vs this factory reset activation).

kingzmm, 2020-10-09:
How to solve this problem

kingzmm, 2020-10-09:
@mdmillerii How to solve this problem

mdmillerii, 2020-10-09:
You can add a delay and hope or accept the system will disappear when you ask it to loose its current state and accept that in the client, looking for a clean state after an appropriate delay.

kingzmm, 2020-10-09:
@mdmillerii Could you please tell me the specific location of the modified file? I did not find the file

mdmillerii, 2020-10-09:
I don't know offhand.

mdmillerii, 2020-10-09:
Actually since it looks like a web server reply look at the journal for a crashed process when dbus communication occurred.

kingzmm, 2020-10-09:
@mdmillerii Do you know the approximate location,Please tell me,Thank you very much

kingzmm, 2020-10-09:
@mdmillerii Do you know restapi factory reset Where is the call

kingzmm, 2020-10-09:
@mdmillerii Could you help me? The simulation can pass, but the actual measurement still reports the same problem

anoo1, 2020-10-09:
@kingzmm can you post the REST call you're using to issue the factory reset? Also do other REST calls return success or error?

kingzmm, 2020-10-10:
@anoo1 I don't know if any other reset will work

anoo1, 2020-10-12:
Just asking if you'd paste here the command you're using to issue the factory reset. The second question was if you've tried any other commands like doing a GET to just read the contents of an interface, to check that the error is only seen on a factory reset command and not on any REST call.

Some additional data that would be useful would be the version of openbmc (cat /etc/os-release), and the journal output when the factory reset was issued, before issuing the factory reset: journalctl -f &

kingzmm, 2020-10-12:
@anoo1 journalctl :
openpower-update-manager[1046]:Failed to read VERSION
openpower-update-manager[1046]:Unable to read from version
openpower-update-manager[1046]:Error version is empty

kingzmm, 2020-10-12:
@anoo1 I suspect it was caused by an error in pnor

kingzmm, 2020-10-12:
@anoo1 Do you think pnor may cause factory reset to report errors?

kingzmm, 2020-10-14:
Resolved: shield void ItemUpdaterStatic::reset() in openpower-software-manager/static/ item_updater_static.cpp，add one return Skip this function

closed 2020-11-03 by kingzmm