openbmc/openbmc #3710: how to add power information in openbmc

akky1605, 2020-10-01:
HI,

* I am new to openbmc and trying to understand how we can add new information in openbmc.
* I have a hwmon entry which provides the average power consumption.
* I need to show this information through my openbmc.

* I thought of one way through using sensor, and created a conf file for the same.
* I created a conf file in meta-phosphor layer at location: openbmc/meta-phosphor/recipes-phosphor/sensors/phosphor-hwmon/obmc/hwmon/soc/i2c@7e804000/power@3c.conf
conf file: power@3c.conf
`
$ cat obmc/hwmon/soc/i2c@7e804000/power@3c.conf                 
LABEL_power1=power1_input
`

I add a phosphor-hwmon_%.bbappend file at location "openbmc/meta-phosphor/recipes-phosphor/sensors"
content:
`$ cat phosphor-hwmon_%.bbappend
FILESEXTRAPATHS_prepend := "${THISDIR}/${PN}:"

CHIPS = "i2c@7e804000/power@3c"
ITEMSFMT = "soc/{0}.conf"

ITEMS = "${@compose_list(d, 'ITEMSFMT', 'CHIPS')}"

ENVS = "obmc/hwmon/{0}"
SYSTEMD_ENVIRONMENT_FILE_${PN} += "${@compose_list(d, 'ENVS', 'ITEMS')}"
`
* I created the package using "bitbake phosphor-hwmon" and installed in the bmc target.

* I am able to see my conf file in BMC target at location :  "/etc/default/obmc/hwmon/soc/i2c@7e804000/"
`
$ cat /etc/default/obmc/hwmon/soc/i2c@7e804000/power@3c.conf
LABEL_power1=power1_input
`
* I am able to see my device in devicetree
` 
ls -l /sys/firmware/devicetree/base/soc/i2c@7e804000/power@3c/
-r--r--r--    1 root     root            10 Sep 24 10:46 compatible
-r--r--r--    1 root     root             6 Sep 24 10:46 name
-r--r--r--    1 root     root             4 Sep 24 10:46 reg
`

* Similarly I am also able to see the device in sys/hwmon entry and able to read the value from it.
* I ran the journalctl command to check the status
`
$  journalctl | grep -i hwmon
Sep 25 03:31:17 raspberrypi3-64 systemd[1]: Started Phosphor Hwmon Poller.
Sep 25 03:31:17 raspberrypi3-64 systemd[1]: xyz.openbmc_project.Hwmon@-soc-i2c\x407e804000-power\x403c.service: Succeeded.
`
$ But when I checked the service status:
`
root@raspberrypi3-64:/# systemctl status xyz.openbmc_project.Hwmon@-soc-i2c\x407e804000-power\x403c.service
● xyz.openbmc_project.Hwmon@-soc-i2cx407e804000-powerx403c.service - Phosphor Hwmon Poller
     Loaded: loaded (/lib/systemd/system/xyz.openbmc_project.Hwmon@.service; static; vendor preset: enabled)
     Active: inactive (dead)

Sep 24 09:18:53 raspberrypi3-64 systemd[1]: Condition check resulted in Phosphor Hwmon Poller being skipped.
`
* I dont know the reason for why the hwmon pollr is skipping.

* I am doubting the conf file  I created.
* Please let me know what I am doing wrong.
* Also it will be really helpful if you let me know if there is any other way to read the power consumption.

Thank you for the help.

Akshay

spinler, 2020-10-08:
It looks to me like you have everything right...  

I guess you could try a udevadm info on the hwmon dir and check the OF_FULLNAME path still matches your conf file path, like
```
udevadm info /sys/class/hwmon/hwmon3                                                   
...
E: OF_FULLNAME=/ahb/apb/bus@1e78a000/i2c-bus@200/power-supply@68
```

akky1605, 2020-10-13:
Hi,

Thanks for the update.

I verified the path using udevadm info. The path is coming correct.

I tried to restart the service.
In restarting I get this error:
```
 systemctl status xyz.openbmc_project.Hwmon@-soc-i2c\x407e804000-power\x403c
● xyz.openbmc_project.Hwmon@-soc-i2cx407e804000-powerx403c.service - Phosphor Hwmon Poller
     Loaded: loaded (/lib/systemd/system/xyz.openbmc_project.Hwmon@.service; static; vendor preset: enabled)
     Active: inactive (dead)

Condition check resulted in Phosphor Hwmon Poller being skipped
```
Can you let me know where this condition check is happening. I am not able to get which condition check is happening & in which file?

spinler, 2020-10-13:
That come from this line in the service file:
```
ConditionFileNotEmpty=/etc/default/obmc/hwmon/%I.conf
```
So that implies your conf file path isn't quite right.

akky1605, 2020-10-14:
Hi, 

Thanks for the update.

I verified the path and it seems to be right.
What I am doubting is the naming convention of LABEL and others in the conf file.

I am currently using the conf file as, power@3c.conf
```
$ cat obmc/hwmon/soc/i2c@7e804000/power@3c.conf                 
LABEL_power1=power1_input 
```

can you please verify if this is correct?

Thank you.

spinler, 2020-10-14:
Assuming there is a power1_input file in your hwmon directory, that looks right.

akky1605, 2020-10-21:
Hi,

Sorry for late reply.
Thanks for the update. My sysfs entry name was power1_average that seems to be the issue.

After changing the sysfs entry name to power1_input, the entries got populated.

One thing if need to keep the sysfs entry name as power1_average, what will be the changes in conf file as I tried with power1_average, it did not work??
So with sysfs entry file name as **power1_average** , I used the conf file as:
```
$ cat obmc/hwmon/soc/i2c@7e804000/power@3c.conf                 
LABEL_power1=power1_average
```
but the above did not work. 
What modification is require for the same??

Thank you in advance.

mdmillerii, 2020-10-21:
the _input vs _average label should be based on the function, the hwmon subsystem has some pretty strict rules on the names of these files.

Please review https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/Documentation/hwmon/hwmon-kernel-api.rst?h=v5.9

akky1605, 2020-10-23:
Thank you for the update and help.
I will close the ticket.

closed 2020-10-23 by akky1605