openbmc/openbmc #3715: x86-64 SDK fails to build

leiyu-bytedance, 2020-10-09:
On c3d88e4d9fcc08e1aae7cc9d0337c0261e996c64, the x86-64 SDK fails to build with below error:
```
ERROR: Nothing RPROVIDES 'packagegroup-core-device-devel' (but /opt/jenkins/workspace/OpenBMC/OpenBMC-Build-SDK/label/openbmc/target/qemux86-64/openbmc-sdk/meta/recipes-core/packagegroups/packagegroup-base.bb RDEPENDS on or otherwise requires it)
NOTE: Runtime target 'packagegroup-core-device-devel' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-core-device-devel']
NOTE: Runtime target 'packagegroup-base-extended' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-base-extended', 'packagegroup-core-device-devel']
ERROR: Required build target 'obmc-phosphor-image' has no buildable providers.
Missing or unbuildable dependency chain was: ['obmc-phosphor-image', 'packagegroup-base-extended', 'packagegroup-core-device-devel']
```
While on 7a08bfe5abd1dec200da73021e42f10b3ed38228, the build is fine.

I know the x86-64 SDK is not officially supported by OpenBMC, but it was supported and we have an internal Jenkins job to build it. So it is created here to log the issue.

leiyu-bytedance, 2021-01-21:
This issue is gone on 8ab9600f96abbf3390507aadb85d50ce4259108c, most likely it's fixed in an earlier version.

closed 2021-01-21 by leiyu-bytedance

lxwinspur, 2021-01-21:
Hi, Lei.
when I used x86-64 SDK to built pldm, All tests are skipped:
```
$ ninja -C build test
ninja: Entering directory `build'
[161/162] Running all tests.
 1/23 libpldm_base_test                            SKIP           0.00s
 2/23 libpldm_platform_test                        SKIP           0.00s
 3/23 libpldm_bios_test                            SKIP           0.00s
 4/23 libpldm_bios_table_test                      SKIP           0.00s
 5/23 libpldm_fru_test                             SKIP           0.00s
 6/23 libpldm_utils_test                           SKIP           0.00s
 7/23 libpldm_pdr_test                             SKIP           0.00s
 8/23 libpldmresponder_base_test                   SKIP           0.00s
 9/23 libpldmresponder_bios_test                   SKIP           0.00s
10/23 libpldmresponder_bios_attribute_test         SKIP           0.00s
11/23 libpldmresponder_bios_string_attribute_test  SKIP           0.00s
12/23 libpldmresponder_bios_config_test            SKIP           0.00s
13/23 libpldmresponder_bios_integer_attribute_test SKIP           0.00s
14/23 libpldmresponder_bios_enum_attribute_test    SKIP           0.00s
15/23 libpldmresponder_pdr_effecter_test           SKIP           0.00s
16/23 libpldmresponder_pdr_sensor_test             SKIP           0.00s
17/23 libpldmresponder_bios_table_test             SKIP           0.00s
18/23 libpldmresponder_platform_test               SKIP           0.00s
19/23 pldmd_instanceid_test                        SKIP           0.00s
20/23 pldmd_registration_test                      SKIP           0.00s
21/23 pldm_utils_test                              SKIP           0.00s
22/23 libpldmresponder_fru_test                    SKIP           0.00s
23/23 libpldmresponder_dbus_to_host_effecter_test  SKIP           0.00s

Summary of Failures:

 1/23 libpldm_base_test                            SKIP           0.00s
 2/23 libpldm_platform_test                        SKIP           0.00s
 3/23 libpldm_bios_test                            SKIP           0.00s
 4/23 libpldm_bios_table_test                      SKIP           0.00s
 5/23 libpldm_fru_test                             SKIP           0.00s
 6/23 libpldm_utils_test                           SKIP           0.00s
 7/23 libpldm_pdr_test                             SKIP           0.00s
 8/23 libpldmresponder_base_test                   SKIP           0.00s
 9/23 libpldmresponder_bios_test                   SKIP           0.00s
10/23 libpldmresponder_bios_attribute_test         SKIP           0.00s
11/23 libpldmresponder_bios_string_attribute_test  SKIP           0.00s
12/23 libpldmresponder_bios_config_test            SKIP           0.00s
13/23 libpldmresponder_bios_integer_attribute_test SKIP           0.00s
14/23 libpldmresponder_bios_enum_attribute_test    SKIP           0.00s
15/23 libpldmresponder_pdr_effecter_test           SKIP           0.00s
16/23 libpldmresponder_pdr_sensor_test             SKIP           0.00s
17/23 libpldmresponder_bios_table_test             SKIP           0.00s
18/23 libpldmresponder_platform_test               SKIP           0.00s
19/23 pldmd_instanceid_test                        SKIP           0.00s
20/23 pldmd_registration_test                      SKIP           0.00s
21/23 pldm_utils_test                              SKIP           0.00s
22/23 libpldmresponder_fru_test                    SKIP           0.00s
23/23 libpldmresponder_dbus_to_host_effecter_test  SKIP           0.00s

Ok:                 0   
Expected Fail:      0   
Fail:               0   
Unexpected Pass:    0   
Skipped:            23  
Timeout:            0   
```

Have you met the same problem as me?

lxwinspur, 2021-01-21:
Built x86-64 SDK successfully on https://github.com/openbmc/openbmc/commit/55fb85d142fd2c8fcb2dc1ea3c27e50974670763

leiyu-bytedance, 2021-01-21:
> when I used x86-64 SDK to built pldm, All tests are skipped:

This is expected because the tests are run as "cross-compiled" and thus they are skipped.
You should try to pick the change in https://lore.kernel.org/openbmc/CAARXrtk20BTpqFac1LiR8f6eHE_g+=rQwd__kU5qkbEsAMSktg@mail.gmail.com/ or use repo-ci to run the tests.