openbmc/openbmc #3716: ipmitool fru list error Incomplete printing

kingzmm, 2020-10-14:
root@palmetto:~# ipmitool fru list
FRU Device Description : Builtin FRU Device (ID 0)
 Board Mfg Date        : Unspecified

Question: Why can only one of the above displayed information be printed? How to complete the printing of ipmitool fru list

solonfan, 2020-10-20:
<ipmitool fru> is enough,i think u didnt specified the FRU info,you should fill the FRU device first.

1.Generate the fru filed,make it included into FRU.bin;write the fru.bin into flash: ipmitool fru write 0 xxx.bin

2.And u could edit/modify the fru info with cmd $ ipmitool fru edit 0 field c 1 ***

Thanks,
SolonFan

kingzmm, 2020-10-20:
@solonfan Thank you for your answer. I have performed all these operations, but the contents displayed on the board, product and chassis are not complete. For example, I want to display the relevant contents of Board/Product/Chassis Part Number, but I don’t know now. Where to perform the operation, I guessed at the beginning that it was written in the relevant yaml file, I put the picture below, hoping to answer
As product
0:
  /system:
    entityID: 25
    entityInstance: 1
    interfaces:
      xyz.openbmc_project.Inventory.Decorator.Asset:
        Manufacturer:
          IPMIFruProperty: Manufacturer
          IPMIFruSection: Product
        SerialNumber:
          IPMIFruProperty: Serial Number
          IPMIFruSection: Product
        Model:
          IPMIFruProperty: Model Number
          IPMIFruSection: Product
        PartNumber:
          IPMIFruProperty: Part Number
          IPMIFruSection: Product
      xyz.openbmc_project.Inventory.Decorator.Revision:
        Version:
          IPMIFruProperty: Version
          IPMIFruSection: Product
      xyz.openbmc_project.Inventory.Item:
        PrettyName:
          IPMIFruProperty: Name
          IPMIFruSection: Product

root@palmetto:~# ipmitool fru list
FRU Device Description : Builtin FRU Device (ID 0)
 Product Manufacturer  : Alibaba
 Product Version       : 0.01
 Product Serial        : 12345678998765432112345

closed 2020-11-03 by kingzmm