openbmc/openbmc #3720: intermittent compile failure - os-release: The metadata is not deterministic and this needs to be fixed

geissonator, 2020-10-21:
This started appearing intermittently in our build after picking up these two commits:
```
    boost: temp fix for upstream yocto boost change (commit: 36fe5df) (details / githubweb)
    poky: subtree update:b23aa6b753..ad30a6d470 (commit: af5e4ef) (details / githubweb)
```
I don't see a rhyme or reason yet, it has hit multiple build machines.

Full error message
```
[441A[J[1;31mERROR[0m: [31mWhen reparsing /var/lib/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/witherspoon/openbmc/meta/recipes-core/os-release/os-release.bb:do_compile, the basehash value changed from 5da39bb75d4e4256ff7e589f6204c0ee79f16031e54aa78a23d630633becceac to c266b6051c26e27a7f5585d632b703511a7a9657fa2960515b570d3ef54a8ef2. The metadata is not deterministic and this needs to be fixed.[0m
[1;31mERROR[0m: [31mThe following commands may help:[0m
[1;31mERROR[0m: [31m$ bitbake os-release -cdo_compile -Snone[0m
[1;31mERROR[0m: [31mThen:[0m
[1;31mERROR[0m: [31m$ bitbake os-release -cdo_compile -Sprintdiff
```

leiyu-bytedance, 2020-10-22:
We hit the same issue on internal build as well.
It's a daily build and currently it occurs on ed31ac3e8946bb

geissonator, 2020-10-22:
Usually when weird things like this happen, it's related to sstate cache. I'm cleaning up that on the systems that are failing to see if it helps.

geissonator, 2020-10-22:
Cleaning up caches does not seem to be helping. https://lists.yoctoproject.org/g/yocto/topic/61340522 seems to offer some clues.

geissonator, 2020-10-22:
I put up https://gerrit.openbmc-project.xyz/c/openbmc/meta-phosphor/+/37593 but not confident it really fixes anything. Maybe works around whatever is happening.

geissonator, 2020-10-27:
So builds are now working with that change but I'm seeing flash images get generated that have invalid os-release hash values. At times, images are coming out of CI with the same hash, even though they have different content. Sometimes there's even a -dirty in the os-release which makes no sense.

geissonator, 2020-10-27:
I've put a revert of my "fix" at https://gerrit.openbmc-project.xyz/c/openbmc/meta-phosphor/+/37705

geissonator, 2020-10-27:
I think the issue is that initially yocto is using a cached os-release, and then during the build process, it switches over to the correct one and hits the error. https://www.kc8apf.net/2017/09/all-about-those-caches/ talks a bit about our use case. I wonder if something changed upstream with our use of BB_DONT_CACHE

geissonator, 2020-10-27:
I posted our issue over on yocto mailing list, lets see if they've got any ideas - https://lists.yoctoproject.org/g/yocto/message/51198

geissonator, 2020-11-03:
I took another stab at this based on upstream yocto feedback with https://gerrit.openbmc-project.xyz/c/openbmc/meta-phosphor/+/37866. Ran into something new here though where phosphor-ipmi-config depends on os-release but if you build phosphor-ipmi-config, it seems to only build the base os-release recipe, and not the .bbappend. It then gets the basic 0.1.0 VERSION which the parser used by phosphor-ipmi-config does not understand how to parse.

```
ERROR[phosphor-ipmi-config-1.0-r1 do_patch: Error executing a python function in exec_python_func() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_python_func() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:do_patch(d)
     0003:
File: 'autogenerated', lineno: 14, function: do_patch
  File "autogenerated", line 14, in do_patch

Exception: IndexError: list index out of range
[0m
Currently  8 running tasks (3633 of 6178)  58% ||
```

I think it's time to revert the poky commit that caused this. This is having significant impacts to our builds.

geissonator, 2020-11-03:
Needed reverts:
https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/37874
https://gerrit.openbmc-project.xyz/c/openbmc/meta-phosphor/+/37885

closed 2020-11-18 by geissonator