openbmc/openbmc #3724: Current master is broken

ya-mouse, 2020-10-30:
The recent commit af5e4ef732faedf66c6dc1756432e9de2ac72988 broke the build:

https://pastebin.com/5tTYHd68

The do_package() step tries to chown the base-files/3.0.14-r89/package/usr directory to 0:0 uid/gid

I've tried to remove build/tmp and re-run several times. Fails with the same error. If I use the previous commit (36fe5df200a94e3ce82ba2dcad16c0a4127f6d46) it builds without any issues.

ya-mouse, 2020-10-30:
and it also fails on base-files package as well with the same error

geissonator, 2020-10-31:
I've seen no issues, and CI has been fine (other then https://github.com/openbmc/openbmc/issues/3720). Are you starting with a fresh build directory? Maybe try clearing out any shared sstate?

ya-mouse, 2020-11-04:
It seems like recent update brings the PSEUDO_IGNORE_PATH support that has `/usr` in it. For the reasons, the build happens under /usr. By removing this rule everything is going to work.

edtanous, 2020-12-03:
@ya-mouse  Considering this is a bug in yocto and not really openbmc, do you think you could file it with yocto using their process?  We'll cross link it here and hopefully drive it to closure.

rpurdie, 2021-01-07:
Do you really need to run the build under /usr ? This was added to PSEUDO_IGNORE_PATHS as we didn't want the overhead of pseudo database trips for files under /usr, whilst we could probably remove it from that variable, it will trigger more database roundtrips in the pseudo code/tasks. You probably could remove it in local configuration?

ya-mouse, 2021-01-10:
Yeah, I think this should work. Thanks!

closed 2021-01-10 by ya-mouse