openbmc/openbmc #3727: how to build witherspoon-tacoma mtd file

chiyanglin-AStar, 2020-11-11:
I am not sure whether it is bug or just less doc.  

I try to build witherspoon-tacoma , and run it in qemu , 

step like : 

. setup witherspoon-tacoma build 
DISTRO=openbmc-openpower MACHINE=witherspoon-tacoma bitbake obmc-phosphor-image

after building images , I didn't find .mtd in images folder !  

what happen or need to run extra instruction ? 

any advice , will be helpful ! Tks a lot!

anoo1, 2020-11-11:
Hi. The tacoma image is for an eMMC instead of a NOR, so the output file would be a .wic.xz. Here are the commands for qemu: https://github.com/openbmc/qemu/wiki/Usage#tacoma-emmc although they're currently a work in progress.

chiyanglin-AStar, 2020-11-12:
got it ! Tks a lot !   I have rebuild and try it on qemu , still can't mount /home!  

But I found https://openpower.xyz/job/openbmc-build/distro=ubuntu,label=builder,target=witherspoon-tacoma/lastSuccessfulBuild/artifact/deploy/images/witherspoon-tacoma/    have .mtd image , where can know this steps or instruction ?

anoo1, 2020-11-12:
To build an .mtd image, revert the following commits that switched tacome to building eMMC:

https://github.com/openbmc/openbmc/commit/c569800bf46dde3c8ebd4962efb6d7f3027712eb
https://github.com/openbmc/openbmc/commit/a9893155ce1a82074984bc3a0e0a5d936af7c10d

chiyanglin-AStar, 2020-11-12:
it is working !  Tks a lot!

closed 2020-11-12 by chiyanglin-AStar

apoorvemohan, 2021-02-04:
> To build an .mtd image, revert the following commits that switched tacome to building eMMC:
> 
> [c569800](https://github.com/openbmc/openbmc/commit/c569800bf46dde3c8ebd4962efb6d7f3027712eb)
> [a989315](https://github.com/openbmc/openbmc/commit/a9893155ce1a82074984bc3a0e0a5d936af7c10d)

@anoo1 Thanks for the instructions. I was able to build an mtd image using these instructions you specified. 

Next, I tried building an mtd image but with UBOOT_MACHINE = "ast2600_openbmc_spl_defconfig" and SPL_BINARY = "spl/u-boot-spl.bin". The build process completed without any error, however, when I launch it with QEMU, it freezes (with no output on the terminal). 

Any pointers on building a witherspoon-tacoma mtd image with u-boot-spl?
I am not sure whether it is bug or am I making any mistake. 

I would really appreciate any help on this. Thanks!

chiyanglin-AStar, 2021-02-04:
try the following , I have built it ,successfully ! and can run on Qemu 

KERNEL_DEVICETREE = "aspeed-bmc-opp-tacoma.dtb"
#UBOOT_MACHINE = "ast2600_openbmc_spl_emmc_defconfig"
UBOOT_MACHINE = "ast2600_openbmc_spl_defconfig"
UBOOT_DEVICETREE = "ast2600-tacoma"
SPL_BINARY = "spl/u-boot-spl.bin"

OBMC_POWER_SUPPLY_INSTANCES = "0 1"

MACHINEOVERRIDES =. "ibm-ac-server:witherspoon:"
require conf/machine/include/ast2600.inc
require conf/machine/include/obmc-bsp-common.inc
require conf/machine/include/openpower.inc
require conf/machine/include/p9.inc

#  For mtd format 
#require conf/distro/include/openpower-virtual-pnor.inc
#require conf/distro/include/phosphor-mmc.inc

SERIAL_CONSOLES = "115200;ttyS4"

# change image from emmc to mtd and add some packages
FLASH_SIZE = "131072"
IMAGE_INSTALL_append = " bash"
IMAGE_INSTALL_append = " ipmitool "
IMAGE_FSTYPES += "cpio.${INITRAMFS_CTYPE}.u-boot"
IMAGE_FSTYPES += "ext4 mmc-ext4-tar"
EXTRA_IMAGE_FEATURES = "debug-tweaks package-management"
# ------------------------------------------------------

MRW_XML = "witherspoon.xml"
PHOSPHOR_MRW_LICENSE = "Apache-2.0"
PHOSPHOR_MRW_LIC_FILES_CHKSUM = "file://LICENSE;md5=e3fc50a88d0a364313df4b21ef20c29e"
PHOSPHOR_MRW_URI = "git://github.com/open-power/witherspoon-xml"
PHOSPHOR_MRW_REV = "c622cb5a5dd3ebc2a4eef558d1b70740f914e6f7"

VIRTUAL-RUNTIME_phosphor-hwmon-config_df-mrw = ""

PREFERRED_PROVIDER_virtual/phosphor-fan-control-fan-config_df-mrw = \
    "phosphor-fan-control-fan-config"
PREFERRED_PROVIDER_virtual/phosphor-fan-presence-config_df-mrw = \
    "phosphor-fan-presence-config"

# The eMMC on Tacoma is just 1GB smaller than Rainier. The template file
# meta-aspeed/wic/emmc-aspeed.wks.in builds an image for the size of Rainier as
# the default. To be able to re-use that template for Tacoma, the size of the
# last partition (hostfw) was made configurable. The default size of the hostfw
# partition is 5GB, therefore setting it to 4GB for Tacoma.
#WKS_HOSTFW_SIZE = "4G"

IMAGE_FEATURES_remove = "obmc-ikvm"

chiyanglin-AStar, 2021-02-04:
Try the following , I can build it successfully , and run it on qemu!

KERNEL_DEVICETREE = "aspeed-bmc-opp-tacoma.dtb"
#UBOOT_MACHINE = "ast2600_openbmc_spl_emmc_defconfig"
UBOOT_MACHINE = "ast2600_openbmc_spl_defconfig"
UBOOT_DEVICETREE = "ast2600-tacoma"
SPL_BINARY = "spl/u-boot-spl.bin"

OBMC_POWER_SUPPLY_INSTANCES = "0 1"

MACHINEOVERRIDES =. "ibm-ac-server:witherspoon:"
require conf/machine/include/ast2600.inc
require conf/machine/include/obmc-bsp-common.inc
require conf/machine/include/openpower.inc
require conf/machine/include/p9.inc

#  For mtd format 
#require conf/distro/include/openpower-virtual-pnor.inc
#require conf/distro/include/phosphor-mmc.inc

SERIAL_CONSOLES = "115200;ttyS4"

# change image from emmc to mtd and add some packages
FLASH_SIZE = "131072"
IMAGE_INSTALL_append = " bash"
IMAGE_INSTALL_append = " ipmitool "
IMAGE_FSTYPES += "cpio.${INITRAMFS_CTYPE}.u-boot"
IMAGE_FSTYPES += "ext4 mmc-ext4-tar"
EXTRA_IMAGE_FEATURES = "debug-tweaks package-management"
# ------------------------------------------------------

MRW_XML = "witherspoon.xml"
PHOSPHOR_MRW_LICENSE = "Apache-2.0"
PHOSPHOR_MRW_LIC_FILES_CHKSUM = "file://LICENSE;md5=e3fc50a88d0a364313df4b21ef20c29e"
PHOSPHOR_MRW_URI = "git://github.com/open-power/witherspoon-xml"
PHOSPHOR_MRW_REV = "c622cb5a5dd3ebc2a4eef558d1b70740f914e6f7"

VIRTUAL-RUNTIME_phosphor-hwmon-config_df-mrw = ""

PREFERRED_PROVIDER_virtual/phosphor-fan-control-fan-config_df-mrw = \
    "phosphor-fan-control-fan-config"
PREFERRED_PROVIDER_virtual/phosphor-fan-presence-config_df-mrw = \
    "phosphor-fan-presence-config"

# The eMMC on Tacoma is just 1GB smaller than Rainier. The template file
# meta-aspeed/wic/emmc-aspeed.wks.in builds an image for the size of Rainier as
# the default. To be able to re-use that template for Tacoma, the size of the
# last partition (hostfw) was made configurable. The default size of the hostfw
# partition is 5GB, therefore setting it to 4GB for Tacoma.
#WKS_HOSTFW_SIZE = "4G"

IMAGE_FEATURES_remove = "obmc-ikvm"

apoorvemohan, 2021-02-04:
@chiyanglin-AStar I tried the following conf but when launching under QEMU, everything freezes:
> $ cat ~/openbmc-mtd1/meta-ibm/conf/machine/witherspoon-tacoma.conf
> KERNEL_DEVICETREE = "aspeed-bmc-opp-tacoma.dtb"
> #UBOOT_MACHINE = "ast2600_openbmc_spl_emmc_defconfig"
> UBOOT_MACHINE = "ast2600_openbmc_spl_defconfig"
> UBOOT_DEVICETREE = "ast2600-tacoma"
> SPL_BINARY = "spl/u-boot-spl.bin"
>
> OBMC_POWER_SUPPLY_INSTANCES = "0 1"
>
> MACHINEOVERRIDES =. "ibm-ac-server:witherspoon:"
> require conf/machine/include/ast2600.inc
> require conf/machine/include/obmc-bsp-common.inc
> require conf/machine/include/openpower.inc
> require conf/machine/include/p9.inc
> 
> #For mtd format
> #require conf/distro/include/openpower-virtual-pnor.inc
> #require conf/distro/include/phosphor-mmc.inc
> 
> SERIAL_CONSOLES = "115200;ttyS4"
> 
> #change image from emmc to mtd and add some packages
> FLASH_SIZE = "131072"
> IMAGE_INSTALL_append = " bash"
> IMAGE_INSTALL_append = " ipmitool "
> IMAGE_FSTYPES += "cpio.${INITRAMFS_CTYPE}.u-boot"
> IMAGE_FSTYPES += "ext4 mmc-ext4-tar"
> EXTRA_IMAGE_FEATURES = "debug-tweaks package-management"
> MRW_XML = "witherspoon.xml"
> PHOSPHOR_MRW_LICENSE = "Apache-2.0"
> PHOSPHOR_MRW_LIC_FILES_CHKSUM = "file://LICENSE;md5=e3fc50a88d0a364313df4b21ef20c29e"
> PHOSPHOR_MRW_URI = "git://github.com/open-power/witherspoon-xml"
> PHOSPHOR_MRW_REV = "c622cb5a5dd3ebc2a4eef558d1b70740f914e6f7"
>
> VIRTUAL-RUNTIME_phosphor-hwmon-config_df-mrw = ""
> 
> PREFERRED_PROVIDER_virtual/phosphor-fan-control-fan-config_df-mrw = "phosphor-fan-control-fan-config"
> PREFERRED_PROVIDER_virtual/phosphor-fan-presence-config_df-mrw = "phosphor-fan-presence-config"
>
> #The eMMC on Tacoma is just 1GB smaller than Rainier. The template file
> #meta-aspeed/wic/emmc-aspeed.wks.in builds an image for the size of Rainier as
> #the default. To be able to re-use that template for Tacoma, the size of the
> #last partition (hostfw) was made configurable. The default size of the hostfw
> #partition is 5GB, therefore setting it to 4GB for Tacoma.
> #WKS_HOSTFW_SIZE = "4G"
> 
> IMAGE_FEATURES_remove = "obmc-ikvm"

Here is the QEMU command I am using: 
>$ qemu-system-arm --version
QEMU emulator version 5.0.0 (Debian 1:5.0-5ubuntu6~ppa0)
Copyright (c) 2003-2020 Fabrice Bellard and the QEMU Project developers
> $ qemu-system-arm -M tacoma-bmc -nographic -no-reboot -nic user -drive file=~/openbmc-mtd1/build/tmp/deploy/images/witherspoon-tacoma/obmc-phosphor-image-witherspoon-tacoma.static.mtd,format=raw,if=mtd

chiyanglin-AStar, 2021-02-04:
try this : 

~/qemu-system-arm -M tacoma-bmc -m 1024 -nic user -drive file=./obmc-phosphor-image-witherspoon-tacoma.static.mtd,format=raw,if=mtd -nographic -net nic -net user,

apoorvemohan, 2021-02-04:
> try this :
> 
> ~/qemu-system-arm -M tacoma-bmc -m 1024 -nic user -drive file=./obmc-phosphor-image-witherspoon-tacoma.static.mtd,format=raw,if=mtd -nographic -net nic -net user,

Still the same behavior. No output on the screen - no command prompt either.

apoorvemohan, 2021-02-06:
[Update] 
@chiyanglin-AStar I was finally able to build mtd image with u-boot-spl for witherspoon-tacoma with the configuration your suggested above. However, it does not work on commits after [f3e7cee](https://github.com/openbmc/openbmc/commit/f3e7ceeb25c4e1809fd54eb7b6ef97ce4ceba3c6).