openbmc/openbmc #3730: 1 Task failed "icu_67.1.bb:do_compile"

SuprajaChirumamindla, 2020-11-16:
Hello!

I have encountered the below error when I tried compiling the OpenBMC with the icu-67.1.bb.
Can someone kindly help me by letting me know how to debug the error.

ERROR ENCOUNTERED:
.....................................................................................................................................................................................................................................................................
ERROR: icu-native-67.1-r0 do_compile: oe_runmake failed
ERROR: icu-native-67.1-r0 do_compile: Execution of '/project/sw/users/suprajac/IntelOpenBMC/openbmc-intel-zip/download/intelbmc/openbmc-intel/build/tmp/work/x86_64-linux/icu-native/67.1-r0/temp/run.do_compile.275978' failed with exit code 1:
Note: rebuild with "make VERBOSE=1 " to show all compiler parameters.
make[0]: Making `all' in `stubdata'
make[1]: Entering directory `/project/sw/users/suprajac/IntelOpenBMC/openbmc-intel-zip/download/intelbmc/openbmc-intel/build/tmp/work/x86_64-linux/icu-native/67.1-r0/build/stubdata'
make[1]: Nothing to be done for `all'.
make[1]: Leaving directory `/project/sw/users/suprajac/IntelOpenBMC/openbmc-intel-zip/download/intelbmc/openbmc-intel/build/tmp/work/x86_64-linux/icu-native/67.1-r0/build/stubdata'
make[0]: Making `all' in `common'
make[1]: Entering directory `/project/sw/users/suprajac/IntelOpenBMC/openbmc-intel-zip/download/intelbmc/openbmc-intel/build/tmp/work/x86_64-linux/icu-native/67.1-r0/build/common'
   g++   ...  ../../icu/source/common/utext.cpp
   g++   ...  ../../icu/source/common/uarrsort.cpp
   g++   ...  ../../icu/source/common/servrbf.cpp
   g++   ...  ../../icu/source/common/servslkf.cpp
   g++   ...  ../../icu/source/common/uidna.cpp
   g++   ...  ../../icu/source/common/usprep.cpp
   g++   ...  ../../icu/source/common/uts46.cpp
   g++   ...  ../../icu/source/common/punycode.cpp
   g++   ...  ../../icu/source/common/util.cpp
   g++   ...  ../../icu/source/common/util_props.cpp
   g++   ...  ../../icu/source/common/parsepos.cpp
   g++   ...  ../../icu/source/common/locbased.cpp
   g++   ...  ../../icu/source/common/cwchar.cpp
   g++   ...  ../../icu/source/common/wintz.cpp
../../icu/source/common/uarrsort.cpp: In function ‘constexpr int32_t sizeInMaxAlignTs(int32_t)’:
../../icu/source/common/uarrsort.cpp:40:34: error: ‘max_align_t’ is not a member of ‘std’
     return (sizeInBytes + sizeof(std::max_align_t) - 1) / sizeof(std::max_align_t);
                                  ^
../../icu/source/common/uarrsort.cpp:40:34: note: suggested alternative:
In file included from /usr/include/c++/4.8.2/cstddef:42:0,
                 from ../../icu/source/common/uarrsort.cpp:21:
/usr/lib/gcc/x86_64-redhat-linux/4.8.5/include/stddef.h:425:3: note:   ‘max_align_t’
 } max_align_t;
   ^
../../icu/source/common/uarrsort.cpp:40:66: error: ‘max_align_t’ is not a member of ‘std’
     return (sizeInBytes + sizeof(std::max_align_t) - 1) / sizeof(std::max_align_t);
                                                                  ^
../../icu/source/common/uarrsort.cpp:40:66: note: suggested alternative:
In file included from /usr/include/c++/4.8.2/cstddef:42:0,
                 from ../../icu/source/common/uarrsort.cpp:21:
/usr/lib/gcc/x86_64-redhat-linux/4.8.5/include/stddef.h:425:3: note:   ‘max_align_t’
 } max_align_t;
   ^
../../icu/source/common/uarrsort.cpp:41:1: error: body of constexpr function ‘constexpr int32_t sizeInMaxAlignTs(int32_t)’ not a return-statement
 }
 ^
../../icu/source/common/uarrsort.cpp: In function ‘void insertionSort(char*, int32_t, int32_t, int32_t (*)(const void*, const void*, const void*), const void*, UErrorCode*)’:
../../icu/source/common/uarrsort.cpp:144:26: error: ‘max_align_t’ is not a member of ‘std’
     icu::MaybeStackArray<std::max_align_t, sizeInMaxAlignTs(STACK_ITEM_SIZE)> v;
                          ^
../../icu/source/common/uarrsort.cpp:144:26: note: suggested alternative:
In file included from /usr/include/c++/4.8.2/cstddef:42:0,
                 from ../../icu/source/common/uarrsort.cpp:21:
/usr/lib/gcc/x86_64-redhat-linux/4.8.5/include/stddef.h:425:3: note:   ‘max_align_t’
 } max_align_t;
   ^
../../icu/source/common/uarrsort.cpp:144:26: error: ‘max_align_t’ is not a member of ‘std’
     icu::MaybeStackArray<std::max_align_t, sizeInMaxAlignTs(STACK_ITEM_SIZE)> v;
                          ^
../../icu/source/common/uarrsort.cpp:144:26: note: suggested alternative:
In file included from /usr/include/c++/4.8.2/cstddef:42:0,
                 from ../../icu/source/common/uarrsort.cpp:21:
/usr/lib/gcc/x86_64-redhat-linux/4.8.5/include/stddef.h:425:3: note:   ‘max_align_t’
 } max_align_t;
[... 80 lines cut ...]
                 ut->pExtra    = &((ExtendedUText *)ut)->extension;
                                                         ^
*** Failed compilation command follows: ----------------------------------------------------------
g++ -isystem/project/sw/users/suprajac/IntelOpenBMC/openbmc-intel-zip/download/intelbmc/openbmc-intel/build/tmp/work/x86_64-linux/icu-native/67.1-r0/recipe-sysroot-native/usr/include -D_REENTRANT -DU_HAVE_ELF_H=1 -DU_HAVE_STRTOD_L=1 -DU_HAVE_XLOCALE_H=1 -I../../icu/source/common -DDEFAULT_ICU_PLUGINS="/project/sw/users/suprajac/IntelOpenBMC/openbmc-intel-zip/download/intelbmc/openbmc-intel/build/tmp/work/x86_64-linux/icu-native/67.1-r0/recipe-sysroot-native/usr/lib/icu"  -DU_ATTRIBUTE_DEPRECATED= -DU_COMMON_IMPLEMENTATION -isystem/project/sw/users/suprajac/IntelOpenBMC/openbmc-intel-zip/download/intelbmc/openbmc-intel/build/tmp/work/x86_64-linux/icu-native/67.1-r0/recipe-sysroot-native/usr/include -O2 -pipe -W -Wall -pedantic -Wpointer-arith -Wwrite-strings -Wno-long-long -std=c++11 -c -DPIC -fPIC -o uarrsort.o ../../icu/source/common/uarrsort.cpp
--- ( rebuild with "make VERBOSE=1 all" to show all parameters ) --------
make[1]: *** [uarrsort.o] Error 1
make[1]: *** Waiting for unfinished jobs....
   g++   ...  ../../icu/source/common/ucnvsel.cpp
*** Failed compilation command follows: ----------------------------------------------------------
g++ -isystem/project/sw/users/suprajac/IntelOpenBMC/openbmc-intel-zip/download/intelbmc/openbmc-intel/build/tmp/work/x86_64-linux/icu-native/67.1-r0/recipe-sysroot-native/usr/include -D_REENTRANT -DU_HAVE_ELF_H=1 -DU_HAVE_STRTOD_L=1 -DU_HAVE_XLOCALE_H=1 -I../../icu/source/common -DDEFAULT_ICU_PLUGINS="/project/sw/users/suprajac/IntelOpenBMC/openbmc-intel-zip/download/intelbmc/openbmc-intel/build/tmp/work/x86_64-linux/icu-native/67.1-r0/recipe-sysroot-native/usr/lib/icu"  -DU_ATTRIBUTE_DEPRECATED= -DU_COMMON_IMPLEMENTATION -isystem/project/sw/users/suprajac/IntelOpenBMC/openbmc-intel-zip/download/intelbmc/openbmc-intel/build/tmp/work/x86_64-linux/icu-native/67.1-r0/recipe-sysroot-native/usr/include -O2 -pipe -W -Wall -pedantic -Wpointer-arith -Wwrite-strings -Wno-long-long -std=c++11 -c -DPIC -fPIC -o utext.o ../../icu/source/common/utext.cpp
--- ( rebuild with "make VERBOSE=1 all" to show all parameters ) --------
make[1]: *** [utext.o] Error 1
make[1]: Leaving directory `/project/sw/users/suprajac/IntelOpenBMC/openbmc-intel-zip/download/intelbmc/openbmc-intel/build/tmp/work/x86_64-linux/icu-native/67.1-r0/build/common'
make: *** [all-recursive] Error 2
WARNING: /project/sw/users/suprajac/IntelOpenBMC/openbmc-intel-zip/download/intelbmc/openbmc-intel/build/tmp/work/x86_64-linux/icu-native/67.1-r0/temp/run.do_compile.275978:1 exit 1 from 'exit 1'

ERROR: Logfile of failure stored in: /project/sw/users/suprajac/IntelOpenBMC/openbmc-intel-zip/download/intelbmc/openbmc-intel/build/tmp/work/x86_64-linux/icu-native/67.1-r0/temp/log.do_compile.275978

ERROR: Task (virtual:native:/project/sw/users/suprajac/IntelOpenBMC/openbmc-intel-zip/download/intelbmc/openbmc-intel/meta/recipes-support/icu/icu_67.1.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1564 tasks of which 1558 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  virtual:native:/project/sw/users/suprajac/IntelOpenBMC/openbmc-intel-zip/download/intelbmc/openbmc-intel/meta/recipes-support/icu/icu_67.1.bb:do_compile
Summary: There were 3 WARNING messages shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

amboar, 2020-11-16:
It looks like you're trying to build https://github.com/intel-bmc/openbmc. Have you confirmed this is a problem with upstream openbmc? It's not clear from the output you've posted which revision of what repo you're trying to build (i.e. the commit ID). If you haven't confirmed this is an issue in upstream openbmc (i.e. the repo you've opened this issue against) then you should consider moving this issue to https://github.com/intel-bmc/openbmc.

Having said that, you're probably compiling Intel's OpenBMC fork on a host distro that is either too old or too new (as you're building `icu-native`, which uses the host distro's compiler as part of the bootstrap process). Consider building it under a supported host distro.

williamspatrick, 2020-11-16:
`/usr/lib/gcc/x86_64-redhat-linux/4.8.5/include/stddef.h`

This indicates to me the “too old of distro” theory. Your OS has GCC 4.8.5, which is ancient.  The 4.8 series was first released in 2013.

SuprajaChirumamindla, 2020-11-20:
I have appended the path of the GCC-10.2.0 binary to the "PATH" variable, but I have no idea why it's still using the older version.
Please let me know if you have any idea why is it happening so.

amboar, 2020-11-20:
@SuprajaChirumamindla can you please list the exact steps you've executed to try to resolve the issue? For example, copy/pasting the commands and output from your terminal. It's hard help you with so little detail, beyond suggesting you use a supported host distro.

mdmillerii, 2020-11-20:
The `PATH` variable is searched from first to last.  If you have a command in two different directories the first one found will be used.   What happens if you place the directory with the new compiler version early in your path before the other copies?

SuprajaChirumamindla, 2020-11-20:
I am a remote user for a server so I have installed the GCC-10.2.0 version in my local directory and appended the path of the GCC-10.2.0 binary to the "PATH" environment variable and checked the version before compiling.
It gives the latest version(GCC-10.2.0) when I check(using the command "gcc --version") but the older version is being used during compilation.I don't understand why is it happening so.
.....................................................................................................................................................................................................................................................................

[suprajac@hw-ts2 openbmc]$ export PATH=/home/suprajac/GCC-10.2.0/bin:/home/suprajac/TAR-1.32/bin:$PATH
[suprajac@hw-ts2 openbmc]$ which gcc
~/GCC-10.2.0/bin/gcc
[suprajac@hw-ts2 openbmc]$ gcc --version
gcc (GCC) 10.2.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

[suprajac@hw-ts2 openbmc]$ echo $PATH
/home/suprajac/GCC-10.2.0/bin:/home/suprajac/TAR-1.32/bin:/project/sw/users/suprajac/IntelOpenBMC/openbmc-intel-zip/download/openbmc/scripts:/project/sw/users/suprajac/IntelOpenBMC/openbmc-intel-zip/download/openbmc/poky/bitbake/bin:/home/suprajac/GCC-10.2.0/bin:/home/suprajac/TAR-1.32/bin:/usr/lib64/qt-3.3/bin:/usr/local/bin:/usr/bin:/usr/local/sbin:/usr/sbin:/home/suprajac/.local/bin:/home/suprajac/bin

.

amboar, 2020-12-01:
@SuprajaChirumamindla Please use the triple back-tick markdown when pasting code or console output into github issues. In addition to improving readability it prevents github's rendering engine interpreting your log output as markdown. What you write looks like:

    ```
    My code, console or log output here
    ```

A guide to using github-flavoured markdown is here:

https://guides.github.com/features/mastering-markdown/

amboar, 2020-12-01:
@SuprajaChirumamindla Whether or not changing your path works will likely depend on whether you already have the package configured under bitbake. I suggest you at least start by cleaning the package out first and retrying. It's not clear from what you've posted what state you're starting from. So at least, please try:

1. setting your `$PATH`
2. sourcing the bitbake environment script
3. `bitbake -c cleanall icu-native`
4. `bitbake icu-native`

williamspatrick, 2020-12-01:
I think that bitbake rarely honors `$PATH` or if it does it only does it on the first run when the build directory is created.  Inside your build directory you will see `tmp/hosttools/gcc` (and `g++`).  These are the programs that are typically used for compiling native code inside bitbake.  If these symlinks are pointed at `/usr/bin/gcc` (or similar) this might explain why `$PATH` isn't being honored.

TBH, your most straight-forward path is going to be using a modern distribution or at least something with Docker.  Yocto itself doesn't support something so old so it isn't easy for us to add on support for you.  `meta-poky/conf/distro/poky.conf` has a variable `SANITY_TESTED_DISTROS`, which gives a good indication of the distributions supported by Yocto.  (As an aside, I have no idea why we have `SANITY_TESTED_DISTROS_append ?= "*"` in most of our `local.conf.sample` files but that is what turns off the warning for an unsupported distro like yours.)

@bradbishop ^ - should we disable this override of `SANITY_TESTED_DISTROS`?

closed 2021-07-08 by stale[bot]