openbmc/openbmc #3732: recipes-kernel/linux-libc-headers/linux-libc-headers_5.8.bb:do_install) failed with exit code '134'

ttsai1, 2020-11-26:
Hi 
I had build openbmc follow https://github.com/openbmc/openbmc  README.md.

but build failed, Any idea for it? 

Ubuntu ver: 14.04.1-Ubuntu
tony@B3S1SVN:/media2/tony/openbmc/openbmc$ . setup romulus build
Machine romulus found in meta-ibm/meta-romulus
You had no conf/local.conf file. This configuration file has therefore been
created for you with some default values. You may wish to edit it to, for
example, select a different MACHINE (target hardware). See conf/local.conf
for more information as common configuration options are commented.

You had no conf/bblayers.conf file. This configuration file has therefore been
created for you with some default values. To add additional metadata layers
into your configuration please add entries to conf/bblayers.conf.

The Yocto Project has extensive documentation about OE including a reference
manual which can be found at:
    http://yoctoproject.org/documentation

For more information about OpenEmbedded see their website:
    http://www.openembedded.org/

Common targets are:
     obmc-phosphor-image
tony@B3S1SVN:/media2/tony/openbmc/openbmc/build$ bitbake obmc-phosphor-image
Loading cache: 100% |                                                                                                                                                                | ETA:  --:--:--
Loaded 0 entries from dependency cache.
Parsing recipes: 100% |###############################################################################################################################################################| Time: 0:01:07
Parsing of 2503 .bb files complete (0 cached, 2503 parsed). 3790 targets, 371 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.48.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-14.04"
TARGET_SYS           = "arm-openbmc-linux-gnueabi"
MACHINE              = "romulus"
DISTRO               = "openbmc-romulus"
DISTRO_VERSION       = "0.1.0"
TUNE_FEATURES        = "arm thumb arm1176jzs"
TARGET_FPU           = "soft"
meta
meta-oe
meta-networking
meta-perl
meta-python
meta-phosphor
meta-aspeed
meta-openpower
meta-romulus         = "master:7cff96a9d077030a18a48c489c93bf5ac93f7a79"

Initialising tasks: 100% |############################################################################################################################################################| Time: 0:00:04
Sstate summary: Wanted 1597 Found 0 Missed 1597 Current 0 (0% match, 0% complete)
NOTE: Executing Tasks
ERROR: Task (/media2/tony/openbmc/openbmc/meta/recipes-kernel/linux-libc-headers/linux-libc-headers_5.8.bb:do_install) failed with exit code '134'
NOTE: Tasks Summary: Attempted 267 tasks of which 0 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /media2/tony/openbmc/openbmc/meta/recipes-kernel/linux-libc-headers/linux-libc-headers_5.8.bb:do_install

vingoseu, 2020-12-25:
I also have the same problem, and have no ideas about it. Anyone resolves?

xianleon, 2020-12-25:
By changing meta/recipes-devtools/pseudo/pseudo_git.bb: SRCREV to  "f3ae0c96e012426a8bcb6f881d1c8c30f9673b14" (old version of pseudo), I was able to continue to build in my Ubuntu 18.04 (WSL2) machine.
Not sure if Ubuntu 20.04 has this issue
Commit d8d1c38bc12c283e7d113a2dfe0691fca804b570 of pseudo package shall give the hint, at least for me. But why "path mismatch" happened?

williamspatrick, 2020-12-28:
Per https://github.com/openbmc/openbmc/blob/master/poky/meta-poky/conf/distro/poky.conf#L43, Ubuntu 14.04 is not supported by Yocto itself, so it is not something we can really provide additional support for.  I do not believe Yocto supports WSL2 either.

Since you have isolated this to a potential issue with pseudo, I suggest reporting upstream to the Yocto community.  I'm not sure how much interest they have in supporting WSL2 officially.

By the way, the log in the first report is not useful to help debug.  All it says is that "do_install" failed in "linux-libc-headers" but there is nothing anyone can do to investigate further.  Yocto should have also pointed you to a full log file which would have given the error that "do_install" encountered.  In the future, please also include this (and if you want to report upstream they will similarly need it).

closed 2021-09-06 by stale[bot]