openbmc/openbmc #3733: Server Health Critical; High Error. No more information

stewartsmith, 2020-11-29:
**Unexpected behavior you saw**
When logging into the web UI to boot my system:

Server health: Critical, `xyz.openbmc_project.Common.Error.InternalFailure`

Here's all the detail:
```
None _PID=1181
```

That's it.

**Expected behavior**
At least some detail on what component failed.

**To Reproduce**
No idea.

**Screenshots**
[image]

**OpenBMC Information:**
```
root@blackbird:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="2.7.0-dev"
VERSION_ID="2.7.0-dev-581-g18878e4f6"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.7.0-dev"
BUILD_ID="2.7.0-dev"
OPENBMC_TARGET_MACHINE="blackbird"
root@blackbird:~# uname -a
Linux blackbird 5.0.7-551af99b5cadeeff8bb97b97b3afa82421b7eca0 #1 Wed Feb 19 11:45:06 UTC 2020 armv6l GNU/Linux
```

williamspatrick, 2020-11-30:
Have you ran this on something newer?  2.7 was released in August 2019 and you’re running a pre-2.7 image.  Is this even an OpenBMC image or a non-contributed fork by the company that made your system?

Did you look for any core dump files?

amboar, 2020-12-01:
Perhaps a better direction for this is, if we haven't improved the log messages in the mean time, what could we do better to make them more informative?

geissonator, 2020-12-04:
We have realized the InternalError got over used and provided pretty useless info. We have an anti-pattern for it in https://github.com/openbmc/docs/blob/master/anti-patterns.md#handling-unexpected-error-codes-and-exceptions and have been trying to either core dump (so we can debug it) or handle errors appropriately and provide more debug data in the journal/log.

amboar, 2020-12-07:
Slightly trollish suggestion, but what about making InternalError assert()? I imagine that'd encourage people to sort them out, if at the cost of a bit of up-front instability.

manojdesd, 2021-04-21:
I faced the same issue and the best thing you can do is, edit the places in code where internal failure is being called and define the problem eg : if <internal failure> being called due to image then image activation failed error.