openbmc/openbmc #3735: BMC fail to read ID flash after HOST boot

chnguyen-ampere, 2020-12-03:
Hi,
I'm using Linux kernel 5.4. I saw an issue relate to the read ID of the BIOS Flash (mx25l25635e) after the HOST boot.

After the HOST boot, I tried to unbind and bind the SPI flash driver again, but the driver fails during reading ID flash.
Here is the output:

1. BMC probe driver success in booting while the HOST hasn't turned ON

```
[    1.673278] aspeed-smc 1e620000.spi: Using 50 MHz SPI frequency
[    1.679396] aspeed-smc 1e620000.spi: mx66l51235f (65536 Kbytes)
[    1.685444] aspeed-smc 1e620000.spi: CE0 window [ 0x20000000 - 0x24000000 ] 64MB
[    1.692929] aspeed-smc 1e620000.spi: CE1 window [ 0x24000000 - 0x2a000000 ] 96MB
[    1.700341] aspeed-smc 1e620000.spi: read control register: 203b0241
[    1.845445] 5 fixed-partitions partitions found on MTD device bmc
[    1.851561] Creating 5 MTD partitions on "bmc":
[    1.856230] 0x000000000000-0x000000060000 : "u-boot"
[    1.863787] 0x000000060000-0x000000080000 : "u-boot-env"
[    1.871664] 0x000000080000-0x0000004c0000 : "kernel"
[    1.879247] 0x0000004c0000-0x000001c00000 : "rofs"
[    1.886686] 0x000001c00000-0x000002000000 : "rwfs"
[    1.895650] aspeed-smc 1e630000.spi: Using 50 MHz SPI frequency
[    1.901806] aspeed-smc 1e630000.spi: mx25l25635e (32768 Kbytes)
[    1.907856] aspeed-smc 1e630000.spi: CE0 window [ 0x30000000 - 0x32000000 ] 32MB
[    1.915346] aspeed-smc 1e630000.spi: CE1 window [ 0x32000000 - 0x38000000 ] 96MB
[    1.922756] aspeed-smc 1e630000.spi: read control register: 203c0241
```

2. After HOST on, try to unbind/bind the driver again. But, the driver can't probe because can't read ID Flash success

```
root@obmc:~# obmcutil chassison
root@obmc:~# obmcutil state
CurrentBMCState     : xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState   : xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState    : xyz.openbmc_project.State.Host.HostState.Running
BootProgress        : xyz.openbmc_project.State.Boot.Progress.ProgressStages.Unspecified
OperatingSystemState: xyz.openbmc_project.State.OperatingSystem.Status.OSStatus.Inactive

root@obmc:~# obmcutil chassisoff
root@obmc:~# obmcutil state
CurrentBMCState     : xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState   : xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState    : xyz.openbmc_project.State.Host.HostState.Off
BootProgress        : xyz.openbmc_project.State.Boot.Progress.ProgressStages.Unspecified
OperatingSystemState: xyz.openbmc_project.State.OperatingSystem.Status.OSStatus.Inactive

root@obmc:~# gpioset 0 226=0
root@obmc:~# echo 1e630000.spi > /sys/bus/platform/drivers/aspeed-smc/unbind
root@obmc:~# echo 1e630000.spi > /sys/bus/platform/drivers/aspeed-smc/bind

[  294.723293] aspeed-smc 1e630000.spi: Using 50 MHz SPI frequency
[  294.729342] aspeed-smc 1e630000.spi: unrecognized JEDEC id bytes: e0 00 0c e0 00 0c
[  294.737155] aspeed-smc 1e630000.spi: Aspeed SMC probe failed -2
[  294.761962] aspeed-smc: probe of 1e630000.spi failed with error -2
-sh: echo: write error: No such device

```

3. After that, I reset the SPI clock frequency selection (t-CK) bit in CE0 Control Register (0x1e630010) to the default init value. Then, I can bind the SPI driver.

```
root@obmc:~# devmem 0x1e630010
0x203C0641
root@obmc:~# devmem 0x1e630010 32 0x203C0241
root@obmc:~# echo 1e630000.spi > /sys/bus/platform/drivers/aspeed-smc/bind
[  273.092538] aspeed-smc 1e630000.spi: Using 50 MHz SPI frequency
[  273.098700] aspeed-smc 1e630000.spi: mx25l25635e (32768 Kbytes)
[  273.104796] aspeed-smc 1e630000.spi: CE0 window [ 0x30000000 - 0x32000000 ] 32MB
[  273.112305] aspeed-smc 1e630000.spi: CE1 window [ 0x32000000 - 0x38000000 ] 96MB
[  273.119719] aspeed-smc 1e630000.spi: read control register: 203c0241

```
I know the driver will optimize reading flash with the best clock.
Once the driver probes again, it will inherit the previous setting (optimize reading), and I saw the ReadID can't read success.

I wonder if it's due to running the read ID flash too fast?

Note: 
1. Of course, I handled the GPIO to control an external bus switch to select the SPI BIOS flash, which is directed to the Host system or BMC SPI controller interface.

chnguyen-ampere, 2020-12-03:
Hi @shenki, @AlexanderAmelkin

Do you have any idea about this issue? It seems to relate to the issue #3358

Thank you!

AlexanderAmelkin, 2020-12-03:
@chnguyen-ampere, what is the machine you're using? #3358 was related to P8 only.
How the host flash behaves in OpenBMC heavily depends on the machine architecture and board schematics.

williamspatrick, 2020-12-03:
@chnguyen-ampere Many systems have a mux that controls if the flash is pointed to the host or the BMC.  Does your BMC have direct control over this mux or is it driven automatically from the power sequence in your system?  Are you sure the mux is in the correct direction when you are doing all of your tests?

chnguyen-ampere, 2020-12-03:
> @chnguyen-ampere Many systems have a mux that controls if the flash is pointed to the host or the BMC. Does your BMC have direct control over this mux or is it driven automatically from the power [...]

Yes, @williamspatrick , I'm sure the mux pin was set correctly in each of my tests.
IF I don't switch the mux, then the driver will read ID with the value is 00, 00, 00, 00, 00, 00
In my case, the ReadID can return the value but this value is wrong (e0 00 0c e0 00 0c)

AlexanderAmelkin, 2020-12-03:
Good, but still, what is your shematics related to host flash? How is it connected? What is the host type (machine) ?

chnguyen-ampere, 2020-12-04:
hi @AlexanderAmelkin,

This is block diagram of the spi flash in my system. The type of host machine is arm64
[image]

closed 2021-07-11 by stale[bot]

mdmillerii, 2021-07-12:
You gave the incorrect value of readid but not the expected value to compare.   The optimization is tested by reading data from flash with different delay values until it changes, so maybe the device has a different response delay for id or the content of the flash doesn't toggle bits that would notice the failure during tuning.