openbmc/openbmc #3736: Kernel panic in find_inode_fast function

JohnChung1993, 2020-12-03:
**Unexpected behavior you saw**
BMC Kernel panic when BMC booting.

```
Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 ttyS4
[   61.834837] 8<--- cut here ---
[   61.837937] Unable to handle kernel paging request at virtual address 20666eb0
[   61.845300] pgd = a7322dfc
[   61.848030] [20666eb0] *pgd=00000000
[   61.851628] Internal error: Oops: 5 [#1] ARM
[   61.855921] CPU: 0 PID: 172 Comm: python Not tainted 5.4.23-6ba7086-dirty-9737e6c #1
[   61.863659] Hardware name: Generic DT based system
[   61.868467] PC is at find_inode_fast+0x34/0xbc
[   61.872919] LR is at iget_locked+0x64/0x168
[   61.877103] pc : [<8026a000>]    lr : [<8026b9bc>]    psr: 20000013
[   61.883357] sp : 91655b68  ip : 91655b88  fp : 91655b84
[   61.888574] r10: 9ebb9dfc  r9 : 9ebaf000  r8 : 806f6bd0
[   61.893793] r7 : 90e5a400  r6 : 90e5a400  r5 : 9ebb9dfc  r4 : 00000b25
[   61.900311] r3 : 65646f6e  r2 : 00000b25  r1 : 9ebb9dfc  r0 : 20666e8c
[   61.906829] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
[   61.913952] Control: 00c5387d  Table: 916bc008  DAC: 00000051
[   61.919696] Process python (pid: 172, stack limit = 0x918220c4)
[   61.925611] Stack: (0x91655b68 to 0x91656000)
[   61.929977] 5b60:                   6d64085f 00002b7f 00000b25 90e5a400 91655bbc 91655b88
[   61.938148] 5b80: 8026b9bc 80269fd8 90e5a400 802cc87c 91655bdc 6d64085f 00000000 6d64085f
[   61.946322] 5ba0: 9262ba00 9229b6d8 9262ba08 91bb64f4 91655bdc 91655bc0 802cebac 8026b964
[   61.954493] 5bc0: 6d64085f 00000000 00000b25 9262ba00 91655c5c 91655be0 802cefb8 802ceb94
[   61.962666] 5be0: 00000b25 91bb64c8 91655c5c 91655bf8 90e04e80 90e04e80 91bb64c8 91bb64f4
[   61.970838] 5c00: 9258ad0d 00000000 91655c20 0000001c 00001812 7cff5f25 01d3a07f 00000000
[   61.979010] 5c20: 00000030 00006d64 00000b1b 80a03008 ffffffff 91bb64c8 92296b28 00000000
[   61.987183] 5c40: 91655cc8 9229b6d8 91655c6c 00000000 91655ca4 91655c60 8025a0cc 802cec30
[   61.995357] 5c60: 91655cc8 91655cc8 92296b28 91655c6c 91655c6c 80a03008 90f3c4e0 9229b754
[   62.003529] 5c80: 92296b28 91655cc8 00000000 90f3c940 90f3c4e0 9267cc40 91655cc4 91655ca8
[   62.011702] 5ca0: 8025a1c8 8025a044 92296b28 91655db0 00000000 807ec11c 91655cf4 91655cc8
[   62.019875] 5cc0: 8025ccac 8025a194 4f1123ec 0000001c 91bb602c 91655ce0 80546314 80a03008
[   62.028047] 5ce0: 9d411a20 807ec11c 91655d14 91655cf8 80322550 8025cc38 807ec11c 91655db0
[   62.036220] 5d00: 00000001 9d60ba40 91655d5c 91655d18 803228ac 80322528 00000000 807ec11c
[   62.044393] 5d20: 91655d90 8024417c 91655d90 92296b28 80a14ca0 00000000 91fe4260 00000001
[   62.052566] 5d40: 9d60ba40 90f3c940 90f3c4e0 9267cc40 91655e0c 91655d60 80323b88 8032277c
[   62.060737] 5d60: 80268070 91655ea8 bf4e4b52 00000000 90f4f400 91bb6000 00000000 90f3c4e0
[   62.068909] 5d80: 00000000 00000000 00000000 00000000 80a81004 91bb6000 91655e0c 91655da8
[   62.077084] 5da0: 802689bc 802684d8 91655dcc 91655e64 90f4dc00 8025965c bf4e4b52 0000001c
[   62.085256] 5dc0: 91bb602c 75d6c5e4 01000000 00000000 80a81000 91655e5c 9267dd40 80a03008
[   62.093429] 5de0: 91655e0c 91655ea0 00000000 91655f58 92296660 91bb6000 9267dd40 00020000
[   62.101601] 5e00: 91655e9c 91655e10 8025db5c 803237a8 80a4de30 00000001 91654000 91655ea8
[   62.109774] 5e20: 00000000 92292e20 00000000 92296660 00000000 00000041 00020000 00000000
[   62.117946] 5e40: 00000000 000041ed 00000000 00000004 ffffffff 91655ebc 90ff700c 91654000
[   62.126119] 5e60: 91655e84 91655e64 91655e64 80a03008 803bef9c 00000009 91655f58 91655ea0
[   62.134293] 5e80: 00000001 801011e4 91654000 00000000 91655f4c 91655ea0 8025e59c 8025d408
[   62.142464] 5ea0: 90e1a550 92296660 bf4e4b52 0000001c 90ff7036 91655f0c 90e1a550 9d1fb990
[   62.150637] 5ec0: 92292e20 00000109 00000002 0000062a 00000000 00000000 00000000 91655ee0
[   62.158810] 5ee0: 91655f94 80a03008 00000009 91761d40 00000000 00000400 00020000 916c43c0
[   62.166983] 5f00: 91655f3c 91655f10 8026e380 8026da98 90ff7000 00000000 ffffff9c 00000002
[   62.175156] 5f20: ffffff9c 801011e4 91654000 80a03008 00000009 ffffff9c 90ff7000 fffff000
[   62.183327] 5f40: 91655f94 91655f50 8024c114 8025e524 5fa0dbd2 00000000 00020000 00000000
[   62.191499] 5f60: 00000004 00000100 00000001 80a03008 01df3728 76f9b4d0 00000008 00000142
[   62.199675] 5f80: 801011e4 91654000 91655fa4 91655f98 8024c20c 8024bfb4 00000000 91655fa8
[   62.207846] 5fa0: 80101000 8024c1fc 01df3728 76f9b4d0 ffffff9c 01eb03d8 00020000 00000000
[   62.216020] 5fc0: 01df3728 76f9b4d0 00000008 00000142 01eae3c8 462c4000 00001001 5fa0dbd2
[   62.224191] 5fe0: 00000000 7ef007d0 45f3cfdc 45f95af4 60000010 ffffff9c 00000000 00000000
[   62.232351] Backtrace: 
[   62.234825] [<80269fcc>] (find_inode_fast) from [<8026b9bc>] (iget_locked+0x64/0x168)
[   62.242653]  r7:90e5a400 r6:00000b25 r5:00002b7f r4:6d64085f
[   62.248329] [<8026b958>] (iget_locked) from [<802cebac>] (squashfs_iget+0x24/0x9c)
[   62.255909]  r10:91bb64f4 r9:9262ba08 r8:9229b6d8 r7:9262ba00 r6:6d64085f r5:00000000
[   62.263729]  r4:6d64085f
[   62.266276] [<802ceb88>] (squashfs_iget) from [<802cefb8>] (squashfs_lookup+0x394/0x4cc)
[   62.274363]  r7:9262ba00 r6:00000b25 r5:00000000 r4:6d64085f
[   62.280035] [<802cec24>] (squashfs_lookup) from [<8025a0cc>] (__lookup_slow+0x94/0x150)
[   62.288043]  r10:00000000 r9:91655c6c r8:9229b6d8 r7:91655cc8 r6:00000000 r5:92296b28
[... 50 lines cut ...]
(gdb) list *(0x8025ccac)
0x8025ccac is in lookup_one_len_unlocked (/usr/src/kernel/fs/namei.c:2563).
2558    in /usr/src/kernel/fs/namei.c
(gdb) list *(0x803228ac)
0x803228ac is in ovl_lookup_layer (/usr/src/kernel/fs/overlayfs/namei.c:289).
284     /usr/src/kernel/fs/overlayfs/namei.c: No such file or directory.
(gdb) list *(0x8025db5c)
0x8025db5c is in path_openat (/usr/src/kernel/fs/namei.c:3204).
3199    /usr/src/kernel/fs/namei.c: No such file or directory.
(gdb) list *(0x8025e59c)
0x8025e59c is in do_filp_open (/usr/src/kernel/fs/namei.c:3558).
3553    in /usr/src/kernel/fs/namei.c
(gdb) list *(0x8024c114)
0x8024c114 is in do_sys_open (/usr/src/kernel/fs/open.c:1097).
1092    /usr/src/kernel/fs/open.c: No such file or directory.
(gdb) list *(0x8024c20c)
0x8024c20c is in __se_sys_openat (/usr/src/kernel/fs/open.c:1118).
1113    in /usr/src/kernel/fs/open.c
(gdb) list *(0x80269fcc)
0x80269fcc is in find_inode_fast (/usr/src/kernel/fs/inode.c:847).
842     /usr/src/kernel/fs/inode.c: No such file or directory.
```

Any suggestion for this issue?
Thanks.

229024527, 2020-12-05:
Can you tell me how to debug in openbmc ?

mdmillerii, 2020-12-06:
It's always a bug to panic and not return an error but you can check if the squashfs image matches the source and was not overwritten by other content.

JohnChung1993, 2020-12-14:
Thanks.
Update to latest kernel version, this issue seems fix.

closed 2020-12-14 by JohnChung1993

JohnChung1993, 2020-12-14:
> Can you tell me how to debug in openbmc ?

For this issue, I copy the kernel panic address and using GDB to vmlinux file in build folder.

Following is an example.
Using list command with panic address in GDB console.

> gdb vmlinux-5.8.17-xxxxxxx-dirty-eb42a1c
>
> (gdb) list *(0x802cebac) 
> 0x802cebac is in squashfs_iget (/usr/src/kernel/fs/squashfs/inode.c:75).