openbmc/openbmc #3738: [meta-phosphor] meta-phosphor dunfell branch is not compatible with poky dunfell branch

sathishvslabs, 2020-12-05:
**Unexpected behavior you saw**
Building meta-phosphor with dunfell branch with corresponding poky source gives build error in bmcweb package:

`Could NOT find Boost: Found unsuitable version "1.72.0", but required is at least "1.73"`

**Expected behavior**
Build should complete properly without any error

**To Reproduce**
Steps to reproduce the behavior:
1. checkout meta-phosphor (dunfell), meta-aspeed (master), poky (dunfell), meta-openembedded (dunfell)
2. Set env with `TEMPLATECONF=meta-evb/meta-evb-aspeed/meta-evb-ast2500/conf . oe-init-build-env`
3. `bitbake obmc-phosphor-image`
4. Observe build failure

**Additional context**

- I am checking out meta-aspeed with master since there is no official branch for dunfell. But the layer compatibility as of 05/12/2020 in master had both dunfell and gatesgarth.

- Changing the bmcweb's SRCREV from local.conf  to `SRCREV_pn-bmcweb = "67a78d8758859a4e4477de932d24fac5fb287dad"` made the build work and able to test the build as well.

williamspatrick, 2020-12-05:
@sathishvslabs - we don’t currently support checking out the meta layers independently because we’ve never had someone propose a use case for it.  We have a real dunfell branch in this repository, which I just updated to the latest poky/dunfell content this week. Are you able to use that repository / branch instead for your intended use?  Are you seeing a similar compile issue there?

sathishvslabs, 2020-12-05:
@williamspatrick We are working with AST2600 BMC. We already have a working Yocto model using **kas** build system in our build environment. So we are pushing to use that for better integration with rest of the system. 

For make the dunfell branch properly work, We have added the boost 0.73.0 (back ported from poky with patches) support in our custom layer to solve this problem.

I don't see a dunfell branch in openbmc/openbmc repository. Can you share us the link for the dunfell branch of openbmc, So that we can test that as a single repository with collection of layers?

geissonator, 2020-12-05:
Whoops, the dunfell branch was not synch'd over from gerrit into github. I just did it so now available at https://github.com/openbmc/openbmc/tree/dunfell

sathishvslabs, 2020-12-05:
Thank you @geissonator I will give it a try and update.

sathishvslabs, 2020-12-05:
I have tried building the dunfell branch of the openbmc and it successfully built without any error. The discrepancy is that meta-phosphor in openbmc dunfell branch and meta-phosphor in its own repository's dunfell branch are not in sync. So openbmc dunfell build works.

I think I can use openbmc dunfell repo and select only the required layers in our kas script for our build.

sathishvslabs, 2020-12-08:
using dunfell branch of openbmc for the purpose. Closing the issue.

closed 2020-12-08 by sathishvslabs

bradbishop, 2020-12-08:
> Whoops, the dunfell branch was not synch'd over from gerrit into github. I just did it so now available at https://github.com/openbmc/openbmc/tree/dunfell

I just deleted the dunfell branches from gerrit/github for meta-phosphor.  They seemed to be pointing off into the weeds anyway.