openbmc/openbmc #3740: Openbmc debug with gdb ... is too large!

229024527, 2020-12-09:
Hello who can help me!
I want't debug pflash with gdb i changed follow file

openbmc/build/conf/local.conf
EXTRA_IMAGE_FEATURES = "dbg-pkgs"

openbmc/meta-ibm/meta-romulus/recipes-phosphor/image/obmc-phosphor-image.bbappend
OBMC_IMAGE_EXTRA_INSTALL_append_palmetto = " mboxd liberation-fonts uart-render-controller pflash gdb"

when i bitbake obmc-phosphor-image
obmc-phosphor-image-romulus.squashfs-xz is too large! appear 
I try to chage meta-phosphor/classes/image_types_phosphor.bbclass FLASH_SIZE=131072 but cannot solve this problem, who can tell me how to debug with other way or solve this problem，Thanks！

leiyu-bytedance, 2020-12-09:
This is because the image size is limited and adding the new package causes the image exceeds the limit.

The gdb is in the the debug package. Suggest to dynamically load the debug package into the BMC in RAM (e.g. tmpfs), so that it won't impact the BMC's flash.

The debug package is built by the CI: https://jenkins.openbmc.org/job/ci-openbmc/distro=ubuntu,label=docker-builder,target=romulus/lastSuccessfulBuild/artifact/openbmc/build/tmp/deploy/images/romulus/obmc-phosphor-debug-tarball-romulus.tar.xz

229024527, 2020-12-11:
Thank you very much，it works~

closed 2020-12-11 by 229024527

lingyaof, 2023-09-04:
Hello all
When I wanted to add **Valgrand** to my project, I found the same problem.  squashfs-xz is too large
Could you tell me how can solve this problem.Thanks!

amboar, 2023-09-04:
You could also try a workflow like described here:

https://amboar.github.io/notes/2022/05/13/opkg-based-development-workflow.html

Kostr, 2023-09-04:
Another solution is to use NFS

lingyaof, 2023-09-13:
Thanks ! 
I only need the memcheck tool, so I deleted unnecessary files in bb. Now it works. 
Thank you for your suggestions!!!