openbmc/openbmc #3741: Implement new interface to update the firmware

chnguyen-ampere, 2020-12-15:
Hi all,

Currently, the phosphor-bmc-code-mgmt supported openbmc update the BMC and BIOS firmware. But, now I would like to have an interface to update the System Control Processor (SCP) Boostrap Firmware into the EEPROM chip.
My system is designed as below :
[image]

Follow the design for BIOS updating with option host-bios-upgrade in phosphor-bmc-code-mgmt. I don't know it's good if I will refer that to implement the SCP updating? 
**Ex**: I will define the host-scp-upgrade option to implement the VersionPurpose == "SCP"

Anyone help me raise your ideas, please!

Thanks!

mdmillerii, 2020-12-15:
You will likely need to add hooks to do this.   Also do you need to send commands to pause your host while you access the bootstrap eeprom and maybe start afterwards?   I believe the phosphor code management repository can be extended. openbmc/phosphor-bmc-code-mgmt. I'd suggest starting by creating a pull request to add a script that applies the firmware with host stop start like the apply for the other images and the review can discuss further changes to call for this image type.

williamspatrick, 2020-12-15:
@chnguyen-ampere , I work at Facebook and we have a team at HCL working on porting openbmc to one of our system designs: yosemite-v2.  They have been similarly exploring how to do firmware update on the CPLD and another micro-controller on our CPU cards called the BIC.  You might find some posts on the mailing list and in Gerrit on the Docs repository that have some overlap with what you're also working on.

> VersionPurpose == "SCP"

I also maintain the phosphor-dbus-interfaces repository, which is where the VersionPurpose enumeration comes from.  We had some discussion with Intel when they were pursuing update the the ME.  I have said there that we're going to stop adding new hardware devices to this enumeration list and instead move to a schema on ExtendedVersion that allows update software to determine compatibility between an image and the managed hardware.  It seems like there is some need for me to define that schema soon...

chnguyen-ampere, 2020-12-16:
> firmware update on the CPLD

epa

chnguyen-ampere, 2020-12-16:
Hi @mdmillerii ,

Yes, the script will handle the host state while accessing the bootstrap eeprom.

I will folow https://github.com/openbmc/docs/blob/dcc8886b3cb9838f6b612947c5ba7940b2044812/designs/firmware-update-over-redfish.md to extend the openbmc/phosphor-bmc-code-mgmt support update SCP-Boostrap Firmware into the EEPROM chip via REST API.

Then, I will create a pull request to openbmc/phosphor-bmc-code-mgmt and phosphor-dbus-interfaces.

Hi @williamspatrick ,

I don't see the CPLD firmware update via REST in https://github.com/facebook/openbmc . I just see update the CPLD FW was updated by some tool as cpldupdate_poc tool.

williamspatrick, 2020-12-16:
The work I mentioned is being done in meta-facebook under this repository, but much of it is open pending review in Gerrit. 

You should not use facebook/openbmc as an example unless you really know what you are looking for. I know it is confusing but that repository is an entirely different codebase and not in any way related to this repository.

chnguyen-ampere, 2020-12-17:
Ok @williamspatrick, I got it! 
I'll follow the PSU firmware update design (https://github.com/openbmc/docs/blob/master/designs/psu-firmware-update.md ) to design and implement for my SCP FW update feature. I also try to create and push review the design document to openbmc/doc gerrit server.

chnguyen-ampere, 2020-12-23:
Hi,

I think we shouldn't add more enum to VersionPurpose (phosphor-dbus-interfaces repository). Refer to the discussion at https://gerrit.openbmc-project.xyz/c/openbmc/phosphor-dbus-interfaces/+/29004 , I think to use the VersionPurpose = "Other" and combine with the ExtendedVersion for a specific purpose will be good.

@williamspatrick, Have we had any plan to handle the ExtendedVersion object from MANIFEST in the phosphor-bmc-code-mgmt? On the openpower-pnor-code-mgmt, I saw an implementation for the ExtendedVersion ( https://github.com/openbmc/openpower-pnor-code-mgmt/commit/7254f0ea592464c96181cef2a4b86de52d8ce888 )

Sorry @williamspatrick, I tried to find some discussion on the mailing list and in Gerrit on the Docs repository about CPLD fw update or BIC, BUT not found. Please help point me to this discussion that you mentioned!

I only see the plan for VR upgrade from FB team (https://lists.ozlabs.org/pipermail/openbmc/2020-September/022815.html). But, it seems in a suspension.

Thanks!

chnguyen-ampere, 2021-01-11:
Hi @vijaykhemka,
I hope you can share with me the information or progress on the "VR upgrade" (https://lists.ozlabs.org/pipermail/openbmc/2020-September/022815.html) . Thank you !

closed 2021-09-06 by stale[bot]