openbmc/openbmc #3742: ast2400 kcs can not set bootdev

229024527, 2020-12-17:
It's my config file for kcs
1. openbmc/build/workspace/sources/linux-aspeed/arch/arm/boot/dts/aspeed-bmc-opp-palmetto.dts
&kcs3 {
        // BMC KCS channel 3
        status = "okay";
        kcs_addr = <0xca2>;
};

2. openbmc/meta-ibm/meta-palmetto/conf/machine/palmetto.conf
PREFERRED_PROVIDER_virtual/obmc-host-ipmi-hw = "phosphor-ipmi-kcs"

when i use ipmitool chassis bootdev pxe it response failed to send dbus message

[image]

when i use journctl follow error were appear

Jan 01 00:01:18 palmetto systemd[1]: Started Process Core Dump (PID 309/UID 0).
Jan 01 00:01:19 palmetto phosphor-chassis-state-manager[241]: Change to Chassis Power State
Jan 01 00:01:23 palmetto phosphor-mapper[189]: Introspect call failed with error: generic:113, No route to host on process: xyz.openbmc_project.Control.Host path: /org/open_power
Jan 01 00:01:23 palmetto systemd[1]: phosphor-ipmi-host.service: Main process exited, code=dumped, status=6/ABRT
Jan 01 00:01:23 palmetto phosphor-mapper[189]: Introspect call failed with error: generic:113, No route to host on process: xyz.openbmc_project.Control.Host path: /xyz
Jan 01 00:01:23 palmetto phosphor-mapper[189]: Introspect call failed with error: generic:113, No route to host on process: xyz.openbmc_project.Ipmi.Host path: /xyz
Jan 01 00:01:23 palmetto systemd[1]: phosphor-ipmi-host.service: Failed with result 'core-dump'.
Jan 01 00:01:24 palmetto systemd[1]: phosphor-ipmi-host.service: Service RestartSec=1s expired, scheduling restart.
Jan 01 00:01:24 palmetto systemd[1]: phosphor-ipmi-host.service: Scheduled restart job, restart counter is at 2.
Jan 01 00:01:24 palmetto systemd[1]: Stopping Phosphor MBOX Daemon...
Jan 01 00:01:24 palmetto systemd[1]: Stopping Phosphor IPMI KCS DBus Bridge...
Jan 01 00:01:24 palmetto systemd[1]: Stopping Network IPMI daemon...
Jan 01 00:01:24 palmetto systemd[1]: mboxd.service: Succeeded.
Jan 01 00:01:24 palmetto systemd[1]: Stopped Phosphor MBOX Daemon.
Jan 01 00:01:24 palmetto systemd[1]: phosphor-ipmi-kcs@ipmi-kcs3.service: Succeeded.
Jan 01 00:01:24 palmetto systemd[1]: Stopped Phosphor IPMI KCS DBus Bridge.
Jan 01 00:01:24 palmetto systemd[1]: phosphor-ipmi-net@eth0.service: Succeeded.
Jan 01 00:01:24 palmetto systemd[1]: Stopped Network IPMI daemon.
Jan 01 00:01:24 palmetto systemd[1]: Stopped Phosphor Inband IPMI.

mine260309, 2020-12-20:
`palmetto` is an OpenPOWER machine and thus does not use ipmi kcs. It uses ipmi-bt instead.

229024527, 2021-01-13:
> `palmetto` is an OpenPOWER machine and thus does not use ipmi kcs. It uses ipmi-bt instead.

Thanks very much，i use ipmi bt it works

closed 2021-01-13 by 229024527