openbmc/openbmc #3744: obmc-init.sh possible unlimited while

marthauk, 2021-01-06:
https://github.com/openbmc/openbmc/blob/master/meta-phosphor/recipes-phosphor/initrdscripts/files/obmc-init.sh#L144
Is there a probability of this takeover to fail and it becoming stuck here? Should there be a timeout for this while?

williamspatrick, 2021-01-06:
@mdmillerii ^

mdmillerii, 2021-01-06:
At this point in the script, we have detected that something had gone fatally wrong or a stop for explicit engineering action was requested.  In addition we know that the interactive shell was explicitly requested (see line 132).   

The while is there to make sure the user explicitly exits the shell and it wasn't accidentally killed by a signal.  Note that the `sh` invocation being checked is an interactive shell.

In normal deployment I would have expected the script to exit at line 135 and the kernel to panic.

For these reasons I don't see a need for a change.

closed 2021-09-06 by stale[bot]