openbmc/openbmc #3745: How to update the BMC firmware to Alternate SPI BMC chip

chnguyen-ampere, 2021-01-07:
Hi all,

Currently, the phosphor-bmc-code-mgmt supported openbmc update the BMC. But, now I would like to have an interface to update the Alternate SPI BMC chip.

My system is designed as below :

[image]

I suggest a solution. If the purpose is alternate bmc updating, I will put the images ("image-kernel", "image-rofs","image-rwfs", "image-u-boot") in /run/initramfs and rename it by inserted the "alt" string ("image-alt-kernel", "image-alt-rofs","image-alt-rwfs", "image-alt-u-boot"). Please see the patch for phosphor-bmc-code-mgmt in below:

[Support_flash_Alternate_BMC_via_Rest_patch.log](https://github.com/openbmc/openbmc/files/5780624/Support_flash_Alternate_BMC_via_Rest_patch.log)

So, When trigger a reboot and the **updater** script (in /run/initramfs) will program the image to alt BMC partition ("alt-kernel", "alt-rofs","alt-rwfs", "alt-u-boot")

Anyone help me raise your ideas, please!

Thanks!

mdmillerii, 2021-01-07:
The witherspoon machine has dual spi flash.  It is using the ubi layout instead of direct MTD partitions.  

In the ubi scheme the alternate image is updated while the current image is running then the u-boot environment is updated to point to the other image.   There is a redundant copy of u-boot and kernel on the alternate flash and the kernel will mount the ubifs image based on the u-boot environment variable inserted into the kernel command line.

To use ubi the kernel has to be compiled with 4k flash blocks.

chnguyen-ampere, 2021-01-08:
Thank @mdmillerii ! What's the advantage of the UBI layout with the MTD direction?

My machine is using the default MTD partition for the BMC SPI layout

leiyu-bytedance, 2021-01-08:
I have a proposal in https://gerrit.openbmc-project.xyz/c/openbmc/docs/+/36407. (Be noted that it's not implemented yet)
@chnguyen-ampere maybe you could check that design and see if it fits your use case.

mdmillerii, 2021-01-08:
Pros for UBI and ubifs:
1.  The storage is in one pool as oposed to fixed volumes.  This means that the space occupied by the kernel can be traded for space occupied by the (read-only base) filesystem and you won't run out of space in the allocated system.
2.  Because the erase blocks are logical (mapped), the whole area occupied by ubi is used for wear leveling the writable file system (`/var` stored in ubifs).  This means any logging is spread over most of the flash instead of the fixed mtd partiton
3.  The layout boots the system without using an initramfs.   (This might be available to raw partitions if the file system layout is changed to not be a full overlay as I have discussed elsehwere).

Cons:
1.  There is a 2 write block  overhead per erase block (512 bytes per 4k for the devces we selected)
2.  The kernel has to use 4k erase blocks.  This is a kernel config item and is not selectable per device or partiton (might be able to take up with upstream).

Unknown:
1. jffs2 has two built-in compressors ubifs seemed to only have one.
2. jffs2 builds all metadata structures in ram during mount.  ubi reads all erase blocks in ram at boot but the fs data has normal.
3. jffs2 adds a link then removes the old entry on rename, while ubifs uses atomic transactions.

I didn't have this list written down and is what I can recall within a few hours.   The flexible space was the deciding factor for deployment.

bradbishop, 2021-01-11:
> On Jan 7, 2021, at 4:19 AM, Chanh Nguyen <notifications@github.com> wrote:
> 
> 
> Hi all,
> 
> Currently, the phosphor-bmc-code-mgmt supported openbmc update the BMC. But, now I would like to have an interface to update the Alternate SPI BMC chip.
> 
> My system is designed as below :
> 
> 
> 
> I suggest a solution. If the purpose is alternate bmc updating, I will put the images ("image-kernel", "image-rofs","image-rwfs", "image-u-boot") in /run/initramfs and rename it by inserted the "alt" string ("image-alt-kernel", "image-alt-rofs","image-alt-rwfs", "image-alt-u-boot"). Please see the patch for phosphor-bmc-code-mgmt in below:
> 
> Support_flash_Alternate_BMC_via_Rest_patch.log
> 
> So, When trigger a reboot and the updater script (in /run/initramfs) will program the image to alt BMC partition ("alt-kernel", "alt-rofs","alt-rwfs", "alt-u-boot")
> 
> Anyone help me raise your ideas, please!
> 
> Thanks!

The issue tracker is for reporting bugs.  Please send your idea to the mailing list.  Thanks!

closed 2021-09-06 by stale[bot]