openbmc/openbmc #3747: Unable to create new system in meta-amd

jdhanasekar, 2021-01-11:
Hi All,

I am trying to create new system in meta-amd. I have followed instruction in this link,```https://gist.github.com/geissonator/77d4f6c6ed17bb0693b4251137f77fb4```.
Cloned ```meta-ethanolx``` package and renamed it as  ```meta-daytona```.

After building AMD Ethanolx successfully, made the below changes in ```meta-daytona```,

I have made change like this,

In ```bblayers.conf.sample```,
[image]

In ```layer.conf```,
[image]

In ```local.conf.sample```,
[image]

Changed ```meta-amd/meta-daytona/machine/ethonalx.conf``` to ```meta-amd/meta-daytona/machine/daytona.conf```.

In ```daytona.conf```, made changes like this,
[image]

**Command executed**

1. ```rm -rf ./build/conf```
2. ```export TEMPLATECONF=meta-amd/meta-daytona/conf```
3.  ```. openbmc-env```
4.  ```bitbake obmc-phosphor-image```

And getting the below error,
[image]

Even though Ethanolx conf files are deleted form build folder, bitbake is still loading ethanolx for machine, instead of daytona. 

Please help me to fix this issue.
If there are mistakes in my experiments, Please correct me. 

Thanks,
Dhanasekar

bradbishop, 2021-01-11:
> On Jan 11, 2021, at 5:31 AM, jdhanasekar <notifications@github.com> wrote:
> 
> 
> Hi All,
> 
> I am trying to create new system in meta-amd. I have followed instruction in this link,https://gist.github.com/geissonator/77d4f6c6ed17bb0693b4251137f77fb4.
> Cloned meta-ethanolx package and renamed it as meta-daytona.
> 
> After building AMD Ethanolx successfully, made the below changes in meta-daytona,
> 
> I have made change like this,
> 
> In bblayers.conf.sample,
> 
> 
> In layer.conf,
> 
> 
> In local.conf.sample,
> 
> 
> Changed meta-amd/meta-daytona/machine/ethonalx.conf to meta-amd/meta-daytona/machine/daytona.conf.
> 
> In daytona.conf, made changes like this,
> 
> 
> Command executed
> 
> 	• rm -rf ./build/conf
> 	• export TEMPLATECONF=meta-amd/meta-daytona/conf
> 	• . openbmc-env
> 	• bitbake obmc-phosphor-image
> And getting the below error,
> 
> 
> Even though Ethanolx conf files are deleted form build folder, bitbake is still loading ethanolx for machine, instead of daytona.
> 
> Please help me to fix this issue.
> If there are mistakes in my experiments, Please correct me.
> 
> Thanks,
> Dhanasekar

You need to rm -rf ./build/tmp as well in this situation.

jdhanasekar, 2021-01-11:
After removing of ./build/tmp, getting same error.

[image]

williamspatrick, 2021-01-11:
> Please set a valid MACHINE in your local.conf

Any of the `foo.conf.sample` files are used to create corresponding ones in your build directory when you first start bitbake on an empty build directory.  Under `conf` you should see `conf/bblayers.conf` and `conf/local.conf`.  Those need to be updated with similar changes to what you made to the `local.conf.sample`, etc.

jdhanasekar, 2021-01-12:
I have removed ```.build/conf``` and ```.build/tmp```
when I have compared ```meta-amd/meta-daytona/conf/bblayers.conf.sample``` with ```build/conf/bblayers.conf``` and ```meta-amd/meta-daytona/conf/local.conf.sample``` with ```build/conf/local.conf```. 
I am observing meta-daytona's conf files are identical with build/conf/ files. 

[image]

**local.conf.sample** vs **local.conf**

[image]

williamspatrick, 2021-01-12:
Can you push up a diff to Gerrit?  Even if there is no intention on merging, it gives others something that is identical to what you have to help debug.  There is probably something small you missed in the conversion but it is going to be hard to identify without seeing exactly what you have done.

jdhanasekar, 2021-01-13:
Hi William,

Before uploading diff to Gerrit. I had given a try. 

I have followed the steps mentioned in ```https://gist.github.com/geissonator/77d4f6c6ed17bb0693b4251137f77fb4.```

This time, I have used ```bitbake -f -c compile u-boot``` instead of ```bitbake obmc-phosphor-image```.

I am getting below error. 
[image]

jdhanasekar, 2021-01-25:
Hi Willam,

I am facing issue when pushing the changes..

[image]

williamspatrick, 2021-01-25:
Your gerrit address does not appear correct.  ‘velankanigroup.com’?

jdhanasekar, 2021-01-25:
Yes...Actually, what should come in that place. ??
I have to use ```github.com```??.
[image]

williamspatrick, 2021-01-25:
Please see: https://github.com/openbmc/docs/blob/master/CONTRIBUTING.md#submitting-changes-via-gerrit-server-to-openbmc

You likely could replace with `gerrit.openbmc-project.xyz` but the recommendation is to make an alias in your .sshconfig (see the link).

jdhanasekar, 2021-01-26:
I am getting below error while pushing,

[image]

williamspatrick, 2021-01-26:
That is implying you already pushed it to Gerrit and do not have any changes. Did you commit these changes?  Maybe you already pushed?  If these changes are committed you could “git commit —amend” and it will create a new commit number so Gerrit will accept. I suspect you never made a commit though... check “git status” to see if you have uncommitted changes.

williamspatrick, 2021-01-26:
Also, I’d prefer you don’t send screenshots. This is all text; just copy and paste it.

jdhanasekar, 2021-01-26:
I am very new to Gerrit Code review process.. I am getting below output for "git status",

```
user@quanta:~/dhanasekar/BMC/AMD/daytona1/openbmc$ git status
On branch master
Your branch is up to date with 'origin/master'.

Untracked files:
  (use "git add <file>..." to include in what will be committed)

	meta-amd/meta-daytona/

nothing added to commit but untracked files present (use "git add" to track)

```

williamspatrick, 2021-01-26:
So you have not added / committed any of your changes. There isn’t anything being sent to Gerrit when you are pushing because git only pushes commits.

jdhanasekar, 2021-01-26:
I have added my changes, below is the output,

```
user@quanta:~/dhanasekar/BMC/AMD/daytona1/openbmc$ git add meta-amd/meta-daytona/
user@quanta:~/dhanasekar/BMC/AMD/daytona1/openbmc$ git status
On branch master
Your branch is up to date with 'origin/master'.

Changes to be committed:
  (use "git reset HEAD <file>..." to unstage)

	new file:   meta-amd/meta-daytona/conf/bblayers.conf.sample
	new file:   meta-amd/meta-daytona/conf/conf-notes.txt
	new file:   meta-amd/meta-daytona/conf/layer.conf
	new file:   meta-amd/meta-daytona/conf/local.conf.sample
	new file:   meta-amd/meta-daytona/conf/machine/daytona.conf
	new file:   meta-amd/meta-daytona/recipes-kernel/linux/linux-aspeed/0001-ARM-dts-aspeed-Add-GPIO-line-names-for-AMD-EthanolX.patch
	new file:   meta-amd/meta-daytona/recipes-kernel/linux/linux-aspeed/ethanolx.cfg
	new file:   meta-amd/meta-daytona/recipes-kernel/linux/linux-aspeed_%.bbappend
	new file:   meta-amd/meta-daytona/recipes-phosphor/configuration/ethanolx-yaml-config.bb
	new file:   meta-amd/meta-daytona/recipes-phosphor/configuration/ethanolx-yaml-config/ethanolx-ipmi-fru-properties.yaml
	new file:   meta-amd/meta-daytona/recipes-phosphor/configuration/ethanolx-yaml-config/ethanolx-ipmi-fru.yaml
	new file:   meta-amd/meta-daytona/recipes-phosphor/configuration/ethanolx-yaml-config/ethanolx-ipmi-sensors.yaml
	new file:   meta-amd/meta-daytona/recipes-phosphor/ipmi/phosphor-ipmi-config.bbappend
	new file:   meta-amd/meta-daytona/recipes-phosphor/ipmi/phosphor-ipmi-config/channel_config.json
	new file:   meta-amd/meta-daytona/recipes-phosphor/ipmi/phosphor-ipmi-config/dcmi_sensors.json
	new file:   meta-amd/meta-daytona/recipes-phosphor/ipmi/phosphor-ipmi-fru_%.bbappend
	new file:   meta-amd/meta-daytona/recipes-phosphor/ipmi/phosphor-ipmi-host_%.bbappend
	new file:   meta-amd/meta-daytona/recipes-phosphor/leds/ethanolx-led-manager-config-native.bb
	new file:   meta-amd/meta-daytona/recipes-phosphor/leds/ethanolx-led-manager-config/led.yaml
	new file:   meta-amd/meta-daytona/recipes-phosphor/sensors/phosphor-hwmon/obmc/hwmon/ahb/apb/bus@1e78a000/i2c-bus@300/lm75a@48.conf
	new file:   meta-amd/meta-daytona/recipes-phosphor/sensors/phosphor-hwmon/obmc/hwmon/ahb/apb/bus@1e78a000/i2c-bus@300/lm75a@49.conf
	new file:   meta-amd/meta-daytona/recipes-phosphor/sensors/phosphor-hwmon/obmc/hwmon/ahb/apb/bus@1e78a000/i2c-bus@300/lm75a@4a.conf
	new file:   meta-amd/meta-daytona/recipes-phosphor/sensors/phosphor-hwmon/obmc/hwmon/ahb/apb/bus@1e78a000/i2c-bus@300/lm75a@4b.conf
	new file:   meta-amd/meta-daytona/recipes-phosphor/sensors/phosphor-hwmon/obmc/hwmon/ahb/apb/bus@1e78a000/i2c-bus@300/lm75a@4c.conf
	new file:   meta-amd/meta-daytona/recipes-phosphor/sensors/phosphor-hwmon/obmc/hwmon/ahb/apb/bus@1e78a000/i2c-bus@300/lm75a@4d.conf
	new file:   meta-amd/meta-daytona/recipes-phosphor/sensors/phosphor-hwmon/obmc/hwmon/ahb/apb/bus@1e78a000/i2c-bus@300/lm75a@4e.conf
	new file:   meta-amd/meta-daytona/recipes-phosphor/sensors/phosphor-hwmon/obmc/hwmon/ahb/apb/bus@1e78a000/i2c-bus@300/lm75a@4f.conf
	new file:   meta-amd/meta-daytona/recipes-phosphor/sensors/phosphor-hwmon/obmc/hwmon/ahb/apb/pwm-tacho-controller@1e786000.conf
	new file:   meta-amd/meta-daytona/recipes-phosphor/sensors/phosphor-hwmon_%.bbappend
	new file:   meta-amd/meta-daytona/recipes-x86/chassis/x86-power-control/0001-Amd-power-control-modifications-for-EthanolX.patch
	new file:   meta-amd/meta-daytona/recipes-x86/chassis/x86-power-control_%.bbappend
	new file:   meta-amd/meta-daytona/recipes.txt

user@quanta:~/dhanasekar/BMC/AMD/daytona1/openbmc$ git commit
file is empty: .git/COMMIT_EDITMSG
user@quanta:~/dhanasekar/BMC/AMD/daytona1/openbmc$ git status
On branch master
Your branch is up to date with 'origin/master'.

Changes to be committed:
  (use "git reset HEAD <file>..." to unstage)

	new file:   meta-amd/meta-daytona/conf/bblayers.conf.sample
	new file:   meta-amd/meta-daytona/conf/conf-notes.txt
	new file:   meta-amd/meta-daytona/conf/layer.conf
	new file:   meta-amd/meta-daytona/conf/local.conf.sample
	new file:   meta-amd/meta-daytona/conf/machine/daytona.conf
	new file:   meta-amd/meta-daytona/recipes-kernel/linux/linux-aspeed/0001-ARM-dts-aspeed-Add-GPIO-line-names-for-AMD-EthanolX.patch
	new file:   meta-amd/meta-daytona/recipes-kernel/linux/linux-aspeed/ethanolx.cfg
	new file:   meta-amd/meta-daytona/recipes-kernel/linux/linux-aspeed_%.bbappend
	new file:   meta-amd/meta-daytona/recipes-phosphor/configuration/ethanolx-yaml-config.bb
	new file:   meta-amd/meta-daytona/recipes-phosphor/configuration/ethanolx-yaml-config/ethanolx-ipmi-fru-properties.yaml
	new file:   meta-amd/meta-daytona/recipes-phosphor/configuration/ethanolx-yaml-config/ethanolx-ipmi-fru.yaml
	new file:   meta-amd/meta-daytona/recipes-phosphor/configuration/ethanolx-yaml-config/ethanolx-ipmi-sensors.yaml
	new file:   meta-amd/meta-daytona/recipes-phosphor/ipmi/phosphor-ipmi-config.bbappend
	new file:   meta-amd/meta-daytona/recipes-phosphor/ipmi/phosphor-ipmi-config/channel_config.json
	new file:   meta-amd/meta-daytona/recipes-phosphor/ipmi/phosphor-ipmi-config/dcmi_sensors.json
	new file:   meta-amd/meta-daytona/recipes-phosphor/ipmi/phosphor-ipmi-fru_%.bbappend
	new file:   meta-amd/meta-daytona/recipes-phosphor/ipmi/phosphor-ipmi-host_%.bbappend
	new file:   meta-amd/meta-daytona/recipes-phosphor/leds/ethanolx-led-manager-config-native.bb
	new file:   meta-amd/meta-daytona/recipes-phosphor/leds/ethanolx-led-manager-config/led.yaml
[... cut ...]
	new file:   meta-amd/meta-daytona/recipes-phosphor/sensors/phosphor-hwmon/obmc/hwmon/ahb/apb/pwm-tacho-controller@1e786000.conf
	new file:   meta-amd/meta-daytona/recipes-phosphor/sensors/phosphor-hwmon_%.bbappend
	new file:   meta-amd/meta-daytona/recipes-x86/chassis/x86-power-control/0001-Amd-power-control-modifications-for-EthanolX.patch
	new file:   meta-amd/meta-daytona/recipes-x86/chassis/x86-power-control_%.bbappend
	new file:   meta-amd/meta-daytona/recipes.txt

user@quanta:~/dhanasekar/BMC/AMD/daytona1/openbmc$ 

```

jdhanasekar, 2021-01-26:
I have pushed my changes and getting below error,

```

user@quanta:~/dhanasekar/BMC/AMD/daytona1/openbmc$ git remote add daytona5 ssh://jdhanasekar@openbmc.gerrit/openbmc/daytona5
user@quanta:~/dhanasekar/BMC/AMD/daytona1/openbmc$ gitdir=$(git rev-parse --git-dir)
user@quanta:~/dhanasekar/BMC/AMD/daytona1/openbmc$ 
user@quanta:~/dhanasekar/BMC/AMD/daytona1/openbmc$ scp -p -P 29418 openbmc.gerrit:hooks/commit-msg ${gitdir}/hooks
commit-msg                                                                                         100% 1790     6.6KB/s   00:00    
user@quanta:~/dhanasekar/BMC/AMD/daytona1/openbmc$ 
user@quanta:~/dhanasekar/BMC/AMD/daytona1/openbmc$ git push ssh://jdhanasekar@openbmc.gerrit/openbmc/daytona5 HEAD:refs/for/master%notify=NONE
fatal: project openbmc/daytona5 not found
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
user@quanta:~/dhanasekar/BMC/AMD/daytona1/openbmc$ 

```

williamspatrick, 2021-01-26:
Please check the URL you are trying to push to.

jdhanasekar, 2021-01-26:
```
user@quanta:~/dhanasekar/BMC/AMD/daytona1/openbmc$ git push ssh://jdhanasekar@openbmc.gerrit/openbmc/openbmc HEAD:refs/for/master%notify=NONE
Counting objects: 70, done.
Delta compression using up to 32 threads.
Compressing objects: 100% (31/31), done.
Writing objects: 100% (70/70), 9.93 KiB | 9.93 MiB/s, done.
Total 70 (delta 53), reused 54 (delta 37)
remote: Resolving deltas: 100% (53/53)
remote: Processing changes: refs: 1, done    
To ssh://openbmc.gerrit/openbmc/openbmc
 ! [remote rejected]     HEAD -> refs/for/master%notify=NONE (no new changes)
error: failed to push some refs to 'ssh://jdhanasekar@openbmc.gerrit/openbmc/openbmc'
```

jdhanasekar, 2021-01-27:
Corrected the url and repo, still getting error,

```
user@quanta:~/dhanasekar/BMC/AMD/daytona1/openbmc$ git commit --amend
[master d33c73c27] Change-Id: Ia19a7a4a10245c1c2f85f6da216997cb6a836daa changes created by dj for fixing issue in new system creation. this changes will not be checked in.
 Author: Ed Tanous <edtanous@google.com>
 Date: Wed Jan 6 13:04:58 2021 -0800
 71 files changed, 2701 insertions(+), 3 deletions(-)
 create mode 100644 meta-amd/meta-daytona/conf/bblayers.conf.sample
 create mode 100644 meta-amd/meta-daytona/conf/conf-notes.txt
 create mode 100644 meta-amd/meta-daytona/conf/layer.conf
 create mode 100644 meta-amd/meta-daytona/conf/local.conf.sample
 create mode 100644 meta-amd/meta-daytona/conf/machine/daytona.conf
 create mode 100644 meta-amd/meta-daytona/recipes-kernel/linux/linux-aspeed/0001-ARM-dts-aspeed-Add-GPIO-line-names-for-AMD-EthanolX.patch
 create mode 100644 meta-amd/meta-daytona/recipes-kernel/linux/linux-aspeed/ethanolx.cfg
 create mode 100644 meta-amd/meta-daytona/recipes-kernel/linux/linux-aspeed_%.bbappend
 create mode 100644 meta-amd/meta-daytona/recipes-phosphor/configuration/ethanolx-yaml-config.bb
 create mode 100644 meta-amd/meta-daytona/recipes-phosphor/configuration/ethanolx-yaml-config/ethanolx-ipmi-fru-properties.yaml
 create mode 100644 meta-amd/meta-daytona/recipes-phosphor/configuration/ethanolx-yaml-config/ethanolx-ipmi-fru.yaml
 create mode 100644 meta-amd/meta-daytona/recipes-phosphor/configuration/ethanolx-yaml-config/ethanolx-ipmi-sensors.yaml
 create mode 100644 meta-amd/meta-daytona/recipes-phosphor/ipmi/phosphor-ipmi-config.bbappend
 create mode 100644 meta-amd/meta-daytona/recipes-phosphor/ipmi/phosphor-ipmi-config/channel_config.json
 create mode 100644 meta-amd/meta-daytona/recipes-phosphor/ipmi/phosphor-ipmi-config/dcmi_sensors.json
 create mode 100644 meta-amd/meta-daytona/recipes-phosphor/ipmi/phosphor-ipmi-fru_%.bbappend
 create mode 100644 meta-amd/meta-daytona/recipes-phosphor/ipmi/phosphor-ipmi-host_%.bbappend
 create mode 100644 meta-amd/meta-daytona/recipes-phosphor/leds/ethanolx-led-manager-config-native.bb
 create mode 100644 meta-amd/meta-daytona/recipes-phosphor/leds/ethanolx-led-manager-config/led.yaml
 create mode 100644 meta-amd/meta-daytona/recipes-phosphor/sensors/phosphor-hwmon/obmc/hwmon/ahb/apb/bus@1e78a000/i2c-bus@300/lm75a@48.conf
 create mode 100644 meta-amd/meta-daytona/recipes-phosphor/sensors/phosphor-hwmon/obmc/hwmon/ahb/apb/bus@1e78a000/i2c-bus@300/lm75a@49.conf
 create mode 100644 meta-amd/meta-daytona/recipes-phosphor/sensors/phosphor-hwmon/obmc/hwmon/ahb/apb/bus@1e78a000/i2c-bus@300/lm75a@4a.conf
 create mode 100644 meta-amd/meta-daytona/recipes-phosphor/sensors/phosphor-hwmon/obmc/hwmon/ahb/apb/bus@1e78a000/i2c-bus@300/lm75a@4b.conf
 create mode 100644 meta-amd/meta-daytona/recipes-phosphor/sensors/phosphor-hwmon/obmc/hwmon/ahb/apb/bus@1e78a000/i2c-bus@300/lm75a@4c.conf
 create mode 100644 meta-amd/meta-daytona/recipes-phosphor/sensors/phosphor-hwmon/obmc/hwmon/ahb/apb/bus@1e78a000/i2c-bus@300/lm75a@4d.conf
 create mode 100644 meta-amd/meta-daytona/recipes-phosphor/sensors/phosphor-hwmon/obmc/hwmon/ahb/apb/bus@1e78a000/i2c-bus@300/lm75a@4e.conf
 create mode 100644 meta-amd/meta-daytona/recipes-phosphor/sensors/phosphor-hwmon/obmc/hwmon/ahb/apb/bus@1e78a000/i2c-bus@300/lm75a@4f.conf
 create mode 100644 meta-amd/meta-daytona/recipes-phosphor/sensors/phosphor-hwmon/obmc/hwmon/ahb/apb/pwm-tacho-controller@1e786000.conf
 create mode 100644 meta-amd/meta-daytona/recipes-phosphor/sensors/phosphor-hwmon_%.bbappend
 create mode 100644 meta-amd/meta-daytona/recipes-x86/chassis/x86-power-control/0001-Amd-power-control-modifications-for-EthanolX.patch
 create mode 100644 meta-amd/meta-daytona/recipes-x86/chassis/x86-power-control_%.bbappend
 create mode 100644 meta-amd/meta-daytona/recipes.txt
 rename {meta-intel => meta-intel-openbmc}/COPYING.MIT (100%)
 rename {meta-intel => meta-intel-openbmc}/COPYING.apache-2.0 (100%)
 rename {meta-intel => meta-intel-openbmc}/LICENSE (100%)
 rename {meta-intel => meta-intel-openbmc}/MAINTAINERS (100%)
 rename {meta-intel => meta-intel-openbmc}/conf/layer.conf (100%)
 rename {meta-intel => meta-intel-openbmc}/conf/machine/include/intel.inc (100%)
 rename {meta-intel => meta-intel-openbmc}/meta-common/recipes-core/host-error-monitor/host-error-monitor_git.bb (100%)
 rename {meta-intel => meta-intel-openbmc}/meta-common/recipes-core/libpeci/libpeci_git.bb (100%)
 rename {meta-intel => meta-intel-openbmc}/meta-common/recipes-core/systemd/obmc-targets.bbappend (100%)
 rename {meta-intel => meta-intel-openbmc}/meta-common/recipes-extended/rsyslog/rsyslog/rotate-event-logs.service (100%)
 rename {meta-intel => meta-intel-openbmc}/meta-common/recipes-extended/rsyslog/rsyslog/rotate-event-logs.sh (100%)
 rename {meta-intel => meta-intel-openbmc}/meta-common/recipes-extended/rsyslog/rsyslog/rsyslog-override.conf (100%)
 rename {meta-intel => meta-intel-openbmc}/meta-common/recipes-extended/rsyslog/rsyslog/rsyslog.conf (100%)
 rename {meta-intel => meta-intel-openbmc}/meta-common/recipes-extended/rsyslog/rsyslog/rsyslog.logrotate (100%)
 rename {meta-intel => meta-intel-openbmc}/meta-common/recipes-extended/rsyslog/rsyslog_%.bbappend (100%)
 rename {meta-intel => meta-intel-openbmc}/meta-common/recipes-intel/ipmi/intel-ipmi-oem_git.bb (100%)
 rename {meta-intel => meta-intel-openbmc}/meta-common/recipes-intel/packagegroups/packagegroup-intel-apps.bb (100%)
[... cut ...]
user@quanta:~/dhanasekar/BMC/AMD/daytona1/openbmc$ scp -p -P 29418 openbmc.gerrit:hooks/commit-msg ${gitdir}/hooks
commit-msg                                                                                         100% 1790     6.6KB/s   00:00    
user@quanta:~/dhanasekar/BMC/AMD/daytona1/openbmc$ 
user@quanta:~/dhanasekar/BMC/AMD/daytona1/openbmc$ git push ssh://openbmc.gerrit/openbmc/Test1 HEAD:refs/for/master%notify=NONE
fatal: project openbmc/Test1 not found
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
```

jdhanasekar, 2021-01-27:
I am able solve the issue in push, but I am facing "signed off" error. Even I run "git commit --amend", but still I observing error. 

```
user@quanta:~/dhanasekar/BMC/AMD/test01$ git push TRY4 HEAD:refs/for/master
Counting objects: 9, done.
Delta compression using up to 32 threads.
Compressing objects: 100% (8/8), done.
Writing objects: 100% (9/9), 5.38 KiB | 5.38 MiB/s, done.
Total 9 (delta 2), reused 0 (delta 0)
remote: Resolving deltas: 100% (2/2)
remote: Processing changes: refs: 1, done    
To ssh://openbmc.gerrit/openbmc/openbmc
 ! [remote rejected]     HEAD -> refs/for/master (commit 4868f62: not Signed-off-by author/committer/uploader in message footer)
error: failed to push some refs to 'ssh://openbmc.gerrit/openbmc/openbmc'

```

williamspatrick, 2021-01-27:
I’m not sure what you did but it appears that your commit was authored by Ed. I’m guessing you amended a commit that was in master rather than creating your own commit?

What I would suggest doing:
```
git fetch
git rebase origin/master
git commit —amend —reset-author
```

Make sure you delete the ChangeId and any Signed-off-by that isn’t you (likely Ed) in this commit message on the last step.

Many of these problems you’re running into aren’t actually related to Gerrit, but Git itself. I suspect you don’t have much experience with Git either?  You might want to go through some git tutorials as well.

jdhanasekar, 2021-01-27:
As per your input, i have deleted ChangeId,

```
user@quanta:~/dhanasekar/BMC/AMD/test01$ git fetch
remote: Counting objects: 4083, done
remote: Finding sources: 100% (6/6)
remote: Total 6 (delta 3), reused 6 (delta 3)
Unpacking objects: 100% (6/6), done.
From https://gerrit.openbmc-project.xyz/openbmc/openbmc
   1b4a85eb9..377d8f40e  master     -> origin/master
user@quanta:~/dhanasekar/BMC/AMD/test01$ git rebase origin/master
First, rewinding head to replay your work on top of it...
Applying: Test1: Creating a new system
user@quanta:~/dhanasekar/BMC/AMD/test01$ git commit --amend --reset-author
[master 97e0e2e1d] Test4: Creating a new system
 32 files changed, 2698 insertions(+)
 create mode 100644 meta-amd/meta-daytonax/conf/bblayers.conf.sample
 create mode 100644 meta-amd/meta-daytonax/conf/conf-notes.txt
 create mode 100644 meta-amd/meta-daytonax/conf/layer.conf
 create mode 100644 meta-amd/meta-daytonax/conf/local.conf.sample
 create mode 100644 meta-amd/meta-daytonax/conf/machine/daytonax.conf
 create mode 100644 meta-amd/meta-daytonax/recipes-kernel/linux/linux-aspeed/0001-ARM-dts-aspeed-Add-GPIO-line-names-for-AMD-EthanolX.patch
 create mode 100644 meta-amd/meta-daytonax/recipes-kernel/linux/linux-aspeed/ethanolx.cfg
 create mode 100644 meta-amd/meta-daytonax/recipes-kernel/linux/linux-aspeed_%.bbappend
 create mode 100644 meta-amd/meta-daytonax/recipes-phosphor/configuration/ethanolx-yaml-config.bb
 create mode 100644 meta-amd/meta-daytonax/recipes-phosphor/configuration/ethanolx-yaml-config/ethanolx-ipmi-fru-properties.yaml
 create mode 100644 meta-amd/meta-daytonax/recipes-phosphor/configuration/ethanolx-yaml-config/ethanolx-ipmi-fru.yaml
 create mode 100644 meta-amd/meta-daytonax/recipes-phosphor/configuration/ethanolx-yaml-config/ethanolx-ipmi-sensors.yaml
 create mode 100644 meta-amd/meta-daytonax/recipes-phosphor/ipmi/phosphor-ipmi-config.bbappend
 create mode 100644 meta-amd/meta-daytonax/recipes-phosphor/ipmi/phosphor-ipmi-config/channel_config.json
 create mode 100644 meta-amd/meta-daytonax/recipes-phosphor/ipmi/phosphor-ipmi-config/dcmi_sensors.json
 create mode 100644 meta-amd/meta-daytonax/recipes-phosphor/ipmi/phosphor-ipmi-fru_%.bbappend
 create mode 100644 meta-amd/meta-daytonax/recipes-phosphor/ipmi/phosphor-ipmi-host_%.bbappend
 create mode 100644 meta-amd/meta-daytonax/recipes-phosphor/leds/ethanolx-led-manager-config-native.bb
 create mode 100644 meta-amd/meta-daytonax/recipes-phosphor/leds/ethanolx-led-manager-config/led.yaml
 create mode 100644 meta-amd/meta-daytonax/recipes-phosphor/sensors/phosphor-hwmon/obmc/hwmon/ahb/apb/bus@1e78a000/i2c-bus@300/lm75a@48.conf
 create mode 100644 meta-amd/meta-daytonax/recipes-phosphor/sensors/phosphor-hwmon/obmc/hwmon/ahb/apb/bus@1e78a000/i2c-bus@300/lm75a@49.conf
 create mode 100644 meta-amd/meta-daytonax/recipes-phosphor/sensors/phosphor-hwmon/obmc/hwmon/ahb/apb/bus@1e78a000/i2c-bus@300/lm75a@4a.conf
 create mode 100644 meta-amd/meta-daytonax/recipes-phosphor/sensors/phosphor-hwmon/obmc/hwmon/ahb/apb/bus@1e78a000/i2c-bus@300/lm75a@4b.conf
 create mode 100644 meta-amd/meta-daytonax/recipes-phosphor/sensors/phosphor-hwmon/obmc/hwmon/ahb/apb/bus@1e78a000/i2c-bus@300/lm75a@4c.conf
 create mode 100644 meta-amd/meta-daytonax/recipes-phosphor/sensors/phosphor-hwmon/obmc/hwmon/ahb/apb/bus@1e78a000/i2c-bus@300/lm75a@4d.conf
 create mode 100644 meta-amd/meta-daytonax/recipes-phosphor/sensors/phosphor-hwmon/obmc/hwmon/ahb/apb/bus@1e78a000/i2c-bus@300/lm75a@4e.conf
 create mode 100644 meta-amd/meta-daytonax/recipes-phosphor/sensors/phosphor-hwmon/obmc/hwmon/ahb/apb/bus@1e78a000/i2c-bus@300/lm75a@4f.conf
 create mode 100644 meta-amd/meta-daytonax/recipes-phosphor/sensors/phosphor-hwmon/obmc/hwmon/ahb/apb/pwm-tacho-controller@1e786000.conf
 create mode 100644 meta-amd/meta-daytonax/recipes-phosphor/sensors/phosphor-hwmon_%.bbappend
 create mode 100644 meta-amd/meta-daytonax/recipes-x86/chassis/x86-power-control/0001-Amd-power-control-modifications-for-EthanolX.patch
 create mode 100644 meta-amd/meta-daytonax/recipes-x86/chassis/x86-power-control_%.bbappend
 create mode 100644 meta-amd/meta-daytonax/recipes.txt

```

Now, I need to push the changes again , like this,

```
git remote add TTTT ssh://openbmc.gerrit/openbmc/openbmc
gitdir=$(git rev-parse --git-dir)
scp -p -P 29418 openbmc.gerrit:hooks/commit-msg ${gitdir}/hooks
git push TTTT HEAD:refs/for/master
```
Am I correct ?.

williamspatrick, 2021-01-27:
Yes. But you only need to run the scp operation once per repository and not on every push. This downloads a git-hook from Gerrit that automatically adds the ChangeId to your commit message.

jdhanasekar, 2021-01-27:
sorry, I have run scp operation now. I have run like this,

```
user@quanta:~/dhanasekar/BMC/AMD/test01$ git remote add TRY6 ssh://openbmc.gerrit/openbmc/openbmc
user@quanta:~/dhanasekar/BMC/AMD/test01$ gitdir=$(git rev-parse --git-dir)
user@quanta:~/dhanasekar/BMC/AMD/test01$ scp -p -P 29418 openbmc.gerrit:hooks/commit-msg ${gitdir}/hooks
commit-msg                                                                                         100% 1790     6.2KB/s   00:00    
user@quanta:~/dhanasekar/BMC/AMD/test01$ git push TRY6 HEAD:refs/for/master
Counting objects: 9, done.
Delta compression using up to 32 threads.
Compressing objects: 100% (8/8), done.
Writing objects: 100% (9/9), 5.37 KiB | 5.37 MiB/s, done.
Total 9 (delta 2), reused 0 (delta 0)
remote: Resolving deltas: 100% (2/2)
remote: Processing changes: refs: 1, done    
To ssh://openbmc.gerrit/openbmc/openbmc
 ! [remote rejected]     HEAD -> refs/for/master (commit 97e0e2e: not Signed-off-by author/committer/uploader in message footer)
error: failed to push some refs to 'ssh://openbmc.gerrit/openbmc/openbmc'

```

When I give "git log", I am seeing my commit there,

```
user@quanta:~/dhanasekar/BMC/AMD/test01$ git log
commit 97e0e2e1dc8a3bffa7dc4009d18f1b4511ec77db (HEAD -> master)
Author: jdhanasekar <jdhanasekar@velankanigroup.com>
Date:   Wed Jan 27 18:29:14 2021 +0530

    Test4: Creating a new system
    
    Change-Id: I9147e7475c12fa7aff5ad6b65a274249ceae57d9

commit 377d8f40e88eeb9ccb1321590a3af3c2caa94dda (origin/master, origin/HEAD)
Author: Alexander Amelkin <a.amelkin@yadro.com>
Date:   Tue Jan 26 23:53:05 2021 +0300

    meta-yadro: os-release: Fix release builds
    
    Add a missing variable initialization that resulted
    in an error during release builds
    
    Signed-off-by: Alexander Amelkin <a.amelkin@yadro.com>
    Change-Id: Id938ecb17f2a7b37816b4ff72c066ed6581f4577

commit 1b4a85eb91c205140e7aa10f8304b92b6d11dc3e
Author: Patrick Williams <patrick@stwcx.xyz>
Date:   Mon Jan 25 17:06:50 2021 -0600

    meta-facebook: fb-powerctrl: remove excess trailing slash
    
    Clean up the bitbake warning.
    
        WARNING: Recipe fb-powerctrl sets S variable with trailing slash
        '.../tmp/work/arm1176jzs-openbmc-linux-gnueabi/fb-powerctrl/1.0-r0/',
        remove it
    
    Signed-off-by: Patrick Williams <patrick@stwcx.xyz>
    Change-Id: Iad85528bfc42f042985f02035bbe8cc41fa020e8

commit 6510566f018b664b901835547891e08024616811
Author: Ed Tanous <edtanous@google.com>
Date:   Wed Jan 6 13:04:58 2021 -0800

    Move meta-intel to meta-intel-openbmc
    
    meta-intel conflicts with the upstream yocto meta layer of the same
    name.  This commit renames meta-intel to meta-intel-openbmc and updates
    all the relevant bblayers files with the new name.
    
    This is to pave the way for allowing the yocto meta-intel in the tree,
    which would allow the use of generic-x86 machine type, which would allow
    building and running tests natively.
    
    Signed-off-by: Ed Tanous <edtanous@google.com>
    Change-Id: Ia19a7a4a10245c1c2f85f6da216997cb6a836daa
:

```

williamspatrick, 2021-01-27:
`remote rejected]     HEAD -> refs/for/master (commit 97e0e2e: not Signed-off-by author/committer/uploader in message footer)`

Gerrit is trying to tell you what is wrong. 

Run: `git commit —amend -s` to add a signed off by.

jdhanasekar, 2021-01-28:
Thank you. I have pushed the changes sucessfully. 

```
user@quanta:~/dhanasekar/BMC/AMD/test01$ git push TRY7 HEAD:refs/for/master
Counting objects: 55, done.
Delta compression using up to 32 threads.
Compressing objects: 100% (30/30), done.
Writing objects: 100% (55/55), 13.06 KiB | 6.53 MiB/s, done.
Total 55 (delta 31), reused 35 (delta 21)
remote: Resolving deltas: 100% (31/31)
remote: Processing changes: refs: 1, new: 1, done    
remote: 
remote: SUCCESS
remote: 
remote:   https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/40075 Test4: Creating a new system [NEW]
remote: 
To ssh://openbmc.gerrit/openbmc/openbmc
 * [new branch]          HEAD -> refs/for/master
```

jdhanasekar, 2021-01-29:
I have uploaded the changes to figure out why I am unable to create a new system. Please view it.

williamspatrick, 2021-01-31:
I made comments in your Gerrit commit of some changes that enabled me to get significantly farther in the build.

supven01, 2021-01-31:
Please follow these steps to bring up a new system.
https://github.com/openbmc/docs/blob/master/development/add-new-system.md

When uploading to gerrit, please split the patches grouped by the recipe.
For e.g., /recipes-phosphor/ipmi/ related will be one patch, meta-amd/meta-daytonax/conf will be another and so on.

I will take a look at it, once you separate out the patches.

Do the following
git reset HEAD~
which will stop at the state before the last commit but your changes will be retained as "unstaged" edits.

Now, go ahead and create multiple commits by adding only the relevant files.

Follow the steps for pushing the review.

jdhanasekar, 2021-02-02:
I have debugged further and found after adding only ```meta``` package in ```bblayers.conf.sample```, like this 

```
# LAYER_CONF_VERSION is increased each time build/conf/bblayers.conf
# changes incompatibly
LCONF_VERSION = "7"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
  ##OEROOT##/meta \
  "
BBLAYERS_NON_REMOVABLE ?= " \  
  ##OEROOT##/meta \
  "

```
I am getting error,
```
ERROR:  OE-core's config sanity checker detected a potential misconfiguration.
    Either fix the cause of this error or at your own risk disable the checker (see sanity.conf).
    Following is the list of potential problems / advisories:

    MACHINE=ethanolx is invalid. Please set a valid MACHINE in your local.conf, environment or other configuration file.

```

jdhanasekar, 2021-02-05:
Created New system successfully.

closed 2021-02-05 by jdhanasekar