openbmc/openbmc #3748: Is there feasible to port openbmc  into a  Aarch64 machine

wangmin239, 2021-01-12:
Hi folks,

Currently, the linux kernel  and OS had been successfully compiled by using the [configuration](https://github.com/openbmc/openbmc/blob/master/poky/meta/conf/machine/include/arm/arch-armv8a.inc). while the u-boot  failed to be compiled. Are there some instructions of compiling u-boot and packaging a image in the openbmc skeleton?  Is there feasible to port openbmc into a Aarch64 machine? And what should be paid attention to during the process of the porting.

 Any help will be greatly appreciated.

Thanks!

thangqn-ampere, 2021-02-25:
Do you mean you want to compile OpenBMC on Aarch64 machine or you want to run OpenBMC on Aarch64?

jk-ozlabs, 2021-02-26:
And just to clarify, do you want the BMC architecture to be aarch64, or the host architecture?

For the former, do you have an aarch64 SoC in mind, that is suitable for a BMC platform?

wangmin239, 2021-03-03:
> Do you mean you want to compile OpenBMC on Aarch64 machine or you want to run OpenBMC on Aarch64?

I am trying  running OpenBMC on Aarch64 machines.  Could you give some suggestions?

wangmin239, 2021-03-03:
> And just to clarify, do you want the BMC architecture to be aarch64, or the host architecture?
> 
> For the former, do you have an aarch64 SoC in mind, that is suitable for a BMC platform?

Yes, I have an aarch64 Soc on a BMC card. At present, I am trying porting openBmc to run on the aarch64 SoC. But I encountered many errors during the compiling openBmc by using a aarch64 compilier.  Is it feasible to port openbmc into a Aarch64 machine? Could you give some help?

jk-ozlabs, 2021-03-04:
OK, super. Yes, this should definitely be do-able, provided you have:

1) yocto support for your compiler; and
2) a kernel tree suitable for your platform

However: you've filed an issue against OpenBMC here, and this is a more of a support request, rather than an issue with the code. 

If you can post to the openbmc mailing list (openbmc@lists.ozlabs.org), there'll probably be a bunch of folks there to help out too. If you can include some indication of the errors you're seeing, along with your process for doing the port, then that would really help.

wangmin239, 2021-03-04:
I will post to the openbmc mailing list to discuss about the porting work.  And the issue is closed. Finally, thanks a lot.

closed 2021-03-04 by wangmin239