openbmc/openbmc #3750: linux-pam: support cracklib and tally2 replacements

geissonator, 2021-01-13:
See https://gerrit.openbmc-project.xyz/c/openbmc/meta-phosphor/+/39683 for details. 

2 libraries we depended on for our pam login path were deprecated and removed. We need to look into the new replacements:
- Removed deprecated pam_cracklib module, use pam_passwdqc (from passwdqc project) or pam_pwquality (from libpwquality project) instead.
- Removed deprecated pam_tally and pam_tally2 modules, use pam_faillock instead.

geissonator, 2021-01-13:
FYI @rthomaiy @joseph-reynolds

joseph-reynolds, 2021-01-29:
The effort to use supported Linux-PAM modules is (currently) here:
https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/40102
https://gerrit.openbmc-project.xyz/c/openbmc/phosphor-user-manager/+/39853

joseph-reynolds, 2021-02-05:
Here is the current status of the effort:
- Linux-PAM is temporarily reverted to 1.3.1 as of : https://github.com/openbmc/openbmc/commit/87ddd3eab4df68e624b5350ccaab28b3b97547c0 -- To make progress, we'll need PAM 1.5.1 to pick up the new recipes.
- The openbmc commit above is based on PAM 1.5.1 so it does not currently build.
- The code is tested and working for local users.  LDAP needs testing.
- The libpwquality package pulls in python and other stuff - https://github.com/openbmc/openbmc/blob/master/meta-openembedded/meta-oe/recipes-extended/libpwquality/libpwquality_1.4.4.bb.  We cannot tolerate the size increase in our small images.  I pushed a patch to remove python, but the other stuff in distutils may be harder to remove.  I believe the right way to do this is to change the recipe.
- I've mostly-coded a tool to convert from the old PAM config file to the new PAM config files.  This is needed to convert any overlayfs files on the BMC, to provide a forward migration path.  A new service would run this tool once after an image with the new PAM stuff is booted for the first time.

joseph-reynolds, 2021-02-06:
I got the PAM replacement modules working, but the libpwquality_1.4.4.bb recipe unconditionally inherits distutils3-base which add Pythons and several megabytes if image size, which we cannot tolerate.  The libpwquality project itself has an option to omit the Python bindings, so I wrote libpwquality.bbappend EXTRA_OECONF += "--enable-python-bindings=no" but the resulting image is still too large.  The only way I could get libpwquality built into my image without the extra Python stuff is to edit the libpwquality recipe itself (remove "inherit distutils3-base").  The problem is this recipe is owned by oe-core and there is no way to change the inherit in a bbappend.

I think the solution is to enhance the recipe upstream to optionally omit the python bindings, perhaps with a PACKAGECONFIG, take that patch in OpenBMC, and configure Python out of this package.  Does that sound right?  The recipe needs to conditionally inherit a class.  Anyone want to help out here?

bradbishop, 2021-02-06:
We could try something like this upstream:

```
PACKAGECONFIG ??= "python-bindings"
PACKAGECONFIG[python-bindings] = "--enable-python-bindings=yes, --enable-python-bindings=no"
PYTHON_INHERIT = "${@bb.utils.contains('PACKAGECONFIG', 'python-bindings', ''python3-distutils-base, '', d)}"
inherit "${PYTHON_INHERIT}"
```

joseph-reynolds, 2021-02-10:
I asked oe-core for help here: https://lists.openembedded.org/g/openembedded-core/message/147913?p=,,,20,0,0,0::Created,,libpwquality,20,2,0,80516859

joseph-reynolds, 2021-05-10:
Status update (non-update): The next step is to fix up the libpwquality recipe "the right way" so python3 can be optionally included, and upstream that change to oe-core.

kapil1024, 2021-06-01:
Have you tried below (or similar) to resolve libpwquality problem?

meta-openembedded/meta-oe/recipes-extended/libpwquality/libpwquality_1.4.4.bb 

> 	-PACKAGECONFIG ??= "${@bb.utils.contains('DISTRO_FEATURES', 'pam', 'pam', '', d)}"
> 	+PACKAGECONFIG ??= "${@bb.utils.contains('DISTRO_FEATURES', 'pam', 'pam', '', d)} python-bindings"
> 	+PACKAGECONFIG[python-bindings] = "--enable-python-bindings=yes, --enable-python-bindings=no"
> 	 PACKAGECONFIG[pam] = "--enable-pam, --disable-pam, libpam"
> 	+RDEPENDS_${PN}_remove = "${@bb.utils.contains('PACKAGECONFIG', 'python-bindings', ' ', '${PYTHON_PN}-core', d)}"
> 	+_PYTHON_SYSCONFIGDATA_NAME[unexport] = "${@bb.utils.contains('PACKAGECONFIG', 'python-bindings', '0', '1', d)}"

meta-phosphor/recipes-extended/libpwquality/libpwquality_%.bbappend 

> 	+PACKAGECONFIG_remove = "python-bindings"

geissonator, 2021-06-01:
@kapil1024 we'll take whatever we can get. We just need someone to send something upstream to meta-openembedded that we could then pull into OpenBMC that would let us remove the python requirement. We import meta-openembedded/meta-oe/recipes-extended/libpwquality/libpwquality_1.4.4.bb from yocto so don't make direct changes in our OpenBMC repo to it (i.e. we treat it read-only and only import what goes into upstream).

joseph-reynolds, 2022-01-26:
This topic was recently mentioned in the Discord OpenBMC #yocto channel.  Notes:
- The reviews for this are tested and working as a proof of concept.  They need to be updated.  They are:
   - https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/41357
   - https://gerrit.openbmc-project.xyz/c/openbmc/phosphor-user-manager/+/39853
- The [openbmc review](https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/41357) has a "convert-pam-configs" service and shell script to convert existing phosphor-user-management settings which were stored in the old PAM module config files to the new PAM module config files.  Comprehensive documentation and test cases are included in the commit.  This conversion service provides forward compatibility for your existing OpenBMC systems to use the new PAM modules without disruption.  (Backward compatibility is not provided.)
- Help is needed for the recipe change; this is detailed in https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/40102/5/meta-openembedded/meta-oe/recipes-extended/libpwquality/libpwquality_1.4.4.bb

human3rr, 2022-06-30:
Working with openembedded-devel to try to resolve solution. 
https://lists.openembedded.org/g/openembedded-devel/topic/92052665#