openbmc/openbmc #3752: QEMU-based BMC doesns't seem to boot completely

SaiHarshaK, 2021-01-14:
Hello, I am new and wanted to try out the emulation following the [docs](https://github.com/openbmc/docs/blob/master/development/dev-environment.md), but after 3rd step in [download-and-start-qemu-session](https://github.com/openbmc/docs/blob/master/development/dev-environment.md#download-and-start-qemu-session), it looks like the shell is stuck as I don't see any kind of prompt.
Last few lines of output after I executed the 3rd step:
```
[    0.698118] ipmi-bt-host 1e789140.ibt: Found bt bmc device
[    0.699155] ipmi-bt-host 1e789140.ibt: Using IRQ 34
[    0.704711] Serial: 8250/16550 driver, 6 ports, IRQ sharing enabled
[    0.710729] 1e787000.serial: ttyS5 at MMIO 0x1e787000 (irq = 34, base_baud = 1546875) is a 16550A
[    0.714470] 1e783000.serial: ttyS0 at MMIO 0x1e783000 (irq = 32, base_baud = 1500000) is a 16550A
[    0.715316] 1e784000.serial: ttyS4 at MMIO 0x1e784000 (irq = 33, base_baud = 1500000) is a 16550A
[    0.727185] printk: console [ttyS4] enabled
qemu-system-arm: warning: I/O thread spun for 1000 iterations

```
Please suggest how to proceed.

williamspatrick, 2021-01-14:
What version of QEMU are you running?  Did you build it yourself from our repo mirror or use something from upstream?

SaiHarshaK, 2021-01-14:
It seems the QEMU version is the issue, I didn't think the version from apt install would be so behind (2.11.1). Thanks for the quick response.
/close

closed 2021-01-14 by SaiHarshaK