openbmc/openbmc #3753: Getting High level threat for VNC server in the vulnerability scan report.

tejas-patil20, 2021-01-15:
## Vulnerabilities
26925 - VNC Server Unauthenticated Access
## Synopsis
The remote VNC server does not require authentication.
## Description
The VNC server installed on the remote host allows an attacker to connect to the remote host as no
authentication is required to access this service.
** The VNC server sometimes sends the connected user to the XDM login
** screen. Unfortunately, Nessus cannot identify this situation.
** In such a case, it is not possible to go further without valid
** credentials and this alert may be ignored.
## Solution
Disable the No Authentication security type.
## Risk Factor
High

## Image Deatils
* OpenBMC release version : 2.8
* libvncserver version : 0.9.12
* Kernel version : 5.4.53

**************************************************************************************************************

* As per the solution mentioned by nessus  report, to disable the no authentication security type and enable vnc authentication security type, like a secure VNC support.
* To achive this what kind of code changes to be done inside libvncserver package ? 
* Or is it already fixed with latest changes ? 

* Do you have any inputs on this @geissonator @edtanous ?

edtanous, 2021-02-07:
First of all, in the future, if you think this is a security concern, you should use the security reporting process documented for the project, and allow the security response team to respond rather than publishing a possible zero-day bug in a public forum.  This allows us time to come up with a response, test patches, and fix the issues before the vulnerability is made public.

With that said, this does not appear to be of any concern at a system level.  Yes, obmc-ikvm server does not require authentication, but the existing code only allows binding to localhost, so the server itself is not exposed outside of the BMC, thus not breaking the authentication envelope;  The explicit lack of authentication is intentional, to avoid inadvertently advertising or implying that the RFB protocol authentication is secure in any way, which it isn't, as RFB has no secure transport (in the base spec), and passwords are truncated and hashed using known broken hashes.

Code that binds to localhost only.
https://github.com/openbmc/obmc-ikvm/blob/861337e8ec92767c4c88237ec5db494a2a67fa8d/ikvm_server.cpp#L47

It should be noted, in the future, when filing bugs, please don't mention "the vulnerability scan report", as a justification for a bug without mentioning the scanner name, or publishing the report.  Security scanners have bugs too, and doing your own analysis and filtering of the report in this case, would've led you to the above conclusion without needing to open the bug.

To those that have the right permissions, this bug can be closed.

geissonator, 2021-02-08:
> To those that have the right permissions, this bug can be closed.

I pinged Brad on this. He's got it on hist list to create a triage group within github to get more maintainers with permission to close out bugs.

closed 2021-02-08 by geissonator