openbmc/openbmc #3754: Failed to read CPU and DIMM temperature via PEIC

JohnChung93, 2021-01-18:
**Note on Bugs**
Sometimes failed to read  CPU and DIMM temperature after host power reset.
But it still could read CPU and DIMM temperature via ME.
Restart PECI driver could read CPU and DIMM temperature normally.

**Unexpected behavior you saw**
```
root@:~# peci_cmds -a 0x30 GetTemp
ERROR 6: Retrieving temperature failed
root@:~# peci_cmds -a 0x31 GetTemp
ERROR 6: Retrieving temperature failed
root@:~# peci_cmds -a 0x31 RdPkgConfig 14 0x0005
ERROR 6: command failed
   cc:0x00
root@:/sys/class/peci_dev/peci-0/device/0-30/peci-cputemp.0/hwmon/hwmon5# cat temp15_input
cat: read error: Resource temporarily unavailable
[250323.607270] peci peci-0: peci_command, cmd=0x04
[250323.667496] peci peci-0: xfer error: -110
[250324.160675] peci peci-0: peci_command, cmd=0x01
[250324.211751] peci peci-0: xfer error: -110
```

**Expected behavior**
It still could read CPU and DIMM temperature via ME.
```
# ipmitool -t 0x2c -b 1 -I lanplus raw 0x2e 0x40 0x57 0x01 0x00 0x30 0x01 0x02 0x01
 57 01 00 ec f1
# ipmitool -t 0x2c -b 6 -I lanplus raw  0x2e 0x42 0x57 0x01 0x00 0x00 0xe 0x03 0x00 0x03
 57 01 00 20 00 00 00
# ipmitool -t 0x2c -b 6 -I lanplus raw  0x2e 0x42 0x57 0x01 0x00 0x00 0xe 0x04 0x00 0x03
 57 01 00 1f 00 00 00
```

**Additional context**
Using following step for reload PECI driver.

```
root@:/sys/bus/platform/drivers/peci_cputemp# echo peci-cputemp.0 > unbind
root@:/sys/bus/platform/drivers/peci_cputemp# echo peci-cputemp.1 > unbind
root@:/sys/bus/platform/drivers/peci_cputemp# echo peci-cputemp.0 > unbind
root@:/sys/bus/platform/drivers/peci_cputemp# echo peci-cputemp.1 > unbind
root@:/sys/bus/platform/drivers/peci_aspeed# echo 1e78b000.peci-bus > unbind
root@:/sys/bus/platform/drivers/peci_aspeed# echo 1e78b000.peci-bus > bind
```

```
250344.152228] peci peci-0: peci_command, cmd=0x01
[250344.211399] peci peci-0: xfer error: -110
[250344.211652] peci peci-0: peci_command, cmd=0x01
[250344.265617] peci peci-0: xfer error: -110
[250345.019330] peci peci-0: Removing peci-client at 0x30
[250345.020007] peci 0-30: peci_client_dev_release: peci-client
[250345.020043] peci peci-0: Removing peci-client at 0x31
[250345.020661] peci 0-31: peci_client_dev_release: peci-client
[250345.020699] peci peci-0: adapter [1e78b000.peci-bus] unregistered
[250345.037421] peci peci-0: cdev of adapter [1e78b000.peci-bus] unregistered
[250345.044646] peci peci-0: peci_adapter_dev_release: 1e78b000.peci-bus
[250348.455558] peci peci-0: cdev of adapter [1e78b000.peci-bus] registered as minor 0
[250348.463678] peci peci-0: adapter [1e78b000.peci-bus] registered
[250348.463717] peci_aspeed 1e78b000.peci-bus: peci bus 0 registered, irq 36
```

seirespcp, 2021-01-18:
> **Note on Bugs**
> Sometimes failed to read CPU and DIMM temperature after host power reset.
> But it still could read CPU and DIMM temperature via ME.
> Restart PECI driver could read CPU and DIMM temperature normally.
Since power-control is likely platform-dependent,
where would you like the solution to go to?

> ```
> root@:/sys/bus/platform/drivers/peci_cputemp# echo peci-cputemp.0 > unbind
> root@:/sys/bus/platform/drivers/peci_cputemp# echo peci-cputemp.1 > unbind [...]
You have unbind twice, but I assume it's just a typo.
Is rebinding the whole bus required or are you doing it to be safe?

JohnChung93, 2021-01-18:
> Since power-control is likely platform-dependent,
> where would you like the solution to go to?
I'm not sure is this a PECI kernel driver issue or something else.
When the issue occurred it could recover without any host and BMC reset only need to reload PECI driver.

> You have unbind twice, but I assume it's just a typo.
> Is rebinding the whole bus required or are you doing it to be safe?
It's a typo. Already correct it. Thanks.
Doing it to be safe.

williamspatrick, 2021-01-18:
Can you let us know what openbmc commit you are running? Which kernel version? Which machine?

I didn’t think our kernel even had the PECI anymore because the author hasn’t completed upstreaming them, but I could be confusing with some other driver.

JohnChung93, 2021-01-19:
> Can you let us know what openbmc commit you are running? Which kernel version? Which machine?

Using openbmc commit : 5457b9264db224ed39215ad425e270d44db37ed2
Using kernel version:  Linux version 5.4.23-6ba7086-dirty-dd054e6
Using machine is our modified.

Using following setting for PECI in kernel.

```
CONFIG_PECI=y
CONFIG_PECI_ASPEED=y
CONFIG_PECI_CHARDEV=y
CONFIG_SENSORS_PECI_CPUTEMP=y
CONFIG_SENSORS_PECI_DIMMTEMP=y
CONFIG_MFD_INTEL_PECI_CLIENT=y
```

edtanous, 2021-02-07:
which machine?

JohnChung93, 2021-02-08:
Using our downstream machine.
Base on ast2500.

williamspatrick, 2021-02-08:
You’re running code that is almost a year old, with a kernel that is multiple versions behind, on a machine that isn’t open sourced. I don’t think there are many people that are going to be able to help you out. 

PECI is an Intel interface and one of their developers wrote the PECI driver. Do you have a support channel through them?

JohnChung93, 2021-02-08:
> PECI is an Intel interface and one of their developers wrote the PECI driver. Do you have a support channel through them?

Yes, I have a support channel through them.
I just want to make sure is there have another machine occurred the same issue and is there have a easy way to reproduce this issue.
Cause this PECI driver issue is hardly to reproducing in our system.
 
I would try to use the latest kernel for fixing this issue. 
Thank you for your help.

edtanous, 2021-02-09:
Please don't file bugs for "downstream" machines.  We do not have your code, so we have no ability to debug any issues for you that might occur.  If you want to file this type of bug in the future, please submit your machine to gerrit.

JohnChung93, 2021-02-12:
Okey. I'm sorry about that.
Thank you for your help.

closed 2021-02-12 by JohnChung93

rdyvptnm, 2023-09-30:
I am having similar issue when the HOST/CPU is turned off and turn it on back. Does any body know how the peci driver/clinet/hwmon drivers are notifed when the HOST/CPU turns on ?

mdmillerii, 2023-09-30:
Phosphor State manager starts and stops targets which can cause unit files to execute and or start and stop arbitrary commands.