openbmc/openbmc #3756: Use OpenSSH by default and drop dropbear

joseph-reynolds, 2021-01-20:
This issue is to change the OpenBMC project default build configuration to use the OpenSSH server instead of the dropbear SSH server.  Reasons:
- OpenSSH has additional security focus (bug fixes) and more up-to-date ciphers.
- OpenSSH implements the "expired password dialog" so an SSH client can change an expired password as part of logging into the BMC via SSH.

Note also:
- OpenSSH has a configuration file, which allows dynamic configuration (which can be good if you need to disable specific vulnerable cipers, or bad if an attacker can reenable vulnerable ciphers).
- OpenSSH may be larger (static readonly file system size).
- Yocto has good support for both OpenSSH and for dropbear.

This issues was also mentioned here https://github.com/ibm-openbmc/dev/issues/947 and in the 2020-01-20 OpenBMC security working group meeting: https://docs.google.com/document/d/1b7x9BaxsfcukQDqbvZsU2ehMq4xoJRQvLxxsDUWmAOI

geissonator, 2021-08-02:
Still a valid feature to try for.

leeeening, 2023-12-27:
I would like to ask, can OpenBMC now default to using OpenSSH and discard dropbear？
I would like to use openssh in openbmc and disable dropbear. If there is a way, please provide the configuration method. Thank you.

leeeening, 2024-01-11:
I would like to ask how to add sshd to Systemd and start it after successfully installing OpenSSH. Thank you.

zviratko, 2024-10-09:
Not a user yet, but seeing Dropbear in there instead of OpenSSH is pretty sad.

leeeening, 2024-10-09:
邮件收到，谢谢

zviratko, 2024-10-09:
Because OpenSSH is much more up-to-date and supported than Dropbear. Dropbear AFAIK doesn't support certificates, no hw acceleration, SFTP... at least I see that they added SK key support in 2022, but that one made it unusable for me a few years ago while OpenSSH supported was there. Unless resources are really tight, then I see no reason to use Dropbear. I consider server management much more important than an office wireless router, which is what Dropbear was meant for...

williamspatrick, 2024-10-09:
> Unless resources are really tight,

FWIW, I made the changes to enable OpenSSH to be used and use it on all the platforms I deploy.

There were/are originally two reasons we used Dropbox:
 
1. Early systems OpenBMC supported only had 32MB of storage and OpenSSH wasn't fitting.
2. There is a ["host serial console redirect over SSH" feature](https://github.com/openbmc/obmc-console/blob/366651d9ef0e03f97b1e1d2b6188f2b452044d1c/conf/obmc-console-ssh@.service.in#L12) that currently only works using dropbear that some people rely on.  I'm not positive that feature can be made to work with OpenSSH.

zviratko, 2024-10-09:
I'm not sure what "-c" argument does, but I'm pretty sure something like ForceCommand will work? Unless it somehow delegates password login to the console, but even that would be doable (though pretty horrible from security standpoint)

zviratko, 2024-10-09:
Probably running "sshd -f otherportconfigfile.conf" with the ForceCommand directive.
But yeah, I understand. I wouldn't expect constrains such as "32MB of storage" in this day and age (and in server hardware), seems like we're stuck in the past (OpenBMC doesn't require Java 6, right? :))

williamspatrick, 2024-10-09:
> But yeah, I understand. I wouldn't expect constrains such as "32MB of storage" in this day and age (and in server hardware)

Perhaps surprising but many "modern" server designs are still using 64MB NOR flash chips for their BMC because going to 128MB would have added $1 or more to their board costs. 🙄 So, unfortunately, flash size constraints are still very real for this project.

amboar, 2024-10-09:
> I consider server management much more important than an office wireless router, which is what Dropbear was meant for...

In addition to the flash constraints, the AST2400 (which is what OpenBMC was originally developed on) contains a single 400MHz ARM926EJ (ARMv5). Despite the importance of server management, there aren't a lot of cycles to spare compared to more modern SoCs.

mdmillerii, 2024-10-09:
I started drafting before other replies and have responded to many comments.

`-c` is a dropbear option to use _forcecommand_  for this instance https://github.com/mkj/dropbear/blob/cdfda00b92d2f4a6edd389e1d55af4dda4fc5a01/manpages/dropbear.8#L114 and is combined with socket activation support direct console access while using pam authentication.

> Also, remove the "ssh" feature from obmc-console.

> I'm not even saying "it can't be done" but it is not trivial work and nobody seems interested in doing it / contributing it.

If you  come up with an incremental openssh configuration (that doesn't involve a whole configuration file for each copy) , you can contribute  a package config option to create units  invoking the console for each socket.  Right now the service just copies the in file https://github.com/openbmc/obmc-console/blob/366651d9ef0e03f97b1e1d2b6188f2b452044d1c/meson.build#L15 leaving more complex configuration to the platforms

As this project is open source you are free to make additional changes to suit your needs, and contributions are accepted especially when they enhance security, subject to not breaking the existing users https://github.com/openbmc/docs/blob/master/CONTRIBUTING.md

>> I wouldn't expect constrains such as "32MB of storage" in this day and age (and in server hardware), 

In addition to 32MB SPI nor chips (which may be soldered for cost), the aspeed 2400/2500/2600  (2700 is just sampling not shipping in products yet afaik) support 512MB/1GB/2GB of RAM (one ddr 2/3/4).   Since this is larger than the flash, we stage the flash image in memory and verify signature before writing.   However you can run from RAM too .

> AST2400 400MHz ARM926EJ (ARMv5)

if you don't have the resources to change the hardware to meet your requirements (be it sourcing a larger SPI, multiple SPI chips for space or backup image or A/B update, adding eMMC storage, etc) you can consider installing additional infrastructure to support network boot.  

I developed the static layout on a 2400 with 32MB SPI nor and 512MB RAM using netboot.   The support to tftp the image and have it signature check before use is built in, and also to fetch but not currently check vis http (`https`  requires additional initrd binaries beyond the BusyBox `wget`).

When sharing a NIC with the host, the NC-SI link is 100   Mb/s.  Dedicated porta will likely support 1Gb  max (if they didn't cheap out when selecting the phy).   That's a wire max of 97 and 970MB/s if the CPU and NIC can keep up.   NC SI might involve a packet copy in the nic with a mcu before the BMC nic).

USB ports are often USB 2 (480Mb) if present and one exposed.  

Others  have demonstrated pxe tftp boot if the BMC (with signature check) and iSCSI and RPM update either from hosting storage or BMC  (even integrating netboot server into a Top if Rack router).   Others have shown using dm verity to securely check block images.   I was involved in several storage layouts , and enjoy discussing options in this space.

> seems like we're stuck in the past (OpenBMC doesn't require Java 6, right? :))

No Java has been in in the upstream repository collection to my memory.

williamspatrick, 2024-10-09:
https://stackoverflow.com/questions/38889600/starting-different-applications-from-ssh-by-using-different-ports

Someone could probably teach obmc-console to supply these `Port`, `Match`, and `ForceCommand` directives as `ssh.d` config snippets.

mdmillerii, 2024-10-09:
Except we go from port to console and then create socket activation of the port.  We create a configuration for each needed port.  We also created udev activation of TTY to start the servers.

https://github.com/openbmc/openbmc/tree/master/meta-ibm/recipes-phosphor/console/obmc-console/p10bmc

Maybe a list of console IDs with port and tty like fstab or crypttab https://www.freedesktop.org/software/systemd/man/latest/crypttab.html and create a dbua generator to create the service files https://www.freedesktop.org/software/systemd/man/latest/systemd.environment-generator.html#