openbmc/openbmc #3760: NETDEV WATCHDOG: eth0 (ftgmac100): transmit queue 0 timed out

usrinivasan2407, 2021-02-04:
I am building openbmc with the latest from git and I see the following.  What is the issue and how do I fix it? Thanks in advance.

 [   78.086436] ------------[ cut here ]------------
[   78.091126] WARNING: CPU: 0 PID: 330 at net/sched/sch_generic.c:442 dev_watchdog+0x26c/0x28c
[   78.099682] NETDEV WATCHDOG: eth0 (ftgmac100): transmit queue 0 timed out
[   78.106560] Modules linked in:
[   78.109653] CPU: 0 PID: 330 Comm: systemd-coredum Not tainted 5.8.17-3cc95ae #1
[   78.117033] Hardware name: Generic DT based system
[   78.121832] Backtrace:
[   78.124335] [<80107248>] (dump_backtrace) from [<801074ec>] (show_stack+0x20/0x24)
[   78.131994]  r7:000001ba r6:806628ac r5:00000009 r4:808f2d64
[   78.137772] [<801074cc>] (show_stack) from [<80409f88>] (dump_stack+0x28/0x30)
[   78.145028] [<80409f60>] (dump_stack) from [<80117154>] (__warn+0xd0/0x104)
[   78.152071]  r5:00000009 r4:8095cd08
[   78.155684] [<80117084>] (__warn) from [<8011754c>] (warn_slowpath_fmt+0xa0/0xc8)
[   78.163251]  r7:806628ac r6:000001ba r5:8095cd08 r4:8095cccc
[   78.169030] [<801174b0>] (warn_slowpath_fmt) from [<806628ac>] (dev_watchdog+0x26c/0x28c)
[   78.177299]  r8:980d3000 r7:80b11880 r6:00000000 r5:980ca600 r4:980d3234
[   78.184047] [<80662640>] (dev_watchdog) from [<80163350>] (call_timer_fn+0x40/0x128)
[   78.191880]  r8:ffffa950 r7:00000100 r6:80662640 r5:980d3234 r4:980d3234
[   78.198682] [<80163310>] (call_timer_fn) from [<80163768>] (__run_timers.part.0+0x184/0x24c)
[   78.207235]  r8:ffffa950 r7:00000000 r6:80b129e0 r5:99045784 r4:980d3234
[   78.213968] [<801635e4>] (__run_timers.part.0) from [<80163890>] (run_timer_softirq+0x44/0x7c)
[   78.222670]  r10:00000030 r9:00000002 r8:00000002 r7:00000004 r6:99044000 r5:80b11880
[   78.230577]  r4:80b129e0
[   78.233151] [<8016384c>] (run_timer_softirq) from [<80101314>] (__do_softirq+0xdc/0x310)
[   78.241317]  r5:80b706e8 r4:00000001
[   78.244932] [<80101238>] (__do_softirq) from [<8011b944>] (irq_exit+0xc4/0x128)
[   78.252332]  r10:00000030 r9:99044000 r8:9e415600 r7:00000001 r6:00000000 r5:80b344a0
[   78.260233]  r4:00000000
[   78.262807] [<8011b880>] (irq_exit) from [<80154db0>] (__handle_domain_irq+0x80/0xd4)
[   78.270734] [<80154d30>] (__handle_domain_irq) from [<80101230>] (avic_handle_irq+0x68/0x70)
[   78.279265]  r9:99044000 r8:00000001 r7:990458bc r6:ffffffff r5:99045888 r4:9e404320
[   78.287101] [<801011c8>] (avic_handle_irq) from [<80100acc>] (__irq_svc+0x6c/0x90)
[   78.294678] Exception stack(0x99045888 to 0x990458d0)
[   78.299807] 5880:                   9804ac20 00000005 980b6020 00000001 9804ac20 00000600
[   78.308065] 58a0: 980b6020 9804ac48 00000001 80b11880 00000030 99045924 990458f8 990458dc
[   78.316250] 58c0: 80512428 805112cc 60000013 ffffffff
[   78.321379]  r5:60000013 r4:805112cc
[   78.324996] [<805123f0>] (aspeed_smc_read_reg) from [<8050b56c>] (spi_nor_read_sr+0xd4/0xf0)
[   78.333518]  r8:ffffb8ef r7:980b6020 r6:00000000 r5:9804ac48 r4:805123f0
[   78.340319] [<8050b498>] (spi_nor_read_sr) from [<8050df48>] (spi_nor_wait_till_ready_with_timeout+0xa4/0x39c)
[   78.350403]  r6:00000000 r5:fffff060 r4:9804ac48
[   78.355050] [<8050dea4>] (spi_nor_wait_till_ready_with_timeout) from [<8050ee28>] (spi_nor_write+0x110/0x24c)
[   78.365048]  r10:00000030 r9:99045ab4 r8:00000030 r7:01c3d310 r6:00000000 r5:9804ac48
[   78.372953]  r4:00000030
[   78.375531] [<8050ed18>] (spi_nor_write) from [<8050107c>] (mtd_write_oob_std+0xa8/0xb0)
[   78.383708]  r10:00000030 r9:00000001 r8:0003d310 r7:00000000 r6:99045aac r5:99045ab4
[   78.391610]  r4:00000000
[   78.394190] [<80500fd4>] (mtd_write_oob_std) from [<8050345c>] (mtd_write_oob+0x98/0xe8)
[   78.402359]  r5:980b2400 r4:9804ac48
[   78.405973] [<805033c4>] (mtd_write_oob) from [<80503680>] (mtd_writev+0x138/0x1a0)
[   78.413715]  r8:00000002 r7:00000000 r6:99045b6c r5:0003d310 r4:985e6000
[   78.420545] [<80503548>] (mtd_writev) from [<803233b0>] (jffs2_flash_direct_writev+0x54/0x5c)
[   78.429168]  r10:00000074 r9:98b407c8 r8:00000000 r7:99045b64 r6:97f11400 r5:0003d2cc
[   78.437076]  r4:00000002
[   78.439652] [<8032335c>] (jffs2_flash_direct_writev) from [<8031b88c>] (jffs2_write_dnode+0x110/0x438)
[   78.449045]  r8:00000030 r7:97f11400 r6:0003d2cc r5:9904f830 r4:98e19740
[   78.455776] [<8031b77c>] (jffs2_write_dnode) from [<8031c39c>] (jffs2_write_inode_range+0x364/0x424)
[   78.465017]  r10:000bc000 r9:97f11400 r8:8fc32000 r7:00000030 r6:98b407c8 r5:985e6000
[   78.472933]  r4:98e19740
[   78.475501] [<8031c038>] (jffs2_write_inode_range) from [<803166e0>] (jffs2_write_end+0x1dc/0x3f0)
[   78.484546]  r10:00001000 r9:97f11400 r8:00000000 r7:00000000 r6:9ee02640 r5:98b407f0
[   78.492450]  r4:98e19740
[   78.495036] [<80316504>] (jffs2_write_end) from [<80222448>] (generic_perform_write+0x140/0x1f8)
[   78.503908]  r10:00000000 r9:80816aa0 r8:00001000 r7:98b408e0 r6:99045e30 r5:000bc000
[   78.511809]  r4:80316504
[   78.514388] [<80222308>] (generic_perform_write) from [<802248d4>] (__generic_file_write_iter+0x174/0x214)
[   78.524125]  r10:99045e30 r9:99045e30 r8:9904caa0 r7:98b408e0 r6:00000000 r5:99045d60
[   78.532027]  r4:00000000
[   78.534595] [<80224760>] (__generic_file_write_iter) from [<80224a8c>] (generic_file_write_iter+0x118/0x180)
[   78.544508]  r10:00000000 r9:99045e30 r8:00000000 r7:98b4086c r6:99045e30 r5:99045d60
[   78.552409]  r4:00010000
[   78.554989] [<80224974>] (generic_file_write_iter) from [<8027916c>] (do_iter_readv_writev+0x164/0x1d8)
[   78.564467]  r7:99045e58 r6:00000000 r5:00000000 r4:9904caa0
[   78.570239] [<80279008>] (do_iter_readv_writev) from [<8027af3c>] (do_iter_write+0x94/0x1a8)
[   78.578768]  r10:98fa3600 r9:98dc02c0 r8:99045e58 r7:00000000 r6:9904caa0 r5:99045e30
[   78.586671]  r4:00000000
[   78.589250] [<8027aea8>] (do_iter_write) from [<8027b078>] (vfs_iter_write+0x28/0x34)
[   78.597166]  r9:98dc02c0 r8:000000c0 r7:7ff41fff r6:9905f0d4 r5:00000010 r4:800bf001
[   78.604943] [<8027b050>] (vfs_iter_write) from [<802b221c>] (iter_file_splice_write+0x28c/0x3f8)
[   78.613815] [<802b1f90>] (iter_file_splice_write) from [<8036b6c0>] (ovl_splice_write+0x9c/0xdc)
[   78.622691]  r10:00000000 r9:98f9d880 r8:99045f08 r7:7ff50fff r6:98dc02c0 r5:9904c280
[   78.630591]  r4:00000000
[   78.633156] [<8036b624>] (ovl_splice_write) from [<802b2f98>] (do_splice+0x364/0x7d4)
[   78.641071]  r9:7ff50fff r8:9900a2a0 r7:98dc02c0 r6:00000000 r5:8036b624 r4:9904c280
[   78.648907] [<802b2c34>] (do_splice) from [<802b34a8>] (sys_splice+0x84/0xd4)
[   78.656060]  r10:9900a2a0 r9:9904c280 r8:00000000 r7:00000000 r6:9904c280 r5:9900a2a0
[   78.663964]  r4:7ff50fff
[   78.666599] [<802b3424>] (sys_splice) from [<80100040>] (ret_fast_syscall+0x0/0x54)
[   78.674258] Exception stack(0x99045fa8 to 0x99045ff0)
[   78.679390] 5fa0:                   7ff50fff 00000000 00000005 00000000 00000006 00000000
[   78.687649] 5fc0: 7ff50fff 00000000 76f4c550 00000154 00000000 00010000 00000000 7eb48aa4
[   78.695831] 5fe0: 00000000 7eb48a20 76dda684 76c58780
[   78.700962]  r10:00000000 r9:99044000 r8:80100224 r7:00000154 r6:76f4c550 r5:00000000
[   78.708860]  r4:7ff50fff
[   78.711412] ---[ end trace c5ce2972c8208f1f ]---

jk-ozlabs, 2021-02-05:
This may be occurring due to invalid PHY configuration of your interfaces. Which platform is this on? If it's a new one, are you sure that the device tree correctly represents the network device configuration (particularly, if the devices are using NCSI).

And can you include the whole dmesg log?

usrinivasan2407, 2021-02-05:
I am running on ast2500 on a Portwell C660-ZR1 and the SPI image that Portwell shipped with booted this (I  have their schematics but no source):
U-Boot 2016.07 fbtp-2168080 (Dec 15 2017 - 14:34:41 -0800)
DRAM:  434 MiB
Watchdog: 300s
to_pwr_on: 0, policy:null, ls:null, scu3c:00000014
Flash:  FMC-CS0: flash ID: 1920c2
        FMC-CS1: flash ID: 0
        FMC-CS1: Unknown flash ID!
        FMC-CS1: Skip unsupported flash!
32 MiB
Using default environment
In:    serial
Out:   serial
Err:   serial
*****
cat /proc/cpuinfo
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 789.70
Features        : half thumb fastmult edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7
Hardware        : AST-G5
Revision        : 0000
Serial          : 0000000000000000
****
[   18.290000] libphy: ftgmac100_mdio: probed
[   18.300000] ftgmac100 ftgmac100.0 eth0: irq 2, mapped at dca40000
[   18.310000] ftgmac100 ftgmac100.0 eth0: generated random MAC address ba:25:66:27:94:bf
****
eth0 mac fixup
ethaddr= 6A:C3:EB:11:37:6D
use u-boot mac addr

I am trying to build my own image with i2c devices and I have tried several configurations swapping between phy/ncsi.  So if you can answer some key questions I'd really appreciate it:

1)   I see three choices for uboot machine which is correct for aspeed 2500 ast-g5
#UBOOT_MACHINE = "ast_g5_ncsi_config"
#UBOOT_MACHINE = "ast_g5_phy_defconfig"
#UBOOT_MACHINE = "ast_g5_phy_config"
BTW, some layers use phy_config and others phy_defconfig, which is correct?

2) which dst file do you suggest I use as my base (to which I can add my i2c devices)?
aspeed-ast2500-evb.dts
ast2500-evb.dts
something else?

3) I successfully booted many images but am now switching to uImage so I can use flashcp instead of having to reburn my spi chip after each build.  I added KERNEL_IMAGETYPE = "uImage" but my bitbake fails with the following & I don't know the correct way to fix this. 
  Kernel: arch/arm/boot/Image is ready
  Kernel: arch/arm/boot/zImage is ready
multiple (or no) load addresses:
This is incompatible with uImages
Specify LOADADDR on the commandline to build an uImage

Thanks in advance!

williamspatrick, 2021-02-05:
`U-Boot 2016.07 fbtp-2168080`

I suspect this is a Facebook/openbmc image and not an image from this repository.  fbtp would be “Facebook Tiogapass” but the Portwell design isn’t the same as the Tiogapass design, so I’m a little surprised to see them shipping that image.

Can you get the contents of /etc/issue and/or /etc/os-release?  That will confirm which repository it is built from and maybe a git tag. 

@amithash contributed some support for Portwell here but I don’t think that work is complete.

usrinivasan2407, 2021-02-05:
Portwell told me the image they shipped with was a facebook openbmc.  I will have to boot back to their image to get the info.  

ASPEED doc says "AST2500 supports 2 Ethernet MAC interfaces. Each can be configured as 1G(RGMII) or 10/100M(RMII/NCSI) speed mode independently."  I am happy with RGMII.  So, should I use "ast_g5_phy_defconfig" or "ast_g5_phy_config"?   

And, do you have a suggestion for my uImage build issue. Thanks.

usrinivasan2407, 2021-02-05:
Here is the info from the original image:
root@bmc-oob:~# cat /etc/os-release
ID="poky"
NAME="Facebook OpenBMC"
VERSION="0.4 (krogoth)"
VERSION_ID="0.4"
PRETTY_NAME="Facebook OpenBMC 0.4 (krogoth)"

root@bmc-oob:~# cat /etc/issue
OpenBMC Release fbtp-2168080

root@bmc-oob:~# dmesg|grep mac
[   18.290000] libphy: ftgmac100_mdio: probed
[   18.300000] ftgmac100 ftgmac100.0 eth0: irq 2, mapped at dca40000
[   18.310000] ftgmac100 ftgmac100.0 eth0: generated random MAC address 2a:06:4f:08:61:2c

root@bmc-oob:~# cat /proc/cpuinfo
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 789.70
Features        : half thumb fastmult edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : AST-G5
Revision        : 0000
Serial          : 0000000000000000

amithash, 2021-02-05:
I checked the code of fb-openbmc for portwell, it is using the random MAC and has NC-SI disabled.
The only thing in there is a user space service which takes that random MAC and stores it in u-boot env (Hence next boot up it will reuse the first generated random MAC) -- This is more or less to ensure that we do not have the IP randomly switch up on each boot.

amithash, 2021-02-05:
https://github.com/openbmc/openbmc/blob/master/meta-portwell/meta-neptune/conf/machine/neptune.conf#L4 is probably not the right one.

usrinivasan2407, 2021-02-05:
Yes, thx.  Instead of ast_g5_ncsi_config I am going with ast_g5_phy_defconfig.  

Any idea how to fix the uImage LOADADDR issue I'm running into with KERNEL_IMAGETYP = "uImage"?

jk-ozlabs, 2021-02-06:
OK, sounds like you're using a hardware PHY for that interface; you'll need to make sure that the device tree isn't specifying NCSI there (is there a `use-ncsi` property on the ftgmac node?).

usrinivasan2407, 2021-02-11:
Hi, thanks everyone for your responses.  Yes, I made sure use-ncsi is not in my dts file & that I'm using a phy config:
UBOOT_MACHINE = "ast_g5_phy_defconfig"
Things to note: 
1) the watchdog trace is for eth1 
2) I don't have either eth interfaces connected as of yet, yet eth1 I see LinkUp message
4) I'm not sure where eth1's static IP is coming from; what is the correct method to assigning a static IP to eth0 for when I connect it?
Thanks in advance!!

*********************U-Boot msgs:
Net:   MAC0 : RGMII
MAC1 : RMII/NCSI
FTGMAC100#0
Error: FTGMAC100#0 address not set.
, FTGMAC100#1
Error: FTGMAC100#1 address not set.

*********************Kernel early msgs:
[    1.922088] libphy: Fixed MDIO Bus: probed
[    1.927736] ftgmac100 1e660000.ethernet: Generated random MAC address 7e:6e:15:29:d4:14
[    1.968641] libphy: ftgmac100_mdio: probed
[    1.973731] RTL8211E Gigabit Ethernet 1e660000.ethernet--1:00: attached PHY driver [RTL8211E Gigabit Ethernet] (mii_bus:phy_addr=1e660000.ethernet--1:00, irq=POLL)
[    1.989518] ftgmac100 1e660000.ethernet eth0: irq 20, mapped at (ptrval)
[    1.996672] ftgmac100 1e680000.ethernet: Error applying setting, reverse things back
[    2.004790] ftgmac100 1e680000.ethernet: Generated random MAC address d6:12:33:65:37:70
[    2.418918] libphy: ftgmac100_mdio: probed
[    2.423495] Generic PHY 1e680000.ethernet--1:00: attached PHY driver [Generic PHY] (mii_bus:phy_addr=1e680000.ethernet--1:00, irq=POLL)
[    2.436864] ftgmac100 1e680000.ethernet eth1: irq 21, mapped at (ptrval)

*********************dmesg & trace msgs:
[   15.052904] systemd[1]: Starting Create Static Device Nodes in /dev...
[   15.213160] systemd[1]: Mounted Kernel Configuration File System.
[   15.532672] systemd[1]: Started Journal Service.
[   21.857412] random: crng init done
[   21.860850] random: 7 urandom warning(s) missed due to ratelimiting
[   24.381972] ftgmac100 1e680000.ethernet eth1: Link is Up - 100Mbps/Full - flow control rx/tx
[   24.431177] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   55.426985] i2c i2c-5: new_device: Instantiated device ipmb-dev at 0x10
[   57.070512] i2c i2c-5: Failed to register i2c client ipmb-dev at 0x10 (-16)
[ 1662.087323] ------------[ cut here ]------------
[ 1662.092020] WARNING: CPU: 0 PID: 745 at net/sched/sch_generic.c:442 dev_watchdog+0x26c/0x28c
[ 1662.100573] NETDEV WATCHDOG: eth1 (ftgmac100): transmit queue 0 timed out
[ 1662.107452] Modules linked in:
[ 1662.110540] CPU: 0 PID: 745 Comm: systemd-coredum Not tainted 5.8.17-3cc95ae #1
[ 1662.117920] Hardware name: Generic DT based system
[ 1662.122719] Backtrace:
[ 1662.125220] [<80107248>] (dump_backtrace) from [<801074ec>] (show_stack+0x20/0x24)
[ 1662.132882]  r7:000001ba r6:806628ac r5:00000009 r4:808f2d84
[ 1662.138659] [<801074cc>] (show_stack) from [<80409f88>] (dump_stack+0x28/0x30)
[ 1662.145914] [<80409f60>] (dump_stack) from [<80117154>] (__warn+0xd0/0x104)
[ 1662.152967]  r5:00000009 r4:8095cd28
[ 1662.156578] [<80117084>] (__warn) from [<8011754c>] (warn_slowpath_fmt+0xa0/0xc8)
[ 1662.164147]  r7:806628ac r6:000001ba r5:8095cd28 r4:8095ccec
[ 1662.169920] [<801174b0>] (warn_slowpath_fmt) from [<806628ac>] (dev_watchdog+0x26c/0x28c)
[ 1662.178191]  r8:980e4800 r7:80b11880 r6:00000000 r5:980cae80 r4:980e4a34
[ 1662.184927] [<80662640>] (dev_watchdog) from [<80163350>] (call_timer_fn+0x40/0x128)
[ 1662.192794]  r8:00021410 r7:00000100 r6:80662640 r5:980e4a34 r4:980e4a34
[ 1662.199603] [<80163310>] (call_timer_fn) from [<80163768>] (__run_timers.part.0+0x184/0x24c)
[ 1662.208129]  r8:00021410 r7:00000000 r6:80b129e0 r5:9e4adeac r4:980e4a34
[ 1662.214859] [<801635e4>] (__run_timers.part.0) from [<80163890>] (run_timer_softirq+0x44/0x7c)
[ 1662.223558]  r10:00000000 r9:00000002 r8:00000002 r7:00000004 r6:9e4ac000 r5:80b11880
[ 1662.231463]  r4:80b129e0
[ 1662.234036] [<8016384c>] (run_timer_softirq) from [<80101314>] (__do_softirq+0xdc/0x310)
[ 1662.242202]  r5:80b706e8 r4:00000001
[ 1662.245816] [<80101238>] (__do_softirq) from [<8011b944>] (irq_exit+0xc4/0x128)
[ 1662.253209]  r10:00000000 r9:0000002f r8:9e415600 r7:00000001 r6:00000000 r5:80b344a0
[ 1662.261111]  r4:00000000
[ 1662.263685] [<8011b880>] (irq_exit) from [<80154db0>] (__handle_domain_irq+0x80/0xd4)
[ 1662.271616] [<80154d30>] (__handle_domain_irq) from [<80101230>] (avic_handle_irq+0x68/0x70)
[ 1662.280142]  r9:0000002f r8:00c5387d r7:00c5387d r6:ffffffff r5:9e4adfb0 r4:9e404320
[ 1662.287980] [<801011c8>] (avic_handle_irq) from [<80100e94>] (__irq_usr+0x54/0x80)
[ 1662.295555] Exception stack(0x9e4adfb0 to 0x9e4adff8)
[ 1662.300691] dfa0:                                     00000034 00f52841 76401848 00f52840
[ 1662.308953] dfc0: 00000000 00000000 7640187c 7640187b 00000000 0000002f 00000000 00752840
[ 1662.317136] dfe0: 74700200 7ea45698 00000040 76a56dd4 20000010 ffffffff
[ 1662.323825]  r5:20000010 r4:76a56dd4
[ 1662.327477] ---[ end trace ed7a05fd8b7ffd7f ]---

*********************ifconfig snippet
eth0      Link encap:Ethernet  HWaddr 7E:55:E8:F6:D4:9A
          UP BROADCAST MULTICAST  MTU:1500  Metric:1

eth1      Link encap:Ethernet  HWaddr 5A:CA:8F:68:BA:DC
          inet addr:169.254.81.229  Bcast:169.254.255.255  Mask:255.255.0.0
          inet6 addr: fe80::58ca:8fff:fe68:badc/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1

jk-ozlabs, 2021-02-12:
Hi!
>     1. the watchdog trace is for eth1

Yeah, it looks like it's getting the generic PHY probed, whereas your eth0 has found a RTL PHY device. Do you know what the eth1 MAC is connected to, hardware wise? Is there a PHY there at all?

>     2. I don't have either eth interfaces connected as of yet, yet eth1 I see LinkUp message

That may be a result of (1); the link-detection code not matching whatever your hardware provides.
 
>     3. I'm not sure where eth1's static IP is coming from; what is the correct method to assigning a static IP to eth0 for when I connect it?

That's just the zeroconf address. The address doesn't have a static address, nor did it get an address assigned over DHCP, so you ended up with just the zeroconf address assigned.

mdmillerii, 2021-02-12:
Inline response with mm>

-------- Original Message --------
From: Jeremy Kerr 
Date: Thu, February 11, 2021 11:46 PM -0600
To: openbmc/openbmc 
CC: Subscribed 
Subject: [EXTERNAL] Re: [openbmc/openbmc] NETDEV WATCHDOG: eth0 (ftgmac100): transmit queue 0 timed out (#3760)

pre { white-space: pre-wrap }

Hi!

1. the watchdog trace is for eth1 

Yeah, it looks like it's getting the generic PHY probed, whereas your eth0 has found a RTL PHY device. Do you know what the eth1 MAC is connected to, hardware wise? Is there a PHY there at all?

mm> is this a dual phy with both macs?  The phy selection code was weak or missing in the ftgmac driver, at least it was last time I looked.  It looks for the first phy on the mdio bus, possibly for both mac.

2. I don't have either eth interfaces connected as of yet, yet eth1 I see LinkUp message 

That may be a result of (1); the link-detection code not matching whatever your hardware provides.

mm> A generic phy might be up but missing clocks since it won't know phy details.

3. I'm not sure where eth1's static IP is coming from; what is the correct method to assigning a static IP to eth0 for when I connect it? 

That's just the zeroconf address. The address doesn't have a static address, nor did it get an address assigned over DHCP, so you ended up with just the zeroconf address assigned.

ushasrinivasan-cornelisnetworks, 2021-03-03:
Closing this.  With ast_g5_phy_config & ensuring that use_ncsi is not on, I stopped seeing this issue with eth0. It started appearing for eth1, but since I don't need eth1, I disabled it.  For now, I'm good with this.

closed 2021-03-03 by ushasrinivasan-cornelisnetworks