openbmc/openbmc #3761: npcm7xx-bootblock-10.10.16-r0 do_fetch: Checksum mismatch

geissonator, 2021-02-12:
I think CI is still passing because it's using an sstate cache file.  If try a fresh build of the gsj system, you hit this error:

````
WARNING: npcm7xx-bootblock-10.10.16-r0 do_fetch: Checksum mismatch for local file /home/andrewg/bitbake_downloads/LICENSE
Cleaning and trying again.
WARNING: npcm7xx-bootblock-10.10.16-r0 do_fetch: Renaming /home/andrewg/bitbake_downloads/LICENSE to /home/andrewg/bitbake_downloads/LICENSE_bad-checksum_650b869bd8ff2aed59c62bad2a22a821
ERROR: npcm7xx-bootblock-10.10.16-r0 do_fetch: Checksum failure fetching https://raw.githubusercontent.com/Nuvoton-Israel/bootblock/INVALID/LICENSE;name=lic
ERROR: npcm7xx-bootblock-10.10.16-r0 do_fetch: Fetcher failure for URL: 'https://raw.githubusercontent.com/Nuvoton-Israel/bootblock/INVALID/LICENSE;name=lic'. Checksum mismatch!
File: '/home/andrewg/bitbake_downloads/LICENSE' has md5 checksum 650b869bd8ff2aed59c62bad2a22a821 when b234ee4d69f5fce4486a80fdaf4a4263 was expected
If this change is expected (e.g. you have upgraded to a new version without updating the checksums) then you can use these lines within the recipe:
```

I don't see any recent changes to the LICENSE file so not sure what's going on. @BenjaminFair any ideas? Should we just submit a recipe change to the new sha?

geissonator, 2021-02-12:
Put up https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/40574 for a quick fix.

mustafa20192, 2021-02-12:
I had the same issue before, you need to add SRCREV.

SRCREV = "7bfef99f7a0354395519c4975f96d66cdda1fb67"

Into 

meta-nuvoton/ recipes-bsp/images/npcm7xx-bootblock_10.10.16.bb

BenjaminFair, 2021-02-12:
Thanks @mustafa20192 , looks like the missing SRCREV is the issue here. We were fetching a 404 page instead of the actual license file

geissonator, 2021-02-12:
cool, I updated my gerrit commit with this instead:  https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/40574

closed 2021-02-12 by geissonator