openbmc/openbmc #3762: witherspoon-tacoma and rainier fail to build: ERROR: phosphor-software-manager-1.0+gitAUTOINC+42bd99361c-r1 do_package: SYSTEMD_SERVICE_phosphor-software-manager-updater-mmc value obmc-flash-mmc@.service does not exist

BMC-Bruce, 2021-02-18:
**Note on Bugs**
Please note that questions or feature requests should be discussed on the
mailing list or Discord. Please see https://github.com/openbmc/openbmc#contact
for info on how to contact us. Github Issues in openbmc/openbmc are for bugs
only. Please remove this section when opening a bug.

**Unexpected behavior you saw**
witherspoon-tacoma and rainier fail to build with bitbake obmc-phosphor-image
ERROR: phosphor-software-manager-1.0+gitAUTOINC+42bd99361c-r1 do_package: SYSTEMD_SERVICE_phosphor-software-manager-updater-mmc value obmc-flash-mmc@.service do es not exist
ERROR: Logfile of failure stored in: /home/bam/my_git/openbmc/openbmc/build/witherspoon-tacoma/tmp/work/armv7ahf-vfpv4d16-openbmc-linux-gnueabi/phosphor-softwar e-manager/1.0+gitAUTOINC+42bd99361c-r1/temp/log.do_package.15648
ERROR: Task (/home/bam/my_git/openbmc/openbmc/meta-phosphor/recipes-phosphor/flash/phosphor-software-manager_git.bb:do_package) failed with exit code '1'
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

**Expected behavior**
witherspoon-tacoma and rainier to successfully build with bitbake obmc-phosphor-image

**To Reproduce**
I do:
    git clone git@github.com:openbmc/openbmc.git
    cd openbmc
    source setup witherspoon-tacoma (or rainier)
    bitbake obmc-phosphor-image
And I get this: NOTE: Running task 2654 of 5213 (/home/bam/my_git/openbmc/openbmc/meta-openpower/recipes-phosphor/host/op-proc-control_git.bb:do_packagedata)
NOTE: recipe op-proc-control-1.0+gitAUTOINC+99ee5ca9c3-r1: task do_packagedata: Started
NOTE: recipe op-proc-control-1.0+gitAUTOINC+99ee5ca9c3-r1: task do_packagedata: Succeeded
NOTE: recipe phosphor-network-1.0+gitAUTOINC+1b5ec9c536-r1: task do_compile: Succeeded
NOTE: recipe phosphor-certificate-manager-0.1+gitAUTOINC+d2393f231c-r1: task do_configure: Succeeded
NOTE: recipe phosphor-ipmi-host-1.0+gitAUTOINC+da06d15cab-r1: task do_compile: Succeeded
NOTE: Tasks Summary: Attempted 2654 tasks of which 0 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/bam/my_git/openbmc/openbmc/meta-phosphor/recipes-phosphor/flash/phosphor-software-manager_git.bb:do_package
Summary: There were 12 WARNING messages shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

real    190m52.328s
user    0m10.565s
sys     0m1.557s

Done.
Wed Feb 17 16:54:00 PST 2021

ERROR: phosphor-software-manager-1.0+gitAUTOINC+42bd99361c-r1 do_package: SYSTEMD_SERVICE_phosphor-software-manager-updater-mmc value obmc-flash-mmc@.service do es not exist
ERROR: Logfile of failure stored in: /home/bam/my_git/openbmc/openbmc/build/witherspoon-tacoma/tmp/work/armv7ahf-vfpv4d16-openbmc-linux-gnueabi/phosphor-softwar e-manager/1.0+gitAUTOINC+42bd99361c-r1/temp/log.do_package.15648
ERROR: Task (/home/bam/my_git/openbmc/openbmc/meta-phosphor/recipes-phosphor/flash/phosphor-software-manager_git.bb:do_package) failed with exit code '1'
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
Yet when I do the same for romulus I have no issue and get a successful build.
I start with no sstate-cache
**Screenshots**
If applicable, add screenshots to help explain your problem.

**OpenBMC Information:**
Revision: c0263ac8ac029de5e7651a22cc864c4b3aaea5c3
  | Build Artifacts
-- | --
  | ChangesRemove bios configurations from meta-layer & bump (details / githubweb)meta-google: recipes-extended: networking: sslh: Import from gBMC (details / githubweb)meta-google: recipes-google: networking: gbmc-sslh: Import from gBMC (details / githubweb)meta-google: recipes-extended: networking: mstpd: Import from gBMC (details / githubweb)meta-google: recipes-google: nftables: Import from gBMC (details / githubweb)meta-google: nftables-systemd: Fix for missing rules (details / githubweb)ipmi-blob-tool: srcrev bump 92a4a62c69..f9ee95f52b (details / githubweb)bmcweb: srcrev bump ae6595ef53..086d32c5a6 (details / githubweb)peci-pcie: srcrev bump 9fa54b52f8..0b79f3e485 (details / githubweb)x86-power-control: srcrev bump 047bcb569b..7c8d19cbcd (details / githubweb)phosphor-dbus-interfaces: srcrev bump b2bbac7e29..257c17da48 (details / githubweb)phosphor-led-manager: srcrev bump 19c48e17a0..b00e56c26e (details / githubweb)openpower-vpd-parser: srcrev bump 6ee637aa43..640004f136 (details / githubweb)phosphor-fan-presence: srcrev bump 871bfeaeb3..fcb0dbcb14 (details / githubweb)webui-vue: srcrev bump 0736d2afc5..8026797012 (details / githubweb)telemetry: srcrev bump 405c1e4bf8..9f14591205 (details / githubweb)phosphor-ipmi-flash: srcrev bump 99b5455933..7d39eb04ac (details / githubweb)phosphor-power: srcrev bump de0fd44f76..10d9405574 (details / githubweb)phosphor-sel-logger: srcrev bump 7d8a3003fb..dbd77b9200 (details / githubweb)
  | Started by upstream project latest-master build number 170originally caused by:Started by an SCM change
  | Revision: c0263ac8ac029de5e7651a22cc864c4b3aaea5c3refs/remotes/origin/master

**Additional context**
Add any other context about the problem here.

geissonator, 2021-02-18:
The bitbake command needs the DISTRO and MACHINE in the command line. Did you try something like this?

DISTRO=openbmc-openpower MACHINE=witherspoon-tacoma bitbake obmc-phosphor-image
or
DISTRO=openbmc-openpower MACHINE=rainier bitbake obmc-phosphor-image

BMC-Bruce, 2021-02-18:
>             <author@noreply.github.com>
>             fail to build: ERROR: phosphor-software-manager-1.0+gitAUTOINC

No, I have not tried that; I have been just following the instructions here
openbmc/openbmc: OpenBMC Distribution (github.com)  in the README.md

BMC-Bruce, 2021-02-18:
Closing this "issue" as not a defect.

closed 2021-02-18 by BMC-Bruce