openbmc/openbmc #3764: ERROR: obmc-phosphor-image-1.0-r0 do_generate_static: /openbmc/build/tmp/deploy/images/evb-ast2600/u-boot.bin' is too large!

bamontales, 2021-02-24:
I'm having this issue while building OpenBMC using AST2600 config. 

ERROR: obmc-phosphor-image-1.0-r0 do_generate_static:/openbmc/build/tmp/deploy/images/evb-ast2600/u-boot.bin' is too large!
ERROR: Logfile of failure stored in: /openbmc/build/tmp/work/evb_ast2600-openbmc-linux-gnueabi/obmc-phosphor-image/1.0-r0/temp/log.do_generate_static.28171
ERROR: Task (/openbmc/meta-phosphor/recipes-phosphor/images/obmc-phosphor-image.bb:do_generate_static) failed with exit code '1'

Do I miss anything on setting up the OpenBMC?

amboar, 2021-02-24:
Hi @bamontales - it would be helpful to have a few more details, such as which commit of openbmc/openbmc you are building from and the commands you ran to configure and execute the build process.

bamontales, 2021-02-24:
Here is the commit `763cfc35d3f8d5da8a48f4a23b90f82c185c8984` where I build the openbmc.

As for the commands, I just followed the instruction written on the README of the openbmc/openbmc.
`sudo apt-get install -y git build-essential libsdl1.2-dev texinfo gawk chrpath diffstat`
`git clone git@github.com:openbmc/openbmc.git`
`cd openbmc`
`TEMPLATECONF=meta-evb/meta-evb-aspeed/meta-evb-ast2500/conf . openbmc-env`
`bitbake obmc-phosphor-image`

Distro: **Ubuntu 18.04.5 LTS**

amboar, 2021-02-24:
@shenki might be across what's going wrong here.

bamontales, 2021-02-25:
I've got some help from ASPEED Tech Support and the below instruction solves this issue.

Enter OpenBMC build environment:
```
git clone https://github.com/openbmc/openbmc openbmc
cd openbmc
TEMPLATECONF=meta-evb/meta-evb-aspeed/meta-evb-ast2600/conf . openbmc-env build-ast2600
```

Add the following lines to the end of build-ast2600/conf/local.conf file for supporting eMMC:
```
require conf/distro/include/phosphor-mmc.inc
```

The default eMMC partition layout (16G) from OpenBMC is too big for AST2600 EVB eMMC (4G), please edit openbmc/meta-aspeed/wic/emmc-aspeed.wks.in file to reduce the size, for example applying the following patch:
```
diff --git a/meta-aspeed/wic/emmc-aspeed.wks.in b/meta-aspeed/wic/emmc-aspeed.wks.in
index ee0fbffd8..b5d8fdcfd 100644
--- a/meta-aspeed/wic/emmc-aspeed.wks.in
+++ b/meta-aspeed/wic/emmc-aspeed.wks.in
@@ -26,5 +26,5 @@ part --fstype=ext4 --fixed-size 64M --source bootimg-partition --label boot-a
 part --fstype=ext4 --fixed-size 64M --source bootimg-partition --label boot-b
 part --fstype=ext4 --fixed-size 1G --source rootfs --label rofs-a
 part --fstype=ext4 --fixed-size 1G --source rootfs --label rofs-b
-part --fstype=ext4 --fixed-size 7G --label rwfs
-part --fstype=ext4 --fixed-size ${WKS_HOSTFW_SIZE} --source rootfs --rootfs-dir=${DEPLOY_DIR_IMAGE}/hostfw/image/ --label hostfw
+part --fstype=ext4 --fixed-size 1G --label rwfs
+# part --fstype=ext4 --fixed-size ${WKS_HOSTFW_SIZE} --source rootfs --rootfs-dir=${DEPLOY_DIR_IMAGE}/hostfw/image/ --label hostfw
```

Then build the image:
```
bitbake obmc-phosphor-image
```

closed 2021-02-25 by bamontales

amboar, 2021-02-25:
thanks for the update @bamontales. @anoo1 looks like there could be some improvement to the emmc support so we don't have bad out-of-box experiences.

anoo1, 2021-02-25:
> @anoo1 looks like there could be some improvement to the emmc support so we don't have bad out-of-box experiences.

Like more documentation?
By default, openbmc builds an image for a 32MB NOR, which in this case doesn't fit the u-boot image. The size of the NOR image can be changed with the FLASH_SIZE variable. Or UBI or MMC configs can be chosen.

Maybe we should upstream the changes from @bamontales if the ast2600 can only support the MMC configuration.