openbmc/openbmc #3765: troubles with 'Start QEMU with downloaded Romulus image'

TheDarhi, 2021-03-02:
Hello,

I have a plroblems with 'Start QEMU with downloaded Romulus image'. When start a script, process stoped on a 'romulus login' or 'Phosphor Power Control'.
https://imgur.com/a/qYPJYGy
What can i do wrong?

geissonator, 2021-03-03:
It looks like you reached the login prompt. You should just login?
https://github.com/openbmc/docs/blob/master/development/dev-environment.md#download-and-start-qemu-session

TheDarhi, 2021-03-05:
Hello ge,

When in need to do a 'Download and Install SDK' part? When i in QMENU or out? 
Because, when i login in QMENU how u say, here i cant install sdk, here is no left space.
https://imgur.com/gallery/XRGAtu4

bjwyman, 2021-03-05:
You would want to install the SDK outside of the QEMU environment, such as in another terminal session.

That is the software development kit, allowing you to cross-compile binaries on your host that you can then run on the target system (QEMU).

TheDarhi, 2021-03-09:
I want use QEMU, but i mean when i need to do this command? Uder QUEMU root@romolus or Under System Root root@debian?
I attache screens, in romolus no space.

amboar, 2021-03-09:
> You would want to install the SDK outside of the QEMU environment, such as in another terminal session.

I think this is a little unclear, but what @bjwyman means is that the SDK is only ever installed outside of the QEMU environment. Specifically, the SDK currently needs to be installed on an x86-64 host system, and it will cross-compile binaries for the image you're running under QEMU.

With the right QEMU invocation, you can use `scp` to copy the binaries into your running session.

amboar, 2021-03-09:
Further, please don't install the SDK for use as root :) Just use a regular user; we make no promises about not destroying your system.

TheDarhi, 2021-05-25:
@amboar @bjwyman @geissonator  Hello,

I record my installation process, can u saw it and say what i do wrong?   --->  https://youtu.be/lz1jfaDyeUE

amboar, 2021-05-25:
From the documentation:

> Login using default root/0penBmc login (Note the 0 is a zero).

TheDarhi, 2021-05-26:
@amboar Hello,

if i in root@romulus , all true?

What i need to do for a starting openbmc?

TheDarhi, 2021-05-26:
@amboar upd

I do all on a notebook and romulus login is doing automaticly.

And all process stoped on Started Phosphor Power0 Control ---> https://imgur.com/a/SsBviW5
What is it? Is a true way?

If yes, what can i do next?

amboar, 2021-05-27:
> And all process stoped on Started Phosphor Power0 Control ---> https://imgur.com/a/SsBviW5
> What is it? Is a true way?

It's waiting for you to log in. Press 'Enter' and the login prompt will reappear.

> If yes, what can i do next?

Whatever you like? What are you trying to do?

TheDarhi, 2021-06-03:
@amboar 

> Whatever you like? What are you trying to do?

To begin with, I would like to deploy it on a virtual machine and see what this software is. Get acquainted with the web interface.

Next, I will try to compile it, and install it on the server motherboard on NVRAM. 
If you know the articles or you have documentation that talks about the implementation of what I need, it would be very cool if you shared these materials.

edtanous, 2022-05-24:
In the future, please feel free to ask questions on the mailing list or discord, which generally have more active participants.  Closing as resolved.

closed 2022-05-24 by edtanous